IntelLabs/kAFL #8: kvm.h patch file problem

B1aNB1aN, 2020-07-16:
There is a problem about kvm vmx pt io control code defined in kvm.h patch file.
The io code KVM_VMX_PT_ENABLE_ADDR3 is conflict with io code KVM_KVMCLOCK_CTRL in kvm.h.
You can see the [issus](https://github.com/RUB-SysSec/redqueen/issues/6) i opened in project redqueen for more information.

il-steffen, 2020-07-16:
Thanks. It seems like the control codes are used on different file descriptors with their own respective handlers, so in my mind this lapse in counting should have no effect for now. Did you observe any impact?

B1aNB1aN, 2020-07-17:
No. Changing it or not depends on you. What i mean is that i want you guys notice it in case it will make kAFL unstable.

closed 2020-07-21 by B1aNB1aN