IntelLabs/kAFL #10: How to make kafl-ready VM snapshot?

vient, 2020-07-22:
Hi! I can't seem to understand how to launch kAFL, specifically I don't understand how to advance from "created VM with QEMU" state to "got `kafl` snapshot".

In README, you are referring to original README.kAFL, where I found following instructions:

> Execute loader binary <...>. VM should freeze. Switch to the QEMU management console and create a snapshot:

I tried to do the same with my Windows VM and vuln_test.exe loader. VM is not freezing by default when using kAFL patched QEMU. It is freezing when launched with non-patched QEMU, as well as when I add `-device kafl,reload_mode=False` to kAFL QEMU.

So I used the last method, QEMU+kAFL patch, added kafl device, executed loader, and VM freezed. Next step is to make a snapshot but QEMU hangs in `pause_all_vcpus()`.

What am I doing wrong? Can you elaborate on ways to prepare fuzzing snapshot?

il-steffen, 2020-07-22:
Hi! I have not tested this mode in a while, most of my targets just launch with -kernel. 

You definitely need the patched Qemu and Linux kernel module so that the kAFL hypercalls are understood. The lock() implementation has changed in Redqueen to perform the snapshot automatically. Have a look at handle_hypercall_kafl_lock() in pt/hypercall.c. It should still work the as described in the readme except you don't have to enter the savevm command yourself. Perhaps check if the snapshot was already created?

You can also try to replace it with the old implementation in kAFL: https://github.com/RUB-SysSec/kAFL/blob/1ece095a7a835887477393b2bba88247b1536d4f/QEMU-PT/pt/hypercall.c#L345

vient, 2020-07-22:
I suppose, the scheme is actually to put loader in VM, launch it (it will freeze and create a snapshot by using LOCK hypercall) and use this snapshot with kAFL?

il-steffen, 2020-07-22:
Yes, you need to run the loader once manually inside the VM *and* snapshot it to make the image ready for use in kAFL.

A little more detail: The first hypercall in the loader (linux_x86_64/src/loader.c) is the LOCK hypercall so (1) this requires a patched Qemu + KVM and (2) you launch this once, by calling qemu directly on commandline. kAFL will only ever use this snapshot, which makes your VM start at the point of that lock (see 'loadvm' in qemu.py). The fuzzer handshake is performed and the loader will (optionally) load the actual agent executable which can be the 'info' tool or one of the examples in linux_x86_64/bin/fuzzer/.

vient, 2020-07-22:
Ok, I am following you here. I successfully created a snapshot which is accepted by kAFL, thanks. Now I have another problem which may be relevant so I don't close issue yet:

kAFL expects RELEASE+PT_TRASHED commands in `__qemu_handshake` but loader (`targets/windows_x86_64/src/loader/loader.c`) sends ACQUIRE instead which causes immediate exit. I don't quite understand QEMU snapshot loading algorithm, but I see that next hypercall after LOCK is indeed ACQUIRE. Since ACQUIRE is first hypercall in all loaders I checked, I suppose it is intended to be so. 

I see that `KAFL_PROTO_PT_TRASHED` is sent only from `synchronization_lock()`, which is called only from `handle_hypercall_kafl_next_payload()`. which is called only on NEXT_PAYLOAD and USER_FAST_ACQUIRE hypercalls. How is this supposed to work? Can you point me in the direction of possible root cause of this problem?

il-steffen, 2020-07-22:
The protocol is a little crazy. I think it is correct to start with ACQUIRE on the guest though. That message does not actually match the ACQUIRE expected later on by Python. You can look at Zephyr for a working sample, it has the same flow except for LOCK (it simply reboots instead of restoring from snapshot).

Before we go there, do you actually see a warning from the python frontend about an unexpected message? qemu.py is checking and printing errors in these cases. You can also set self.debug_mode=True in qemu.py to get a more detailed log of the handshake.

If it just freezes or fails later after the handshake, chances are that other parts of the agent implementation are buggy or running into a first crash already as part of initialization. I have spend a lot of time debugging the handshake when actually the guest simply crashing due to other issues.

To answer your questions, if my notes and memory serve well, the protocol works like this:

* ACQUIRE from Guest to Qemu enables PT tracing (except at first time when handshake not done)
* RELEASE from Guest to Qemu disables PT tracing (only on first time it sets the handshake as done instead)
* NEXT_PAYLOAD from Guest to Qemu stops the VM, triggers an ACQUIRE message from Qemu to Python
* RELEASE from Python to Qemu will causes the VM to resume. It is issued after the fuzzing payload has been written to shared memory and is a response to the NEXT_PAYLOAD (seen as ACQUIRE message in Python)

So the first ACQUIRE seen by qemu.py as part of qemu_handshake() is actually triggered by the first NEXT_PAYLOAD hypercall. This and any subsequent ACQUIRE messages are answered by writing the new input to the payload buffer and then sending a RELEASE command to Qemu (see set_payload/send_payload).

Qemu can insert a PT_TRASHED message to signal a problem in PT decoding. I have not actually seen this happen but PT has a bunch of Errata. The guest can also insert a range of additional hypercalls after the handshake (first ACQUIRE/RELEASE pair) to setup the CR3 filter or hook one or more PANIC/KASAN events. 

Finally, the USER_FAST_ACQUIRE seems to be a shortcut to bundle hypercalls  RELEASE, NEXT_PAYLOAD (and I think also SET_CR3) into one hypercall at the end of the loop. It is used in the usermode samples in Redqueen (tests/user_bench). I don't think you get the same speedup for kernel fuzzing, never tried it.

vient, 2020-07-23:
I can show you exact places:

Snapshot is created on https://github.com/IntelLabs/kAFL/blob/master/targets/windows_x86_64/src/loader/loader.c#L168. Note that next hypercall is ACQUIRE.

Then I am trying to use it with kAFL anf get `assert(False)` on this line https://github.com/IntelLabs/kAFL/blob/master/kAFL-Fuzzer/common/qemu.py#L476. As you see, `__qemu_handshake` expects RELEASE or PT_TRASHED to be received first which contradicts (as I understand) your statement that ACQUIRE should be sent first (which is what loader is doing).

So, on this line I get `Got b'R', Expected: b'DZ'` and `Slave has died`.

vient, 2020-07-23:
Well, maybe I misunderstood the situation a bit. Here is the log with `targets/windows_x86_64/src/fuzzer/hprintf_test.c`:

```
[HPRINTF]    [+] Starting... C:\Users\ADMINI~1\AppData\Local\Temp\\fuzzing_engine.exe
[+] Allocating buffer for kAFL_payload struct
[+] Memset kAFL_payload at address 150000 (size 131072)
[+] Attempting to open vulnerable device file (\\.\testKafl)
[+] Submitting buffer address to hypervisor...
[+] Submitting current CR3 value to hypervisor...
[FATAL] Slave 0: Error in debug_recv(): Got b'R', Expected: b'DZ'
Process Slave 0:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/user/kafl/kAFL-Fuzzer/fuzzer/process/slave.py", line 59, in slave_loader
    slave_process.loop()
  File "/home/user/kafl/kAFL-Fuzzer/fuzzer/process/slave.py", line 121, in loop
    if not self.q.start():
  File "/home/user/kafl/kAFL-Fuzzer/common/qemu.py", line 461, in start
    self.__qemu_handshake()
  File "/home/user/kafl/kAFL-Fuzzer/common/qemu.py", line 476, in __qemu_handshake
    self.__debug_recv_expect(qemu_protocol.RELEASE + qemu_protocol.PT_TRASHED)
  File "/home/user/kafl/kAFL-Fuzzer/common/qemu.py", line 347, in __debug_recv_expect
    assert False
AssertionError
[FATAL] Slave has died - check logs!
```

So it dies somewhere in this range https://github.com/IntelLabs/kAFL/blob/master/targets/windows_x86_64/src/fuzzer/hprintf_test.c#L60-L65.

il-steffen, 2020-07-23:
Hi vient,

Please note that the ACQUIRE send by guest is not the same as the ACQUIRE seen by qemu.py. Qemu is in between those two and does not actually forward the ACQUIRE. It just causes Qemu to lock(), and wait for a RELEASE message from Python. It is the NEXT_PAYLOAD that triggers the ACQUIRE on Python side.

On your HPRINTF log, it seems that qemu.py is still in handshake mode while your guest is already executing the NEXT_PAYLOAD/ACQUIRE/RELEASE loop. The initial RELEASE from the loader (right after lock()) should have gotten you past that failing line 476. Can you check what happened to the RELEASE/ACQUIRE hypercalls from the loader?

The first NEXT_PAYLOAD inside the agent should then trigger the ACQUIRE message in line 487. It is correct to perform the ACQUIRE hypercall only after NEXT_PAYLOAD since it will simply lock the VM until Python sends a RELEASE in send_payload().

vient, 2020-07-23:
It seems that all hypercalls in the loader after LOCK are "lost" in my case. Thus, I'm returning to the "create snapshot" stage.

vient, 2020-07-23:
Also, I want to note that using `-machine pc-q35-2.4` in QEMU command line leads to host hangup after loader launch. I'm using `-machine pc-i440fx-2.6` as a bypass. VM OS is Windows, maybe that's why I see this problem.

il-steffen, 2020-07-23:
Yes, I updated that recently since the UEFI target works better with q35 and also seems to launch a little faster. Perhaps try to snapshot again with -machine pc or whatever works for your target. It is probably a good idea to use the same settings on snapshot resume.

vient, 2020-07-23:
It seems that hypercall "mismatch" happens because of race between QEMU and kAFL. I patched kAFL to run QEMU under GDB so I can set breakpoint on hypercall dispatching in `kvm_cpu_exec`, after second break I delete the breakpoint and fuzzing continues normally. Without GDB, hypercall "mismatch" happens every time.

il-steffen, 2020-07-23:
Can you clarify who is failing to wait on what here? Looking back at the thread, you are saying the guest is sending the RELEASE too fast for Qemu/kAFL to handle it properly? Does it help if you insert a couple msec waiting time after the LOCK?

vient, 2020-07-24:
Yes, you understand right, QEMU executes faster than kAFL, and kAFL misses some commands from QEMU. Adding 

``` cpp
volatile unsigned long long t;
for (t = 0; t < 1000000000ull; ++t) {}    // 1000*1000*1000
```

seems to help (it's about a 3 seconds delay on my machine). Amusingly, `Sleep` WinAPI does not help, maybe there is something wrong with system clock in VM. When I try to make delay smaller by removing a zero (making it ~0.3 seconds), `kafl_info.py` still works but `kafl_fuzz.py` is no more working.

_____

> Edit: host hangups seem to appear randomly

Restarting display manager helps but I still don't understand the root cause. I'm using kAFL on fully updated Ubuntu 20.04, tried to switch from gdm3+Gnome to lightdm+xfce4, nothing changed.

il-steffen, 2020-07-28:
I can't reproduce this locally ATM, most of my targets just work with -kernel/-initrd and I managed to stay away from Windows development thus far.

I'm not sure how we are losing the initial RELEASE "char" on the control socket. AFAIK it should stay in the pipe until read by kAFL. 
[0013-lock-on-RELEASE.patch.gz](https://github.com/IntelLabs/kAFL/files/4988841/0013-lock-on-RELEASE.patch.gz)
Perhaps it helps to make the guest wait for kAFL issuing the RELEASE command in turn. The attached patch should do that. I don't see any changes when testing with my own targets.

The host hangups are curious. The only issues I'm aware of are insufficient disk/shm space and Qemu sometimes going OOM after a couple days. Perhaps start a separate issue if you think this is related to kAFL.

vient, 2020-07-28:
Applying this patch leads to `Imported payload produced no new coverage, skipping...` message, with and without for-loop sleep in loader. Test cases added in runtime to `imports` folder also produce this message.
`kafl_info.py` also does not work with this patch.

il-steffen, 2020-07-28:
Hmm, but did get you to the same point as with the busy-loop above? It seems you are getting past the handshake now.
There is no further console output beyond these startup messages and possible runtime errors/warnings. Have you tried to open kafl_gui.py on the workdir to see the status?

There is a known issue around seed import which can result in an empty queue (zero "Paths") if you supply too few seeds. In this case the GUI will show the fuzzer stuck in [waiting] or "import" stage. Supplying a couple more seeds (~3-10) in the seed directory or at runtime via imports/ will typically fix this issue.

vient, 2020-07-29:
> Have you tried to open kafl_gui.py on the workdir to see the status?

Yes, it shows "importing" constantly with 1 total exec.

> Supplying a couple more seeds (~3-10) in the seed directory or at runtime via imports/ will typically fix this issue.

I tried to add additional tests to `imports` a couple of times, nothing changed. What concerns me more btw is that `kafl_info` does not work too. It means that I can't obtain driver IP range to fuzz anymore.

I'll try to look into this issue a bit later but since sleeping fix works good, can't promise anything.

il-steffen, 2020-07-29:
Ok let me know. I agree there is something fishy going on with kafl_info hanging like this. I will test it some more with a Linux image.

il-steffen, 2021-04-05:
'sleep' issue should be fixed in 20bc950

closed 2021-04-05 by il-steffen