IntelLabs/kAFL #12: Redqueen question

vient, 2020-08-04:
Hi! 

I noticed problems getting full coverage on [test Windows driver](https://github.com/IntelLabs/kAFL/blob/master/tests/test_cases/simple/win10_x86-64/Driver.c#L13-L57). It looked like Redqueen was not working sometimes, and kAFL was not finding full coverage test set even after tens of millions of attempts. Initially I suspected a race condition occurring somewhere between kAFL Redqueen and QEMU Redqueen, but after reading the code found [this line](https://github.com/IntelLabs/kAFL/blob/master/kAFL-Fuzzer/fuzzer/technique/redqueen/cmp.py#L58). After removing it, Redqueen started to work perfectly, full coverage test set was found in seconds using Redqueen and not deterministic or havoc approaches.

My question is: looks like you added this line, I don't see it in original RUB-SysSec/redqueen, so what's its purpose? Is it somehow beneficial on large targets?

il-steffen, 2020-08-04:
Hi,

The line is not in original redqueen but in the later grimoire code base which our python frontend is based on: https://github.com/RUB-SysSec/grimoire/blob/fae90ee436fe626f54ec2421269de8340cf06bf9/fuzzer/technique/redqueen/cmp.py#L65

It looks like this code is trying to reduce the search space of redqueen by ignoring things that are surely found by AFL later on. I guess only @eqv can say which is supposed to work better.

Could you try without effector maps (-d) and see if that fixes things?

eqv, 2020-08-04:
Yeah that's pretty much it. No point in trying mutations that would be done by normal mutators anyway. I don't think there will be any serious side effects if you remove that condition. In long (24h or so) fuzzing campaigns it will probably not matter much either way. For short campaigns that can probably increase the performance significantly.

vient, 2020-08-04:
Mentioned line deleted (better say, `is_simple_mutation` function changed to `return lhs == rhs`), effector maps on, 3 runs: full coverage (59 edges) found in ~30 seconds.

Line not deleted, effector maps off, 3 runs: 2/3 attempts failed to produce full coverage in 5 minutes, both found only 36/59 edges. 1/3 attempt found full coverage in 30 seconds.

The time here (30 seconds) may be not representative since I'm using `sleep` fix mentioned in https://github.com/IntelLabs/kAFL/issues/10#issuecomment-663480222. ~8 seconds are spent on QEMU restart, which happens two times in full coverage cases (initial start + 1 crash found). Performance is ~3000 execs/s.

vient, 2020-08-04:
>  In long (24h or so) fuzzing campaigns it will probably not matter much either way

So, the effect was not measured. I think of the opposite argument here: using Redqueen no matters what helps us to get easily achievable coverage fast. Maybe it does not matter in the long run but it gives us a nice opportunity to find something interesting right from the start so we can triage it while fuzzing continues. The question here is, how costly Redqueen becomes on large targets? Is it possible that your change in Grimoire may be seen as increasing performance?

_____

I am performing a long run with your patch and with effector maps disabled. This time, by the power of random, it found 29/59 edges in first 15 seconds. No new coverage was found in 15 minutes (in 3 million execs), I am continuing fuzzing.

eqv, 2020-08-04:
We did experiments with this. The main issue was that the Redqueen stage is performed as one large atomic chunk of work. With large inputs that contain many candidate positions this can delay the havoc stage and scheduling of newly found inputs. In some of our cases this drastically delayed the fuzzing process. The better option would be to make the redqueen stage interruptible so that other inputs and havoc can be scheduled in the mean time.

il-steffen, 2020-08-04:
> Line not deleted, effector maps off, 3 runs: 2/3 attempts failed to produce full coverage
> in 5 minutes, both found only 36/59 edges. 1/3 attempt found full coverage in 30 seconds.

Thanks. That excludes a recent extension I did to the deterministic stages from possible source of errors. I will try to reproduce this for a different target.

vient, 2020-08-04:
I see your point.

Also, I've only now noticed that I was running with `-D` option... Turning deterministic mode back on indeed produces nice results with your patch.

While the original issue is a false alarm, it may be beneficial to make `is_simple_mutation` check in Redqueen dependent on the deterministic mode availability. Other way may be to show a warning when `-redqueen` is supplied in the same time as `-D`.

Another thing to note, [Redqueen check](https://github.com/IntelLabs/kAFL/blob/master/kAFL-Fuzzer/fuzzer/technique/redqueen/cmp.py#L74) is performed against `helper.AFL_ARITH_MAX` while `config.config_values["ARITHMETIC_MAX"]` is used [in state logic processing](https://github.com/IntelLabs/kAFL/blob/master/kAFL-Fuzzer/fuzzer/state_logic.py#L408).

_____

Last question: why [this check](https://github.com/IntelLabs/kAFL/blob/master/kAFL-Fuzzer/fuzzer/technique/redqueen/cmp.py#L76) is used? Processing zeroes produces too much tests as well?

il-steffen, 2020-08-04:
FYI: In my experience RQ is a lot more effective that deterministic or havoc stages, i.e. it typically finds much more in significantly less time. I noticed this so strongly that I changed the scheduler.py and queue.py to focus on Redqueen and also schedule new inputs much earlier. Instead of linearly iterating through a queue and spending some time on each item, our queue is frequently re-sorted to put "promising" items and items that are still in early stages toward the top. Weights are balanced so that we (approximately) solve the init/redqueen/grimoire stages for all fav nodes first, then continue with deterministic stages for highest-fav nodes, and then also start processing non-favorite nodes. Havoc is last, although you will see very highly scored items (lots of new bytes) to enter havoc a little earlier. 

I also added a time limit + resume to the deterministic stages since we can be stuck there for a long time, when time may be better spend pushing any newly discovered item though RQ again.

I did not do a full-fledged benchmark series for this but for for any non-trivial sample I have this seems to work a lot better than the previous strawman scheduler. Based on that experience I would also say finding nodes earlier is exponentially better than waiting for AFL arithmetic phase. But I suppose it depends on the performance impact on RQ phase..

il-steffen, 2020-08-04:
> Also, I've only now noticed that I was running with `-D` option... Turning deterministic mode
> back on indeed produces nice results with your patch.

Ah, great. But what patch do you mean, this kAFL repo here?

> While the original issue is a false alarm, it may be beneficial to make `is_simple_mutation` check in
> Redqueen dependent on the deterministic mode availability. Other way may be to show a warning
> when `-redqueen` is supplied at the same time as `-D`.
>
> Another thing to note, [Redqueen check](https://github.com/IntelLabs/kAFL/blob/master/kAFL-Fuzzer/fuzzer/technique/redqueen/cmp.py#L74) is performed against `helper.AFL_ARITH_MAX` while
> `config.config_values["ARITHMETIC_MAX"]` is used [in state logic processing](https://github.com/IntelLabs/kAFL/blob/master/kAFL-Fuzzer/fuzzer/state_logic.py#L408).

Agreed. If RQ performance is still good with this change, I would simply make this the new default and only add an option to "ignore simple mutations found by deterministic stage". I will look at it over the next days, or feel free to post the patch yourself.

BTW are there any numbers to show the yield per time/execs on havoc vs deterministic? I agree that it makes things feel slow, but looking at algorithms I cannot imagine that disabling the deterministic stage gives a tangible advantage in the average case.

vient, 2020-08-04:
> what patch do you mean

`is_simple_mutation` one, which was introduced, as you said, in Grimoire...

> If RQ performance is still good with this change, I would simply make this the new default and only add an option to "ignore simple mutations found by deterministic stage".

...so you don't need to do anything.
_____
> BTW are there any numbers to show the yield per time/execs on havoc vs deterministic?

My small attempt:

### ordinary mode

fuzzer works at around 3450 exec/s

stopped after no new test was found for 30m, 31/59 edges

```
00:00:11: Found   1 from   0 using import       [a.                                        ] (favs=10, stage=final, exit=R, lvl=0, perf=0.01, score=0.09, t=325.21)
00:00:11: Found   2 from   1 using calibrate    [\xd3.\xd3.\x84\x84\x84l\x84\x84\x84.\xedx\] (favs=3, stage=final, exit=R, lvl=1, perf=0.00, score=0.29, t=59.14)
00:00:12: Found   3 from   1 using afl_havoc    [wdE.\xbe....\xff.\xbe\xbe\xb3\xbe\xbe\xaa\] (favs=4, stage=final, exit=R, lvl=1, perf=0.00, score=0.29, t=64.58)
00:00:12: Found   4 from   1 using afl_havoc    [PPPP\x80.\xba*K4PP\xba\xba*K4PPPj]\xd6P1\x] (favs=2, stage=final, exit=R, lvl=1, perf=0.00, score=0.43, t=34.54)
00:01:35: Found   5 from   3 using afl_splice   [Pw\x80\x80\x80\x80\x80\x80\x9dVp.\x97Npp\x] (favs=2, stage=final, exit=R, lvl=2, perf=0.00, score=0.45, t=33.65)
00:01:38: Found   6 from   5 using afl_arith_1  [Pwn\x80\x80\x80\x80\x80\x9dVp.\x97Npp\x9d\] (favs=2, stage=final, exit=R, lvl=3, perf=0.00, score=0.42, t=35.74)
00:06:05: Found   7 from   5 using afl_splice   [w0,---\x80--v\x8f\x8fwww                  ] (favs=4, stage=final, exit=R, lvl=3, perf=0.00, score=0.15, t=129.47)
00:06:09: Found   8 from   7 using afl_arith_1  [w00---\x80--v\x8f\x8fwww                  ] (favs=4, stage=final, exit=R, lvl=4, perf=0.00, score=0.15, t=130.10)
```

### no deterministic

stopped after no new test was found for 30m, 27/59 edges

```
00:00:11: Found   1 from   0 using import       [a.                                        ] (favs=10, stage=final, exit=R, lvl=0, perf=0.01, score=0.06, t=464.91)
00:00:11: Found   2 from   1 using calibrate    [\xd3.\xd3.\x84\x84\x84l\x84\x84\x84.\xedx\] (favs=2, stage=final, exit=R, lvl=1, perf=0.00, score=0.50, t=33.94)
00:00:11: Found   3 from   1 using afl_havoc    [wdE.\xbe....\xff.\xbe\xbe\xb3\xbe\xbe\xaa\] (favs=4, stage=final, exit=R, lvl=1, perf=0.00, score=0.15, t=129.86)
00:00:12: Found   4 from   1 using afl_havoc    [PPPP\x80.\xba*K4PP\xba\xba*K4PPPj]\xd6P1\x] (favs=1, stage=final, exit=R, lvl=1, perf=0.00, score=0.41, t=32.00)
00:01:31: Found   5 from   1 using afl_havoc    [Pw\x80\x80\x80\x80\x80\x80\x9dVp.\x97Npp\x] (favs=1, stage=final, exit=R, lvl=1, perf=0.00, score=0.36, t=35.85)
00:05:55: Found   6 from   1 using afl_havoc    [w0,---\x80--v\x8f\x8fwww                  ] (favs=4, stage=final, exit=R, lvl=1, perf=0.00, score=0.10, t=204.11)
00:12:44: Found   7 from   5 using afl_splice   [PwnnYnp.....nnnnnnnn\x80\xff\xff\xffdHQ\xb] (favs=5, stage=final, exit=R, lvl=2, perf=0.00, score=0.08, t=259.26)
```

### ordinary mode, `for j in range(1 << (1 + stacking))` changed to `for j in range(1 << stacking)` [here](https://github.com/IntelLabs/kAFL/blob/de8829798115096c3a8fece85c0051c8ab13e43d/kAFL-Fuzzer/fuzzer/technique/havoc.py#L59)

fuzzer works at around 3750 exec/s

full coverage found in 17:05, big part of it was found by deterministic after a lucky splice

```
00:00:11: Found   1 from   0 using import       [a.                                        ] (favs=10, stage=final, exit=R, lvl=0, perf=0.01, score=0.18, t=159.71)
00:00:11: Found   2 from   1 using calibrate    [\xe4\xe4\xfa...\xd3\xd3.\xd3\xd3\xd3\xd3\x] (favs=1, stage=final, exit=R, lvl=1, perf=0.00, score=0.49, t=32.76)
00:00:12: Found   3 from   1 using afl_havoc    [ww\xff\x98\x98\x98wwwwwwwww\x98\x98\x98\x9] (favs=1, stage=final, exit=R, lvl=1, perf=0.00, score=0.54, t=35.14)
00:00:13: Found   4 from   1 using afl_havoc    [PPPPPPPPPPPPFPKKKK>KKKKKKKKKKKK           ] (favs=1, stage=final, exit=R, lvl=1, perf=0.00, score=0.93, t=16.07)
00:01:31: Found   5 from   2 using afl_splice   [w0yw0y\x90z\x90\x90\x90\x90\x84\x90.\xff\x] (favs=1, stage=final, exit=R, lvl=2, perf=0.00, score=0.42, t=44.72)
00:08:16: Found   6 from   5 using afl_splice   [w00d.\xc0w0..\xc0\xc0\xc0\xc0\xc0^\xc0d\xa] (favs=12, stage=final, exit=R, lvl=3, perf=0.00, score=0.28, t=119.75)
00:08:18: Found   7 from   6 using afl_flip_1/1 [w00t.\xc0w0..\xc0\xc0\xc0\xc0\xc0^\xc0d\xa] (favs=3, stage=final, exit=R, lvl=4, perf=0.00, score=0.86, t=19.74)
00:15:58: Found   8 from   6 using afl_splice   [Pwmmmmmm{m\xff\xffmmmm`mmomm\xfa.mm\xfa..\] (favs=3, stage=final, exit=R, lvl=4, perf=0.00, score=3.83, t=4.18)
00:16:08: Found   9 from   8 using afl_flip_2/1 [Pwnmmmmm{m\xff\xffmmmm`mmomm\xfa.mm\xfa..\] (favs=2, stage=final, exit=R, lvl=5, perf=0.00, score=3.64, t=4.40)
00:16:23: Found  10 from   9 using afl_arith_1  [PwnTmmmm{m\xff\xffmmmm`mmomm\xfa.mm\xfa..\] (favs=2, stage=final, exit=R, lvl=6, perf=0.00, score=3.71, t=4.32)
00:16:36: Found  11 from  10 using afl_flip_1/1 [PwnTommm{m\xff\xffmmmm`mmomm\xfa.mm\xfa..\] (favs=2, stage=final, exit=R, lvl=7, perf=0.00, score=4.25, t=3.77)
00:16:51: Found  12 from  11 using afl_arith_1  [PwnTowmm{m\xff\xffmmmm`mmomm\xfa.mm\xfa..\] (favs=2, stage=final, exit=R, lvl=8, perf=0.00, score=4.07, t=3.93)
[... cut ...]
00:01:38: Found   6 from   5 using afl_arith_1  [Pwn\x80\x80\x80\x80\x80\x9dVp.\x97Npp\x9d\] (favs=2, stage=final, exit=R, lvl=3, perf=0.00, score=0.42, t=35.74)
00:06:05: Found   7 from   5 using afl_splice   [w0,---\x80--v\x8f\x8fwww                  ] (favs=4, stage=final, exit=R, lvl=3, perf=0.00, score=0.15, t=129.47)
```
Why was second test prioritized over the first one? They are both on the same level 3. I see that they have different amount of favs, what does it mean? Looks like fuzzer may need more than a half hour to even out test priority.

I am a bit concerned about afl_havoc performance in these tests, afl_splice did good though.

il-steffen, 2020-08-04:
The fav count is the number of new bytes (edges) that this input triggers on top of other/prior inputs. More specifically, there may be other inputs that also hit one of those edges but they have lower speed/score or similar. I think that logic is similar as in AFL queue culling (although probably not exactly the same and perhaps buggy). I use this as one of the strongest factors in the sorting function. The score + speed of the kafl_plot tool are supposed to match the corresponding values from the sorting function but I think they may be outdated.

Even this toy example seems to confirm that -D will significantly impact the achieved coverage/time. Only in the first 2 tests the results are, I would say, inconclusive (need more runs over longer time?). AFAIK, havoc is doing the same basic mutations but they are applied randomly and stacked, so you have an additional level of noise and collision to content with. My understanding is that this stage really only makes sense if the basic deterministic mutations have all been tried once, especially since they are finite and only need to be done once. If you have any other insight to this, I'm happy to learn.

You are saying havoc should be better - but compared to what, and how can we fix it? I reviewed the deterministic stages and havoc in kAFL against vanilla AFL a while back and they all seem to be equivalent. The major difference seems to be in the number of time/rounds invested in havoc, and that auto-dict is missing in our deterministic stage.

I agree that we spend too much time doing havoc on the initial/seed input. I suppose this is because the initial input is fast and short and gets all the fav bytes assigned that execute regardless of the input. The current scheduler already uses the time spent in havoc to reduce the overall score. We can use a stronger factor there to reduce the bias or remove it completely: [scheduler.py:98](https://github.com/IntelLabs/kAFL/blob/de8829798115096c3a8fece85c0051c8ab13e43d/kAFL-Fuzzer/fuzzer/scheduler.py#L98)

I'm happy to adjust or completely replace this scheduler but I think we need a more representative target and measure multiple runs over several hours. ARITH_MAX=128 looks interesting for example, but I would only make it default if it shows good results across multiple targets. Makes sense?

If you look at tests/user_bench/, there are some relatively simple scripts to get typical userspace targets like objdump and nasm working in a Linux busybox/initramfs environment. I think this can be a good starting point for a regular performance/regression test. Perhaps we can also get the [magma benchmark](https://hexhive.epfl.ch/magma/) working to get some more recent results to compare against.

vient, 2020-08-05:
> You are saying havoc should be better - but compared to what, and how can we fix it?

Well, I don't really know. What can you say about these tweets? https://twitter.com/gamozolabs/status/1284854239232053248 and https://twitter.com/gamozolabs/status/1284854896483659776.

I can also say that I have a feeling more time was spent in afl_splice than in afl_havoc on this target.

> ARITH_MAX=128 looks interesting for example, but I would only make it default if it shows good results across multiple targets. Makes sense?

Since it's deterministic stage, it surely won't make things worse. I think the question is, is it beneficial in the terms of edges per second to make deterministic stage 4 times longer? At least it can be said that that way we fully bruteforce each byte in the test before moving to havoc stage — a nice property.

> I think this can be a good starting point for a regular performance/regression test.

Yeah, I think I can start testing with some Linux programs. Do you consider Objdump and NASM to be the best two examples from user_bench? I don't have computing resource to run extensive tests on all programs there but may run one or two.

eqv, 2020-08-05:
I would strongly advise caution when changing large parts of the mutator on experiments with one or two targets. It's VERY easy to  overfit to one individual target, and similarly be fooled by randomness (even with up to 20 runs per target). We have seen multiple changes that seemed very helpful on 20 vs 20 runs in a given target that turned out to be not helpful at all.

il-steffen, 2020-08-05:
TDB:

> make is_simple_mutation check in Redqueen dependent on the deterministic mode availability.
> Other way may be to show a warning when -redqueen is supplied at the same time as -D.

> Another thing to note, Redqueen check is performed against helper.AFL_ARITH_MAX while
> config.config_values["ARITHMETIC_MAX"] is used in state logic processing.

Lets discuss benchmarking/optimizations separately.

il-steffen, 2020-11-23:
Commit 4559817 fixes these.

closed 2020-11-23 by il-steffen

kotee4ko, 2023-07-23:
Greetings!

I reach some troubles with redqueen feature: it doesn't handle `strcmp()` calls on linux userspace target.

Launch command:
```
kafl fuzz \
        --kernel /boot/vmlinuz-$(uname -r) \
        --initrd forkserver.cpio.gz \
        --memory 512 \
        --sharedir $PWD \
        --purge -t 2 -ts 1 --log-hprintf --log-crashes \
        --append "nokaslr oops=panic nopti mitigations=off console=ttyS0 hprintf=7" \
        -ip0 0x401000-0x4af000 \
        --redqueen --redqueen-hammer --redqueen-simple --redqueen-hashes --grimoire --radamsa \
        -p 1 --qemu-path /usr/src/QEMU-Nyx/x86_64-softmmu/qemu-system-x86_64

```

Some debugging of qemu-nyx approve, that redqueen strcmp() hooks are reached and works as expected, or, at least, see strcmp() calls is target and could access guest args.

The target, as always, is simple, statically built, small harness with several branches to test integers and strings extractions.

```
In file: /usr/src/QEMU-Nyx/nyx/redqueen.c
   948 static void format_strcmp(uint8_t *buf1, uint8_t *buf2)
   949 {
   950     char  out_buf[REDQUEEN_MAX_STRCMP_LEN * 4 + 2];
   951     char *tmp_hex_buf = &out_buf[0];
   952     for (int i = 0; i < REDQUEEN_MAX_STRCMP_LEN; i++) {
 ► 953         tmp_hex_buf += sprintf(tmp_hex_buf, "%02X", (uint8_t)buf1[i]);
   954     }
   955     *tmp_hex_buf++ = '-';
   956     for (int i = 0; i < REDQUEEN_MAX_STRCMP_LEN; i++) {
   957         tmp_hex_buf += sprintf(tmp_hex_buf, "%02X", (uint8_t)buf2[i]);
   958     }

pwndbg> p tmp_hex_buf 
$5 = 0x7fffeb000310 "C9"
pwndbg> p buf2
$6 = (uint8_t *) 0x7fffeb0004a0 "PWNED!!!"
pwndbg> p buf1
$7 = (uint8_t *) 0x7fffeb000460 "ɵ\032<\237b22\032\062\062\016\020g\022UUU"
```

full backtrace:

```
pwndbg> bt full
#0  0x0000555555c0f3a4 in format_strcmp (buf1=buf1@entry=0x7fffeb000460 "ɵ\032<\237b22\032\062\062\016\020g\022UUU", buf2=buf2@entry=0x7fffeb0004a0 "PWNED!!!") at /usr/src/QEMU-Nyx/nyx/redqueen.c:953
        i = <optimized out>
        out_buf = "C9\000\000\000\000\000\000 p\034\334\377\177\000\000\000\000\000\000\000\000\000\000@\244\210\004\000\000\000\000@s\311VUU\000\000\000\003홨\"\261\065`\003\000\353\377\177\000\000\b\000\000\000\000\000\000\000@\244\210\004\000\000\000\000\360\003\000\353\377\177\000\000h\376\377\377\377\377\377\377 p\034\334\377\177\000\000\016p\375\367\377\177\000\000S+\254UUU\000\000\b\000\000\000\000\000\000\000\260\"\265VUU\000\000\340\353\377\377\377\177\000\000\b\000\000\000\000\000\000\000@\244\210\004\000\000\000\000\000\003홨\"\261\065\357\200H", '\000' <repeats 13 times>, "\360\003\000\353\377\177\000\000\000\360\377\377\377\377\017\000\357\200H\000\000\000\000\000\210K\300UUU"...
        tmp_hex_buf = 0x7fffeb000310 "C9"
        res = 0x555555ac21c7 <flatview_read_continue+119> "L\001\365M\001\364L)\363\017\204Q\001"
        env = <optimized out>
        __func__ = "format_strcmp"
        rip = <optimized out>
        __PRETTY_FUNCTION__ = "format_strcmp"
#1  0x0000555555c0f948 in test_strcmp (arg1=140737136034976, arg1@entry=140737353969678, arg2=arg2@entry=4751599) at /usr/src/QEMU-Nyx/nyx/redqueen.c:1002
        cpu = 0x555556c97340
        buf1 = "ɵ\032<\237b22\032\062\062\016\020g\022UUU\000YXﾭ\300\000\000\000\357\276ޭ\276", '\363' <repeats 23 times>, "\255\336쭾\357\256", <incomplete sequence \374>
        buf2 = "PWNED!!!\000Something went wrong\n\000This should never happen :)\n\000xeon"
        __PRETTY_FUNCTION__ = "test_strcmp"
#2  0x0000555555c10024 in test_strcmp_sys_v () at /usr/src/QEMU-Nyx/nyx/redqueen.c:1035
        env = <optimized out>
        arg1 = 140737353969678
        arg2 = 4751599
        __func__ = "test_strcmp_sys_v"
        mode = 1
        cpu = <optimized out>
[... 88 lines cut ...]
        ret = <optimized out>
        pd = <optimized out>
        unwind_buf = {
          cancel_jmp_buf = {{
              jmp_buf = {140737136056064, 65258806425213630, 140737488343454, 140737488343455, 140737136035840, 8396800, -65286209098687810, -65243037933315394},
              mask_was_saved = 0
            }},
          priv = {
            pad = {0x0, 0x0, 0x0, 0x0},
            data = {
              prev = 0x0,
              cleanup = 0x0,
              canceltype = 0
            }
          }
        }
        not_first_call = 0
#13 0x00007ffff6944a2f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
pwndbg> 
```

But kafl doesn't insert findings to corpus.

Any advices?
Thanks.