IntelLabs/kAFL #13: Mutation and scheduling improvements

il-steffen, 2020-08-05:
> You are saying havoc should be better - but compared to what, and how can we fix it?

Well, I don't really know. What can you say about these tweets? https://twitter.com/gamozolabs/status/1284854239232053248 and https://twitter.com/gamozolabs/status/1284854896483659776.

I can also say that I have a feeling more time was spent in afl_splice than in afl_havoc on this target.

> ARITH_MAX=128 looks interesting for example, but I would only make it default if it shows good results across multiple targets. Makes sense?

Since it's deterministic stage, it surely won't make things worse. I think the question is, is it beneficial in the terms of edges per second to make deterministic stage 4 times longer? At least it can be said that that way we fully bruteforce each byte in the test before moving to havoc stage — a nice property.

> I think this can be a good starting point for a regular performance/regression test.

Yeah, I think I can start testing with some Linux programs. Do you consider Objdump and NASM to be the best two examples from user_bench? I don't have computing resource to run extensive tests on all programs there but may run one or two.

_Originally posted by @vient in https://github.com/IntelLabs/kAFL/issues/12#issuecomment-668898629_

vient, 2020-08-05:
I'm currently trying to fuzz objdump from tests/user_bench, and it behaves strangely in my opinion. While I have ~2500 exec/s with deterministic, Redqueen and Grimoire turned on, there are almost no new tests found, definitely not by those three deterministic steps. Do you think it's typical for objdump fuzzing?

Another thing I noted is that Radamsa fails sometimes: radamsa process dies (goes to T state) as well as slave process that launched it, after that this slave is lost forever for kAFL, it just sits there doing nothing. That's why I turned it off in `tests/user_bench/run.sh`.

Besides Radamsa, I just use provided scripts:
``` bash
cd $KAFL_ROOT
./tests/user_bench/build.sh binutils
./tests/user_bench/run.sh pack objdump
./tests/user_bench/run.sh run objdump
```

eqv, 2020-08-05:
We later removed radamsa as a mutator due to this issues.
I think the radamsa stuff is due to the fact that we used radamsa in network mode. If you use radamsa in the mode where it writes 500 files into a folder, it should be doing fine. Might be very nice patch.

il-steffen, 2020-08-05:
> Well, I don't really know. What can you say about these tweets?
> https://twitter.com/gamozolabs/status/1284854239232053248
> https://twitter.com/gamozolabs/status/1284854896483659776.

Sure, definitely worth a try. He seems to work without deterministic stage though so I would like to see some benchmarks to understand how this perfoms best. Can it replace afl_havoc/afl_splice, or should we add it to the list next to other havoc methods, or perhaps should this be one of the havoc_handlers[] that are selected during stacked mutations?

You may have already seen in [state_logic.py:L256](https://github.com/IntelLabs/kAFL/blob/de8829798115096c3a8fece85c0051c8ab13e43d/kAFL-Fuzzer/fuzzer/state_logic.py#L256) that "havoc" is also a general stage in kAFL. It implements AFL havoc/splice as well as havoc-style mutations based on RQ/grimoire/radamsa. I suggest to add this mutator there and make afl_havoc/splice optional for purpose of benchmarking. We can also try to add it to havoc_handler.py as extension of afl_havoc stacked mutations. 

> I can also say that I have a feeling more time was spent in afl_splice than in afl_havoc on this target.
>
> Since it's deterministic stage, it surely won't make things worse. I think the question is, [...]

Agreed. This will impact the required #execs but I also noticed that a lot of these are filtered out by is_not_bitflip(). If this really works well (as indicated in your last test run) perhaps we should try a systematic single-byte-bruteforce as a replacement for all the single-byte bitflip/arith/int mutations.

> Do you consider Objdump and NASM to be the best two examples from user_bench?
> I don't have computing resource to run extensive tests on all programs there but may run one or two.

This is a tough one. As a baseline I think we need something simple/binary to balance the whole pipeline of stages/mutators, as well as something with exploding paths (the master/scheduler can become a performance bottleneck if you get lots of findings per second, but more importantly for real life we want to be sure the scheduler works reasonable with huge queues). So objdump or djpeg for "simple" and something like nasm (bison, bin/who?) for exploding paths? To see the benefit of grimoire and/or radamsa you need something with syntax, like libxml2 or json.

il-steffen, 2020-08-05:
I revised the radamsa integration and thought this to be fixed, but it seems I was wrong. 
I did see consistent improvements when fuzzing something like json with radamsa enabled. However I also didn't do the n hours many-times benchmark across targets.

il-steffen, 2020-08-05:
Completely agree with [@eqv](https://github.com/IntelLabs/kAFL/issues/12#issuecomment-669129119
) comment. Emphasis in above comment should be on "baseline", and "best two examples from user_bench".
 It should not be difficult to include some of the current favored targets from fuzz-bench or magma and then we should verify across a range of these.

il-steffen, 2020-08-05:
> Do you think it's typical for objdump fuzzing?

Not sure. I think it takes some time to find the expected input headers. You can supply a seed and/or compare against stock AFL fuzzing. There may also be some major bugs here which we can fix without large-scale benchmarks.

vient, 2020-08-05:
> I'm currently trying to fuzz objdump from tests/user_bench, and it behaves strangely in my opinion. While I have ~2500 exec/s with deterministic, Redqueen and Grimoire turned on, there are almost no new tests found, definitely not by those three deterministic steps (well, Grimoire immediately finds empty test, that's all). Do you think it's typical for objdump fuzzing?

I am performing two runs of objdump fuzzing, each with 4 workers, second one has ARITH_MAX=128.

70 minutes passed, I have 7 and 8 tests found respectively, all of them by havoc or splice. Looks suspicious.

il-steffen, 2020-08-05:
> objdump fuzzing / only 7 and 8 tests found respectively

Yes, I think I mixed up objdump vs readelf. nasm also seem to work as expected. For djpeg and perhaps also objdump, the script should probably take the -ip0 range for the library and not the main executable. 

Sorry, the scripts are more meant as a getting started help. Its difficult to make them work across platforms and targets.

vient, 2020-08-05:
> objdump seems to work, just throw in a proper .elf as seed

Confirmed (with readelf). Strange, I expected it to find ELF signature at least, guess that's not how readelf is working.

> Note that the "Edges" GUI widget is massively overcounting edges

Is there any other metric in kafl_gui that can be used to compare fuzzer instances?

il-steffen, 2020-08-05:
> Is there any other metric in kafl_gui that can be used to compare fuzzer instances?

Well, I use Edges :-) The overcount is not usually so strong but I just noticed 13k edges in objdump which is probably wrong. I tend to have a slight overcount from random/non-deterministic inputs for which I added the -funky mode.

vient, 2020-08-05:
What can you say about `p(col)`? Are large values bad? For reference, I have p(col) 80% after fuzzing readelf for 7 hours. As I understood from source, increasing bitmap size may be non-trivial.

il-steffen, 2020-08-05:
You may have seen that 'Edges' in the GUI is actually just the byte count in the bitmap. So this approximates the real edges is subject to collisions or non-deterministic ('funky') behavior. I don't trust those 13k or even 52K edges, I would definitely try the cov tool to validate these. Perhaps the nasm or bison samples work better, sorry :+1: 

p(col) is just a GUI feature and directly computed from Edges + bitmap size. I never tried larger bitmap sizes but this is probably just a matter of ensuring all instances of bitmap_size constants are updated across python and qemu-pt code.

Regarding kafl_tfilter, I believe this is completely unused. Best I can tell there were some hypercalls in the original kAFL which allowed you to blacklist certain addresses/edges with the goal to ignore non-deterministic code portions. The parallel mode works out-of-the-box for the provided bios/zephyr/user_bench samples since these do not use snapshots. They tend to boot fast enough and the user_bench/ samples have their own forkserver. For the Windows/Linux samples using -vm_dir commandline you have to provide multiple RAM overlays so that each Slave instance will pick up its respective overlay_<n>.qcow file. I never actually tried this but I guess copying them might just work.

il-steffen, 2020-08-07:
I found the problem with the djpeg sample, you need to copy djpeg-static rather than djpeg to include libjpeg into the -ip0 range. libpng also seems to work. well. I think these are better for initial testing. The objdump/readelf/nasm produce so many paths that you spend the first day just in redqueen and deterministic stages.

il-steffen, 2020-08-08:
Commit 9e42306 fixes djpeg and adds pngtest and xmllint. xmllint triggers a lot of Qemu crashes, need to check..

vient, 2020-08-13:
Radamsa seems to be working now.

il-steffen, 2020-08-13:
FYI It seems that favorites are currently filtered not just by covered edges and performance but also based on same bit count per edge. Removing the val/old_val checks reduces the fav count quite a bit and seems to achieve better coverage over time. Opinions? https://github.com/IntelLabs/kAFL/blob/e61b553017b0b0d7fdbffaa98e84991925327783/kAFL-Fuzzer/fuzzer/queue.py#L112

eqv, 2020-08-13:
I think that's somewhat of an open question. I would be surprised if this was making a big difference one way or the other in the long run/across many targets.

closed 2022-02-14 by il-steffen