IntelLabs/kAFL #14: Slave has died — fs_shm.seek(0) error in set_payload

vient, 2020-08-06:
So, I tried fuzzing readelf from user_bench a bit, and after several hours I got this:

```
[FATAL] Failed to set new payload - Qemu crash?
[FATAL] Failed to set new payload - Qemu crash?
[FATAL] Failed to set new payload - Qemu crash?
[FATAL] Failed to set new payload - Qemu crash?
Process Slave 2:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/vient/git/kAFL/kAFL-Fuzzer/fuzzer/process/slave.py", line 59, in slave_loader
    slave_process.loop()
  File "/home/vient/git/kAFL/kAFL-Fuzzer/fuzzer/process/slave.py", line 133, in loop
    self.handle_node(msg)
  File "/home/vient/git/kAFL/kAFL-Fuzzer/fuzzer/process/slave.py", line 107, in handle_node
    results, new_payload = self.logic.process_node(payload, meta_data)
  File "/home/vient/git/kAFL/kAFL-Fuzzer/fuzzer/state_logic.py", line 92, in process_node
    new_payload = self.handle_initial(payload, metadata)
  File "/home/vient/git/kAFL/kAFL-Fuzzer/fuzzer/state_logic.py", line 185, in handle_initial
    havoc.mutate_seq_havoc_array(payload, self.execute, num_execs)
  File "/home/vient/git/kAFL/kAFL-Fuzzer/fuzzer/technique/havoc.py", line 64, in mutate_seq_havoc_array
    func(data)
  File "/home/vient/git/kAFL/kAFL-Fuzzer/fuzzer/state_logic.py", line 324, in execute
    bitmap, is_new = self.slave.execute(payload, parent_info)
  File "/home/vient/git/kAFL/kAFL-Fuzzer/fuzzer/process/slave.py", line 251, in execute
    exec_res = self.__execute(data)
  File "/home/vient/git/kAFL/kAFL-Fuzzer/fuzzer/process/slave.py", line 245, in __execute
    return self.__execute(data, retry=retry+1)
  File "/home/vient/git/kAFL/kAFL-Fuzzer/fuzzer/process/slave.py", line 245, in __execute
    return self.__execute(data, retry=retry+1)
  File "/home/vient/git/kAFL/kAFL-Fuzzer/fuzzer/process/slave.py", line 245, in __execute
    return self.__execute(data, retry=retry+1)
  File "/home/vient/git/kAFL/kAFL-Fuzzer/fuzzer/process/slave.py", line 235, in __execute
    self.q.set_payload(data)
  File "/home/vient/git/kAFL/kAFL-Fuzzer/common/qemu.py", line 737, in set_payload
    self.fs_shm.seek(0)
ValueError: mmap closed or invalid
[FATAL] Slave has died - check logs!
```

What may have caused it? I had 8 workers, one of them died, others were perfectly normal. I observed same pattern at least one more time (one worker dies, other work) but since it takes a long time I don't know if other workers would die too eventually. Also, kafl_gui was running in both cases all the time. In this particular case kafl_gui have frozen, restart helped.

vient, 2020-08-06:
> Maybe something can be done to restart broken slaves automatically?

Seems like this is [a known TODO](https://github.com/IntelLabs/kAFL/blob/9bf87d87325f081b3e286c40cab699134c5987bb/kAFL-Fuzzer/fuzzer/communicator.py#L47).

il-steffen, 2020-08-06:
I've seen this before. After crash or restart the qemu handshake and (I think) even setting a dummy payload as part of handshake will work, but it fails afterwards on the first set_payload(). The error points to an invalid mmap descriptor and adding a [small delay](https://github.com/IntelLabs/kAFL/blob/9bf87d87325f081b3e286c40cab699134c5987bb/kAFL-Fuzzer/common/qemu.py#L533) on restart() seemed to fix the issue. I could not root-cause the bug but putting the time.sleep() there allowed me to remove it in slave.py and other tools.

That restart() function takes a shortcut in case of '-forkserver', which is exactly the user_bench execution mode. Can you try adding a similar time.sleep() before the early return in forkserver mode? If that does not fix the issue I can try to bisect my older repo.

The real issue here seems to be a race on the mmap/shm files but I could not resolve it back then. Both qemu.py and Qemu are doing a truncate on the files and both maintain their own SHM size limits. Its probably better if one end sets up the files and sizes and the other only checks + confirms.

Regarding slave restart, the slave.py will already try to restart qemu and that tends to work on random communication errors or crashes. As you can see above it actually tried to __execute() 4 times. That does not help on consistent/fatal errors of course. However it would be nice if the GUI would clearly indicate which instances are actually not executing (it will currently just print the last stored exec/s). Also a general resume function would be nice and should be feasible given all the node info is already stored in the workdir..

FYI, the GUI will usually not interfere with anything. If you look at the code, it just polls the various status files in the workdir.

vient, 2020-08-07:
> Can you try adding a similar time.sleep() before the early return in forkserver mode?

It didn't help, during night run, 120M execs, 2/12 slaves dies.

il-steffen, 2020-08-07:
Removing that whole 'if -forkserver:' clause seems to help as workaround. This way we have a full Qemu exit/restart on errors but also on timeout/crash/kasan which the userspace forkserver would have catched for us. There is a significant slowdown if your app has lots of crashes, like the LAVA samples.

il-steffen, 2020-08-08:
Commit 9e42306 uses a separate API to reset qemu on socket errors (typically a Qemu crash) while keeping the forkserver advantage when resetting just the target inside a VM. Recovering from transient Qemu crashes should work properly now for these usermode/forkserver targets.

vient, 2020-08-13:
QEMU seems to be restarting properly now, thanks.

closed 2020-08-13 by vient