IntelLabs/kAFL #15: Failed to reuse ram.qcow2 in parallel fuzzing mode.

Bl1nnnk, 2020-08-26:
The snapshot file "ram.qcow2" is used concurrently by the instances in parallel fuzzing mode. But maybe modern QEMU introduced a new feature "file lock"?

The error log looks like:
```
qemu-system-x86_64: -hdb /home/jungu/kafl/snapshots/win8_x64/ram.qcow2: Failed to get "write" lock
```

Does this error take place in your environment?

il-steffen, 2020-08-28:
Hi! I have not seen this but I also barely use the VM-style launching.
Have you tried with qemu-4.2? Just enable it in install.sh and point your kafl.ini to it.

nola-milkin, 2020-08-30:
Hello! I have the same problem and this error appear on both qemu-5.0 and qemu-4.2.
I solved it just to take advantage of parallel fuzzing by creating another `ram.qcow2`. I have rewritten in `qemu.py` to use `vm_ram` exactly the same as `vm_dir` (`ram_<id>.qcow2` and `overlay_<id>.qcow2`). Is there a prettier way?

il-steffen, 2020-08-30:
It seems this is a feature and there are half-documented options to get around it.
https://bugs.launchpad.net/juju/+bug/1816170/comments/8

I'm not sure from the description if you just need to supply -U/--force-share to qemu-img when creating the backing image, or supply shared-rw (renamed to force-share=on?) on qemu cmdline, or both. Also it looks like force-share=on may only work for -blockdev, which is the modern version of -drive..

Bl1nnnk, 2020-08-31:
I also use the method that @nola-milkin mentioned, it's useful.

il-steffen, 2021-03-11:
The separate vm_ram option was removed some time ago when testing on Windows. Do we know if the above approach also allows using multiple cores for Windows VM fuzzing? Cc @vient

vient, 2021-03-11:
Do you mean if we can make several RAM copies and use them for Windows fuzzing instead of removing `vm_ram` option? For that answer is yes, this was the first thing I tried. The reason why I went to remove `vm_ram` instead is because I don't understand it's meaning: RAM still needs to be copied as well as overlay, we might just combine them in one file then.

il-steffen, 2021-03-11:
Thanks. I just also confirmed that the Linux VM use case works with just the overlay_N file. Will update the guide accordingly.

closed 2021-03-11 by il-steffen