IntelLabs/kAFL #17: Failed to gather trace data on snapshot mode for Linux

hack-rabbit, 2020-09-09:
Hi,
First of all, I am grateful to you for your efforts to maintain this fascinating project.

I'm testing this version of kAFL with Linux snapshot(Ubuntu Server 18.04.7) and the vulnerable driver sample given at `tests/test_cases/simple/linux_x86-64`. I compiled and loaded the kernel module on the QEMU VM and could get an address range for that like below with `info` feature.
~~~
<< kAFL-Fuzzer/kafl_info.py: Agent Info Dumper >>

kAFL Linux x86-64 Kernel Addresses (41 Modules)

START-ADDRESS      END-ADDRESS		DRIVER
0xffffffffc02b0000-0xffffffffc02b4000	kafl_vuln_test
0xffffffffc02a1000-0xffffffffc02a6000	ppdev
...
~~~

However when I start fuzzing with `kafl_fuzz.py`, I keep getting "Imported payload produced no new coverage" no matter of the number of `retries` or the sample inputs. I slightly modified the code for logging and it seems that the VM keeps emitting `MSG_BUSY`. (This time I tried with 3 inputs as follows: "aa", "aaa", "abcd")
~~~
<< kAFL-Fuzzer/kafl_fuzz.py: Kernel Fuzzer >>

Importing payload from /home/user/kAFL/out/imports/seed_00002
{'type': 1, 'task': {'type': 'import', 'payload': b'aa\n'}}
Imported payload produced no new coverage, skipping..
Importing payload from /home/user/kAFL/out/imports/seed_00000
{'type': 1, 'task': {'type': 'import', 'payload': b'abcd\n'}}
Imported payload produced no new coverage, skipping..
Importing payload from /home/user/kAFL/out/imports/seed_00001
{'type': 1, 'task': {'type': 'import', 'payload': b'aaa\n'}}
Imported payload produced no new coverage, skipping..
{'type': 5}
{'type': 5}
~~~

More weird thing is that it happens again when I try with 2 inputs "SERGEJ" and "KERNELAFL" which were to crash the sample driver, except for some feeling of delayed results but no change on the `debug.log`.
~~~
<< kAFL-Fuzzer/kafl_fuzz.py: Kernel Fuzzer >>

Importing payload from /home/user/kAFL/out/imports/seed_00000
{'type': 1, 'task': {'type': 'import', 'payload': b'SERGEJ\n'}}
Imported payload produced no new coverage, skipping..
Importing payload from /home/user/kAFL/out/imports/seed_00001
{'type': 1, 'task': {'type': 'import', 'payload': b'KERNELAFL\n'}}
Imported payload produced no new coverage, skipping..
{'type': 5}
{'type': 5}
~~~

Would you please try the snapshot mode on your environment and check this happens? I am curious that this situation happens only to me or if there really exists some issue on the snapshot mode. Thanks.

+ BTW, I applied the 'sleeping' patch mentioned [here(issue #10)](https://github.com/IntelLabs/kAFL/issues/10) to bypass the race issue between the fuzzing logic and the KVM.

il-steffen, 2020-09-10:
Hi!

It looks like the agent is not executing properly. The test payloads should trigger corresponding panic hypercalls or at least a timeout in qemu.py check_recv(). Can you try the hprintf_test agent in the same folder? It should work for that same test case (kafl_vuln_test) and print some extra outputs. Also enable --debug on top of -v and check the qemu_serial* and qemu_trace* files in the workdir for errors.

The MSG_BUSY is a result of inputs not triggering feedback. The queue is empty as a result and the scheduler has no useful work to distribute. Another check should trigger within a few seconds and notify you that the bitmap is empty and probably not much useful going on.

hack-rabbit, 2020-09-10:
Thanks for your detailed answer.

I tried the `hprintf_test` agent and it seems that agent works properly. I can see the agent printing out "Injecting data...", "Injection finished..." via `hprintf` function.
~~~
__debug_hprintf: [Errno 2] No such file or directory: '/tmp/kAFL_printf.txt.7'
{'type': 1, 'task': {'type': 'import', 'payload': b'aa\n'}}
[HPRINTF]	Injecting data...
Injection finished...
[HPRINTF]	Injecting data...
Injection finished...
[HPRINTF]	Injecting data...
Injection finished...
[HPRINTF]	Injecting data...
Injection finished...
Imported payload produced no new coverage, skipping..
Importing payload from /home/user/kAFL/out/imports/seed_00000
~~~
I also enabled `--debug` on top of `-v` and checked `debug.log`. (There were no contents at all at `qemu_serial*` and `qemu_trace*`) Again the agent seems working in the loop.
~~~
0:00:00.862471] [QEMU 0]	[SEND] ENABLE_PATCHES
[0:00:00.862508] [QEMU 0]	[RECV] ENABLE_PATCHES
[0:00:00.862514] [QEMU 0]	[SEND] RELEASE
[0:00:00.864418] [QEMU 0]	[RECV] ACQUIRE
[0:00:00.864548] [QEMU 0]	Send payload..
[0:00:00.864555] [QEMU 0]	[SEND] RELEASE
[0:00:00.865360] [QEMU 0]	[RECV] ACQUIRE
[0:00:00.865422] [QEMU 0]	Send payload..
[0:00:00.865428] [QEMU 0]	[SEND] RELEASE
[0:00:00.866222] [QEMU 0]	[RECV] ACQUIRE
[0:00:00.866282] [QEMU 0]	Send payload..
[0:00:00.866288] [QEMU 0]	[SEND] RELEASE
[0:00:00.867047] [QEMU 0]	[RECV] ACQUIRE
[0:00:00.867136] [SLAVE 0]	`Imported payload produced no new coverage, skipping..
~~~
Is there anything related to errors that I should have noticed?

hack-rabbit, 2020-09-10:
I edited the `hprintf_test.c` to log the result of `write` and the system calls looks also working properly.
~~~
{'type': 1, 'task': {'type': 'import', 'payload': b'aa\n'}}
[HPRINTF]	Injecting data...
write() returned 3.
Injection finished...
...
Imported payload produced no new coverage, skipping..
Importing payload from /home/user/kAFL/out/imports/seed_00000
{'type': 1, 'task': {'type': 'import', 'payload': b'abcd\n'}}
[HPRINTF]	Injecting data...
write() returned 5.
~~~

il-steffen, 2020-09-10:
Interesting. Can you double-check that the correct IP filter range is applied during fuzzing and kaslr/meltdown fixes are disabled in the guest. Also, supplying the crashing inputs directly as seed may confuse the fuzzer but you should definitely see the crash notifications (PANIC hypercall in qemu and 'crash detected' in debug.log) With the hprintf you can check more directly if a crash actually occurs in the guest.

For serial logs you also need to add console=ttyS0 to the kernel cmdline. See BOOTPARAM in qemu.py. 

I don't think the SIGSEGV should be there, that may be something to check on. But since the agent is loading and info also shows the kmod, it seems the snapshot is correctly set up.

hack-rabbit, 2020-09-10:
I can ensure that the correct IP filter range is applied, but I might have missed about the KASLR/Meltdown fixes. Do you mean that I should turn off them manually(like editing grub boot file to remove `kaslr`) or is there any script that automatically disable KASLR/Metdown fixes?

il-steffen, 2020-09-10:
If you look at common/qemu.py there is a set of default parameters supplied via -append commandline but this only works for direct kernel boot and not with image/snapshot boot. The same should be added to your grub.cfg and then snapshot it again.

If this does not help, I suggest to try the samples in tests/user_bench/ to see if the basic setup + platform is sound. These are more automated and better tested than the VM snapshot mode.

il-steffen, 2020-09-10:
(qemu) unkown packet : 29 c0
qemu-system-x86_64: /home/user/kAFL/qemu-5.0.0/pt/decoder.c:699: decode_buffer: Assertion `false' failed.

Looks like an unrecognized packet in the PT output. Please have a look if this persists with the user_bench examples and on another platform/core if you have any. It looks like libxdc has a handler for this opcode so integrating that one might also help. I did not have the time to look at that.

il-steffen, 2020-10-15:
I noticed you are using an older 4.4 kernel that probably does not understand the mitigations=off shortcut. Try disabling the mitigations individually as explained here: https://linuxreviews.org/HOWTO_make_Linux_run_blazing_fast_(again)_on_Intel_CPUs

il-steffen, 2020-11-04:
It seems there is a change in Ice Lake and Tiger Lake that can cause this bug. I adopted it for our qemu version and I can confirm it working on Tiger Lake. Perhaps you can try it out?

[0001-adopt-PT-decoder-fix-to-enable-ICL-TGL.patch.zip](https://github.com/IntelLabs/kAFL/files/5490745/0001-adopt-PT-decoder-fix-to-enable-ICL-TGL.patch.zip)

il-steffen, 2020-11-26:
Believed to be fixed in fff59777 and upstream Nyx/libxdc.

closed 2020-11-26 by il-steffen