IntelLabs/kAFL #18: Failed to match op str on bison

lain3d, 2020-10-08:
Thanks for efforts maintaining this cool project.

Platform: Ubuntu Server 18.04 
CPU: Intel i7 8700k
QEMU: 5.0.0
Kernel: 5.8.12

I'm also including both bison_fuzz_initrd.gz and bison_info_initrd.gz as well as full debug.log.

I initially was trying to use the snapshot fuzzing but I kept getting the```got b'R', Expected: b'DZ'``` issue, even with attempting to add a delay. When I added ```synchronization_lock(cpu);``` after ```hypercall_snd_char(KAFL_PROTO_RELEASE);``` my target would hang forever.

So I took the advice from #17 and set up the user_bench bison test. I got past the qemu_handshake but it quits with the SHM/socket error. It is not an invalid packet like in #17 :

```
failed to match opstr [rdi*8 + 0x10]
qemu-system-x86_64: /home/rich/qemu-5.0.0/pt/asm_decoder.c:156: asm_decoder_parse_op: Assertion `false' failed.
```

Any idea for this?

[bison_fuzz_initrd.gz](https://github.com/IntelLabs/kAFL/files/5351381/bison_fuzz_initrd.gz)
[bison_info_initrd.gz](https://github.com/IntelLabs/kAFL/files/5351382/bison_info_initrd.gz)
[debug.log](https://github.com/IntelLabs/kAFL/files/5351386/debug.log)

lain3d, 2020-10-08:
Also supplying terminal log:
[terminal_out.txt](https://github.com/IntelLabs/kAFL/files/5351553/terminal_out.txt)

il-steffen, 2020-10-11:
Hi there,

Thanks for supplying context. Your initrd images work fine here. I tested on a i7-8665U with both 5.7.12 and 5.8.12, and I know i5-8600 and i7-9700 are working fine as well.

I noticed in your debug.log that some arcane vmlinuz-4.15 is loaded. Can you switch that back to use the same as the host? Also, the assert() is part of Redqueen code and but you have disabled redqueen in the options. Are you sure this is the matching log? Redqueen is working fine here but disabling it (or ignoring the assert violation) may be an easy fix. 

Both, this and the assert encountered in #17 look very much like an issue with the obtained binary or its decoding. But since your binary works here, the issue probably due to some mismatch in the involved setup, libraries, compiler version etc. Here are some ideas:

- Please double-check that kaslr and spectre/meltdown mitigations are disabled in the guest. You can copy-paste the qemu commandline from the log and add console=ttyS0 / -serial mon:stdio. 
- The target IP0 range seems fine, but please also check that your target is correctly detected as 64 or 32 bit arch. The SUBMIT_CR3 hypercall does an autodetection that is printed to the qemu trace log (use --debug on top of -v). The longmode boolean is used by get_capstone_mode() during disassembly. You can override the auto-recognition with KAFL_USER_SUBMIT_MODE. I believe the userspace forkserver uses that manual route but perhaps there is a conflict somewhere.. 
- Please check that the linux/qemu patches are correctly applied and perhaps try a different 
gcc version for qemu. Use the compile.sh to rebuild. I'm also attaching my 5.8 linux config just in case. [config-5.8.12-kAFL+.gz](https://github.com/IntelLabs/kAFL/files/5361089/config-5.8.12-kAFL%2B.gz)

il-steffen, 2020-11-04:
I encountered the same assertion failure after installing libcapstone 4 from sources. It is picked up by qemu: 

ldd qemu-5.0.0/x86_64-softmmu/qemu-system-x86_64 |grep cap
	libcapstone.so.4 => /usr/local/lib/libcapstone.so.4 (0x00007f76fa210000)

When I remove this one, qemu seems to statically link against its own version in qemu-5.0.0/capstone and that does not seem to cause those errors.

closed 2021-06-21 by il-steffen