IntelLabs/kAFL #19: PT support is detected by KVM

mingxiaoshan123, 2020-11-04:
[os]
Linux version 5.3.0-64-generic (buildd@lcy01-amd64-026)(gcc version 9.2.1 20191008 (Ubuntu 9.2.1-9ubuntu2))

[cpu]
Intel(R) Core(TM) i7-9700 CPU @ 3.00GHz

il-steffen, 2020-11-04:
You don't seem to be running the patched kernel. Follow the steps in `./install note` to install the kernel. On most systems this will give you a new default boot option. If it still does not work, check if the patches have been applied in the linux build tree, that the tree builds without errors, and that CONFIG_KVM_VMX_PT is enabled.

mingxiaoshan123, 2020-11-04:
> You don't seem to be running the patched kernel. Follow the steps in `./install note` to install the kernel. On most systems this will give you a new default boot option. If it still does not work, [...]

thanks for your help !!!

closed 2020-11-04 by mingxiaoshan123