IntelLabs/kAFL #21: Kernel Panic when running loader

mingxiaoshan123, 2020-11-09:
<< kAFL Usermode Load for Linux x86-64 >>
Kernel Panic Handler Address:	ffffffffb9e9b10e
segment fault

il-steffen, 2020-11-09:
Please provide some detailed context with your bug reports.
I think a common case for the loader binary to segfault is that you launched it directly on the host or in a VM without the patched Qemu or (I think) without -kafl commandline parameter to qemu. In this case it the kAFL hypercalls are not supported and so the loader will fail.

SiKreuz, 2020-11-17:
Well, I have the same problem. The VM prints these lines after running the loader:
```
<< kAFL Usermode Load for Linux x86-64 >>
Kernel Panic Handler Address: xxxxxxxxxxxxxxxx
```

The console from which I started the VM shows me these lines:
```
[QEMU-PT] Core:  Creating snapshot <kafl> ...
[QEMU-PT] Core: Done. Shutting down..
```

After about 5 minutes waiting the VM crashes and the console says `Segmentation fault (core dumped)`.
Looking at the snapshot I get this information:
```
image: overlay_0.qcow2
file format: qcow2
virtual size: 20 GiB (21474836480 bytes)
disk size: 18.3 MiB
cluster_size: 65536
backing file: ../arch/linux.qcow2
Snapshot list:
ID        TAG                     VM SIZE                DATE       VM CLOCK
1         kafl                        0 B 2020-11-17 12:20:07   00:00:19.616
Format specific information:
    compat: 1.1
    lazy refcounts: false
    refcount bits: 16
    corrupt: false
```

I'm not sure if the snapshot should only be 0B, but the rest looks fine to me. I tried to continue with running the `kafl_info.py` script with
```
python3 kAFL-Fuzzer/kafl_info.py -work_dir <work-dir> -vm_dir <snapshot-dir> -agent targets/linux_x86_64/bin/info/info -mem 1024 -v
```
and get the feedback that Qemu couldn't launch. The log tells me the following:
```
(qemu) qemu-system-x86_64: This is a disk-only snapshot. Revert to it  offline using qemu-img
```

I'm not sure whether the problem is the Kernel Panic and segment fault or there's something wrong with the snapshot.

il-steffen, 2020-11-17:
Did you follow the kAFL or the new Windows tutorial when creating the VM image? The older kAFL readme uses a separate ram disk file that was supplied as additional argument. A recent patch by @vient updated this part to use a single -vm_dir option, however that flow is only documented in the Windows tutorial.

SiKreuz, 2020-11-17:
Oh, I overlooked that I have still started the VM with the extra RAM. Well, there's still this strange termination of the VM after running the loader. But the kafl snapshot does now have a valid size and executing `kafl_info.py` works. Thank you anyway.
I had some more problems with the outdated REAMDE. I will update this later and send you a pull request.

il-steffen, 2020-11-17:
Great, thanks a lot. The qemu crash is a little sad but hopefully we can soon upgrade to use the Nyx backend with all new bugs..

closed 2022-02-14 by il-steffen