IntelLabs/kAFL #22: Debian binary

bouss83, 2020-11-16:
Hello, 
While installing redqueen on an Ubuntu 16 host, is it possible to target a binay that was compiled on a Debian os ?
I am getting some issues with the dependancies, not foung locally when generating the packed binaries (binay_info + binary_fuzz).
Any idea about this ? thank you.

il-steffen, 2020-11-16:
The examples for fuzzing userspace targets compile the target on the host and then load it into the target VM (via initrd or program/agent). But this is only a shortcut for quick install/testing. Normally your target binary is part of the VM image (or kernel or ramdisk) and you only load the agent (loader) to communicate and launch your actual target. 

Look at the VM image fuzzing flow to see how to install a custom loader/agent inside a VM image:
https://github.com/IntelLabs/kAFL/blob/master/doc/README.kAFL.md

Note that this guide is a little outdated, you may have to check with the Windows fuzzing tutorial for most recent flow of creating and loading the VM image with kAFL.

Rayram2010, 2020-11-16:
> The examples for fuzzing userspace targets compile the target on the host and then load it into the target VM (via initrd or program/agent). But this is only a shortcut for quick install/testing. [...]

Thanks for the reply. Actually I see the logic when it comes to drivers but not a given binary. Could you say how to point the target binary from the host to guest while running command "kafl_user_prepare.py" in redqueen to generate the packed binaries ? How to use the loader for this? I do not find anything in the given docs.

vient, 2020-11-16:
Take a look at tests/user_bench, kafl_user_prepare.py is used there as a part of the process in run.sh.

Rayram2010, 2020-11-16:
> Take a look at tests/user_bench, kafl_user_prepare.py is used there as a part of the process in run.sh.

Yes but this is still does recompile the target on host. I managed to do this with redqueen without problem on lava and other targets. My question is on fuzzing a binary present in the guest VM only (A Debian). I suppose that packing it (to generate binary_info and binary_fuzz) must be done on the guest since the dependencies are not present on the Ubuntu 16 host. May be the solution is not redqueen but kAFL while building a custom loader that lauches the binary in the VM before freezing it.  Will the range address ip0 of the binary appear after runing kafl_info.py in such a scenario ? What do you think?

il-steffen, 2020-11-16:
You can try to download the repo inside the guest and build the info+fuzz initramfs there, then copy them back to the host.

Note that the build scripts in tests/user_bench/ are just a way to get started and document/test the flow. If your binary is similar to the examples they might work. Otherwise you will have to customize the [ld_preload forkserver ](https://github.com/IntelLabs/kAFL/tree/master/targets/linux_x86_64-userspace/src) and/or how your application is loaded inside the initrd.

That forkserver is the only fundamental difference in the two approaches. You could implement your own agent that launches the target application inside the VM, but then it may be difficult to tell the CR3 and -ip0 range of that target process. The forkserver does that for you, and it can also restart the target directly instead of asking kAFL to reload the VM snapshot.

Rayram2010, 2020-11-17:
Thanks a lot. I will give it a try following your advice.

Rayram2010, 2020-11-18:
I packed the binaries within the VM (needed to deactivated Qemu self_chek by the way) but QEMU is required to run kafl_info.py. So, I moved the resulting binary_info outside and run the same script kafl_info.py on host but the process crashes (C return in Qemu protocol). It is stuck in the __debug_recv_expect() while loop. Do you have an idea about the cause?

il-steffen, 2020-11-18:
This is the guest crashing while trying to execute info, so the issue is still with the agent. You could try to execute the info binary manually within the guest. It is expected to crash at the very end because the hypercall at the very end of ld_preload_info.c will fail outside kAFL. But it seems you are getting a crash before that. The 'C'rash you are seeing is likely one of the panic() calls in userspace_loader.c which packages your target binary together with ld_preload_info.c.