IntelLabs/kAFL #23: Fuzzing VM: Error in debug_recv(): Got b'R', Expected: b'DZ'

SiKreuz, 2020-11-17:
When I start fuzzing with
```
./kAFL-Fuzzer/kafl_fuzz.py -work_dir <work-dir> -seed_dir <seed-dir> -vm_dir <snapshot-dir> -mem 1024 -agent targets/linux_x86_64/bin/fuzzer/kafl_vuln_test --purge -v -ip0 0xXXXXXXXXXXXXXXX-0xXXXXXXXXXXXXXXX
```
I get this error:
```
[WARNING] Slave 0: Error in debug_recv(): Got b'R', Expected: b'DZ'
[FATAL] Failed to launch Qemu, please see logs. Error: Killed Qemu due to protocol error.
[FATAL] Slave has died - check logs!
[FATAL] Master exit: All slaves have died.
```

I do not understand what this error actually means. It doesn't depend on the IP trace filter range. Everything was set up properly using the provided scripts, the `kafl_info.py` is working. The `debug.log` also does not provide more information than above.

vient, 2020-11-17:
This issue is mentioned here https://github.com/IntelLabs/kAFL/blob/master/doc/TUTORIAL.Windows_fuzzing.md#troubleshooting-1. Does it help?

SiKreuz, 2020-11-17:
I have already seen this troubleshooting at the end. Am I understanding correctly, that there is a problem with the timing? So these two lines mentioned [here](https://github.com/IntelLabs/kAFL/issues/10#issuecomment-663480222) should be pasted somewhere. But honestly I don't get where. I'm doing a linux fuzz with the loader from `targets/linux_x86_64`. Do I have to put it in the `loader.c` and re-compile and re-run it on the VM to get it working?

vient, 2020-11-17:
> Do I have to put it in the `loader.c` and re-compile and re-run it on the VM to get it working?

Yes, after HYPERCALL_KAFL_LOCK. For example, in Windows loader lines go somewhere here https://github.com/IntelLabs/kAFL/blob/master/targets/windows_x86_64/src/loader/loader.c#L172.

Better synchronization should be implemented between QEMU and fuzzer in the future, right now this hack should work.

SiKreuz, 2020-11-17:
Thank you! It's working now.
At least the error message is gone and the error doesn't appear anymore. Now I have this on my display and I'm not sure if there's something happening or just in idle...
```
Imported payload produced no new coverage, skipping..
```

vient, 2020-11-17:
Well, again, there *may be* synchronization problems that cause such message but usually it means that there is indeed no coverage registered in your provided IP range. You can try to manually add some (3-5-10) tests in `$WORK_DIR/imports` while fuzzer is running to see if new coverage will be registered after all or not.

il-steffen, 2020-11-17:
Add a few more seeds and make sure some of them land in the corpus. Inputs that do not produce new coverage are discarded. In the worst case, wrong trace settings or other mistakes can lead all inputs being discarded, and then there is nothing to mutate on.

SiKreuz, 2020-11-17:
I added some more seeds and tried more ip ranges, but it's always the same.
@vient manually adding stuff to the imports-dir did just threw the same message.
@il-steffen what exactly do you mean with landing in the corpus?

il-steffen, 2020-11-17:
Check the outputs of kafl_gui or kafl_plot to see what inputs the fuzzer actually found/recognized so far. If the seeds produced no coverage then you may end up with zero "paths" found, i.e. the corpus/queue of inputs is empty. In this case, after all seeds are processed, there is nothing for the fuzzer to do but to try random inputs. This is not done by default because usually the problem is with the PT trace setup and not with the inputs.

If you run into this problem with the published hello world sample, the error is most likely in the PT setup. Make sure that kASLR and spectre/meltdown mitigations are disabled in the guest VM. You may have to edit grub.cfg and create a new snapshot. If that seems correct, double check the ip0 range obtained from kafl_info. You can also try the hprintf agent to see if the agent is executing properly.

SiKreuz, 2020-11-18:
Ok. I disabled kASLR and spectre/meltdown mitigations in the guest VM and made a new snapshot. I double checked the ip range (I've choosen this line from `kafl_info` output: `0xffffffffc008f000-0xffffffffc0156000   ext4`) and pasted this ip range into the `kafl_fuzz` command:
```
./kAFL-Fuzzer/kafl_fuzz.py -work_dir ../VMs/working/ --purge -seed_dir ../VMs/seed/ -vm_dir ../VMs/snapshot/ -mem 1024 -ip0 0xffffffffc008f000-0xffffffffc0156000 -agent targets/linux_x86_64/bin/fuzzer/kafl_vuln_test -v
```

 I still get the same output. The `kafl_gui` output tells me that there didn't happen anything...
```
░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░[kAFL v0.2 ]░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
┃Runtime:    0m02s │ #Execs:         0 │ Exec/s:         0 │ Slaves:       1/2 ┃
┃CPU Used:      0% │ User:        0.1% │ Guest:       0.0% │ Stability:     0% ┃
┃Mem Used:    9.3% │ Avail:       6.6G │ Total:      7.7G% │ Reset/s:        0 ┃
────────────────────────────────────────────────────────────────────────────────
┃Path Info:        │ Bitmap Stats:     │ Findings:                             ┃
┃ Total:         0 │                   │  Crash:            0 (N/A)   None Yet ┃
┃ Seeds:         0 │  Edges:         0 │  AddSan:           0 (N/A)   None Yet ┃
┃ Favs:          0 │  p(col):     0.0% │  Timeout:          0 (N/A)   None Yet ┃
┃ Norm:          0 │  Pending:    0.0% │  Regular:          0 (N/A)   None Yet ┃
────────────────────────────────────────────────────────────────────────────────
┃Yld: Init:      0 │ Grim:      0 │ Redq:      0 │ Det:       0 │ Hvc:       0 ┃
┃Fav: Init:      0 │ Rq/Gr:     0 │ Det:       0 │ Hvc:       0 │ Fin:       0 ┃
┃Nrm: Init:      0 │ Rq/Gr:     0 │ Det:       0 │ Hvc:       0 │ Fin:       0 ┃
────────────────────────────────────────────────────────────────────────────────
┃> Slave  0:         import │ node:  N/A  │ fav/lvl:     N/A  │ exec/s:   N/A  ┃
────────────────────────────────────────────────────────────────────────────────
░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░[Payload Info]░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
┃Parent:   N/A │ Size:     N/A │ Bytes:    N/A │ Bits:     N/A │ Exit:         ┃
────────────────────────────────────────────────────────────────────────────────
┃0x0000000: 69 6d 70 6f 72 74 69 6e 67 2e 2e 2e              │importing...     ┃
```

il-steffen, 2020-11-18:
You say you use -ip0 range from ext4 driver, but you seem to use the kafl_vuln_test agent, not the fs_fuzzer agent. Change the -ip0 range to target the kafl_vuln_test kmod or change the agent to actually try some ext4 mounting.

SiKreuz, 2020-11-18:
Oh yes. That's true. Changed the agent to `targets/linux_x86_64/bin/fuzzer/ext4`. That should be right? But I still get the same output.
Sorry for that many questions. It's the first time I get in touch with kAFL and this type of fuzzing in general. Thank you very much for your fast answers!

il-steffen, 2020-11-18:
Try the hello world sample first to see if the basics are working. For ext4, try adding a couple ext4 headers/inputs as seeds. If the seeds get rejected in some generic wrapper you may not actually trigger ext4.kmod and the fuzzer will discard the seeds (zero paths, your problem here)

SiKreuz, 2020-11-19:
I set up the hello world sample. Needed some time but have all successfully compiled now. When I run the `qemu-system-x86_64` command from this [README](https://github.com/IntelLabs/kAFL/blob/master/targets/zephyr_x86_32/README.rst), then I get three lines with the last one telling me `Booting from ROM..` and then terminating. Is that right?
Well, if I run the `kafl_fuzz` now I once again get this error again, but slightly different:
```
[WARNING] Slave 0: Error in debug_recv(): Got b'K', Expected: b'DZ'
```
Can I somehow fix this like I did with the VM before?

il-steffen, 2020-11-19:
Are you saying the Linux samples are working now? 

The issue you see with Zephyr is probably a known a bug/regression in Zephyr. My helper scripts do not pin the Zephyr version and whatever we're checking out these days does not seem to work with qemu -enable-kvm. It does an endless reboot loop if you try it with vanilla qemu like this. Now, the default launch arguments for zephyr inside kAFL include a -catch_resets which will detect these silent reboots and report them as "ASAN" errors. The error reporting is not optimal here, but basically the messages says: "kAFL ran into KASAN type crash while trying to initiate connection with the fuzz agent.". 

The fix is to build an older zephyr that boots with KVM. I had success with `west init --mr "v2.3.0"`, perhaps you can find a more recent version...

SiKreuz, 2020-11-19:
No, the linux samples are still not working for me. But I'll first try to solve this hello word sample for now. 
Your fix did work. The zephyr os is booting properly now. I started kafl_fuzz and the gui tells me that it is running, but there's no cpu usage or executions. I'm not sure if I understand all the arguments correctly:
- How exactly read the desired ip0 range from the zephyr.map? I'm a little bit confused of this huge file.
- Which one is the correct seed directory? I tried `targets/zephyr_x86_32/seeds/`
```
python3 kAFL-Fuzzer/kafl_fuzz.py -ip0 0xXXXXXXXXXXXXXXX-0xXXXXXXXXXXXXXXX -mem 16 -extra ' -no-reboot -no-acpi' -kernel targets/zephyr_x86_32/zephyrproject/zephyr/samples/hello_world/build/zephyr/zephyr.elf -seed_dir targets/zephyr_x86_32/seeds/ -work_dir ~/fuzzing/VMs/working/ --purge -v
```

il-steffen, 2020-11-19:
All of the steps are automated in the compile.sh, including the extraction of -ip0. You probably don't get any executions because the zephyr.elf that  you launch is the standard Zephyr sample which literally just prints hello world.

closed 2020-11-19 by SiKreuz