IntelLabs/kAFL #25: Tested compatibility with Comet Lake?

dcdelia, 2020-12-18:
Hi, this issue is not about a technical difficulty but a question for well-experienced kAFL users and developers.

I heard through the grapevine that some processor models can give erratic PT data for fuzzing with kAFL or sibling projects like Redqueen. Has anyone tested if kAFL works fine with the PT facilities of recent Comet Lake processors? I'm eyeing an i9-10900K for buying a machine where I plan to use kAFL and derived projects. Or perhaps suggestions for kAFL-tested high-end i9 CPUs?

Apologies if my request may look odd or off-topic, I thought this may be the best place where one can ask :)

il-steffen, 2020-12-18:
I don't have Comet Lake at the ready but I have used it successfully on gen8, gen9, gen11. Performance also scales quite nicely with the number of cores and even (hyper-)threads.

In my experience the erratic behavior is not tied to particular CPU models but more related to some possible potential race condition and overall system load. Reducing the used cores or disabling HT had no obvious effects to me, but it may be related to overall VM image/RAM size (there are some known bugs for fuzzing full VM images which will hopefully be fixed with the upcoming Nyx release)

I typically use -funky to validate new findings a couple times before adding them to the queue. The fuzzer also collects some stats about this but they are not displayed in the gui right now (num_funky and num_reload are important to watch for performance).

dcdelia, 2020-12-19:
Thank you @il-steffen, this is tremendously helpful :-) Looking forward to Nyx!

closed 2020-12-19 by dcdelia