IntelLabs/kAFL #26: Release Builds

MatthewTingum, 2020-12-23:
First off, thanks for stepping up and maintaining this project! 

This project has made it very easy for someone interested in whole-system fuzzing to jump on board. I see that this has been ported to newer kernels which is awesome. The original project made it seem as if Ubuntu is the only OS which can support this. I seem to recall some conversation backing this. Is there any truth to this claim?

Can an iso of *some* distro be slipstreamed to include a patched kernel in addition to qemu and fuzzing utilities?

I have several projects where I'd like to integrate kAFL. This usually means recompiling the kernel with my personal patches or something a tad better like the patches found here. To meet my own needs, I was thinking of mirroring the Linux kernel and applying patches to some long term tag. Newer kAFL releases could be rebased on future Linux tags. I'm definitely over-scoping here, but this would make development a lot easier... at least for the KVM portion. It is definitely easier for the typical user to download a specific version and apply patches.

mark-stopka, 2020-12-23:
You can use openSUSE and Open Build Service with KIWI...

https://osinside.github.io/kiwi/working_with_images/build_in_buildservice.html

MatthewTingum, 2020-12-23:
Thanks will check out!

By the way, I have no intention of putting words in maintainer's mouth... `Is there any truth to this claim`. I seem to recall this being a thing.

il-steffen, 2020-12-23:
Matthew, I'm not aware of any distro-specific dependencies. The install.sh is only tested with recent Ubuntu and Debian though, so if you're going with something wildly different you will have to trace down some basic dependencies for building qemu, linux and your fuzz target of choice. The python runtime also has a fun mix of distro and pip based packages.

Any reputable distro will also have a way to create packages from a custom kernel tree. If you are lucky you can just run 'make rpm-pkg' rather than make install.

MatthewTingum, 2021-01-02:
I had hardly bothered to read the documentation on this fork when I posted this. My bad. Seems its easy enough to get this up and running on most any distro. If anyone needs some kind of quickstart I'm sure they can put together one that fits their own needs. If I run into any distro related issues I'll report back. Apologies for the "discussion" issue.

closed 2021-01-02 by MatthewTingum