IntelLabs/kAFL #30: Don't get any coverage

SiKreuz, 2021-01-20:
Hey,

I'm currenctly again trying to get the fuzzing work on an real ubuntu guest.
So I have a kAFL setup with all the stuff properly running and now I want to fuzz my target. I installed an ubuntu-server on the virtual machine and got this module I like to fuzz:
```
0xffffffffc028d000-0xffffffffc03c0000 btrfs
```
Not for any purpose, but just to get known to kAFL in general, and because you already offered an agent for btrfs. I placed some random seeds, but also some img files with btrfs on it. Well, I just get the feedback, that no new coverage was produced.
```
$ ./kAFL/kAFL-Fuzzer/kafl_fuzz.py -work_dir working/ --purge -seed_dir seed/ -vm_dir qemu/snapshot/ -mem 1024 -ip0 0xffffffffc028d000-0xffffffffc03c0000 -agent kAFL/targets/linux_x86_64/bin/fuzzer/btrfs -v

[...]
Importing payload from /home/simon/working/imports/seed_00000
Imported payload produced no new coverage, skipping..
[...]
```
May you help me a bit. What else can I try? Am I still doing something wrong with the seeds or something else? What could it be? Thanks

Host: Ubuntu 20.04.1 LTS with 5.8.12-kAFL+ kernel
Guest: Ubuntu-Server 20.04.1 LTS with default kernel

il-steffen, 2021-02-08:
Hi Simon,

This is usually a problem of ASLR or spectre/meltdown-related changes to MMU handling. Try disabling any such security/randomization features. Older 4.x kernels are known to work and I believe I saw 5.2 or .4 also working with `cmdline="nopti mitigations=off"`.

There is an open pull request for an updated Linux guide, this one also looks really good: https://github.com/IntelLabs/kAFL/issues/27#issuecomment-757012550

il-steffen, 2021-04-05:
Hi @SiKreuz - this is probably fixed in the current testing branch. Let me know if not.

Kakashiiiiy, 2021-04-12:
Hi @il-steffen ,

it seems it is not fixed right now, I do not get any coverage either. I am using the latest Arch kernel inside the VM and I have the corresponding arguments set in the cmdline.

It seems that the argument `oops=panic` (From docs/Linuxfuzz) results in a crash if we load the test driver from targets/linux_x86_64

il-steffen, 2021-04-13:
Are you sure it is the same problem? Confirm that you are executing something, e.g. feeding the crashing input should result in a panic and raise a hypercall from guest back through KVM/Qemu to the python frontend. You can also use hprintf() to try and print something via hypercalls.

Simply loading the driver should not result in a crash, unless perhaps you are trying to run them in original/unpatched Qemu/KVM.

Kakashiiiiy, 2021-04-17:
I was a bit unpercise. It does not crash it freezes, however I just booted now without oops=panic and it works until I fuzz. The seed gets imported correctly I've printet it, but as the top-issue here I get no coverage. 
```
> $ python3 kAFL-Fuzzer/kafl_fuzz.py -vm_dir snapshot/ -S kafl -agent targets/linux_x86_64/bin/fuzzer/hprintf_test -seed_dir seed/ -mem 512 -work_dir workdir -ip0 0xffffffffc05ca000-0xffffffffc05cf000  -v --purge

<< kAFL-Fuzzer/kafl_fuzz.py: Kernel Fuzzer >>

Importing payload from /home/bschlueter/University/Semester_6/Bachelor_Thesis/Fuzzer/kAFL/workdir/imports/seed_00000
b'abcdefg1234567890'
b'abcdefg1234567890'
b'abcdefg1234567890'
b'abcdefg1234567890'
Imported payload produced no new coverage, skipping..
```
The snapshot seems to work as well, at least I can execute kafl_info.py. Right now I have no idea whats wrong or where is the probleme.  Unfortunealy there are not much debugging messages  or troubleshooting guides^^.
I've also made a clean new install of linux 5.11.4-kafl and qemu but the problem is still there

```
> $ python3 kAFL-Fuzzer/kafl_info.py -vm_dir snapshot/ -S kafl -agent targets/linux_x86_64/bin/info/info -mem 512 -v -work_dir workdir | grep kafl
<< kAFL-Fuzzer/kafl_info.py: Agent Info Dumper >>
0xffffffffc05ca000-0xffffffffc05cf000	kafl_vuln_test
```

il-steffen, 2021-04-19:
@Kakashiiiiy You are printing the strings from python frontend. Compare that to whats happening inside the agent/guest. The kafl_vuln_test sample has a corresponding "hprintf" agent that should produce some extra output: [hprintf_test.c](https://github.com/IntelLabs/kAFL/blob/a7a8ae3e5ccd044ac185c94e11bc152f8c0a231c/targets/linux_x86_64/src/fuzzer/hprintf_test.c)

To check the feedback path, start with exec_res in [qemu.py](https://github.com/IntelLabs/kAFL/blob/a7a8ae3e5ccd044ac185c94e11bc152f8c0a231c/kAFL-Fuzzer/common/qemu.py). You can dump to array or get its hash. For the same input, you should get the same non-zero input.

BTW, it looks like you are at RUB working on a thesis? Perhaps contact @schumilo to work on latest upstream version.

Kakashiiiiy, 2021-04-21:
I've made a mistake compiling the Kernel Module kafl_vuln_test.ko . It works now. I compiled it outside of the VM and scp it in. Compiling it inside solved the problem for me.

vient, 2021-04-21:
Wonder why it matters

closed 2021-06-21 by il-steffen