IntelLabs/kAFL #33: Testing the testing branch

il-steffen, 2021-03-12:
Multiple proposed fixes and updates in the testing branch, including long-standing issues like Qemu hanging on snapshot. 
Also a new Linux tutorial based on #27 and #31. 

I have kind of moved on to using the Qemu-PT/KVM-PT from Nyx, but let me know how it works and we can merge any fixes.

vient, 2021-04-02:
Works fine for me. 

I have some changes on top of your testing branch which I plan to push soon, with all these changes kAFL works really stable on my setup

vient, 2021-06-21:
It seems that fixes in testing branch are generally helpful and do not cause additional problems. On the other hand, other people may not notice it and stumble to problems which are already solved in testing. I propose to merge it to master in current state before merging any existing pull request.

il-steffen, 2021-06-21:
Agreed, done.

closed 2021-06-21 by il-steffen