IntelLabs/kAFL #34: the Obtaining driver virtual address  is blank

Junjie-Fan, 2021-04-02:
Thank you for your work! I use the  kafl tool to fuzz windows kernel,.when i turn to the step "Obtaining driver virtual address" to get the address of drivers use the command ./kAFL/kAFL-Fuzzer/kafl_info.py -work_dir ./work -vm_dir . -mem 1024 -agent kAFL/targets/windows_x86_64/bin/info/info.exe. there is no output!  I do not know why it happy. my host is ubutu20.04 and my  window in qemu is window10 enterprise evaluation version 20H2

vient, 2021-04-02:
I've had problems with info just yesterday. Not sure if it will help you but I made a fix for my issue and will publish it soon (the problem is race between kAFL and QEMU on program writing/loading).

Junjie-Fan, 2021-04-02:
thank you for your work and hope to see your new publish soon!

vient, 2021-04-02:
Right, did you try 'testing' branch? It already contains some fixes that may affect you.

Junjie-Fan, 2021-04-02:
No,I haven't. I will try it these days! I'm really grateful for your analysis and your excellent advice! I hope it would be run correctly.

Junjie-Fan, 2021-04-09:
I‘m sorry to tell you that when I used the system of Windows server the kAFL works well,But when I use Win10 user system the driver address is none so that I could not take the next step.I was confused about that.Maybe there are some materials for me to learn more about it!Or maybe i should read the source code through to make it clear.Thank all your help for me till now!

Junjie-Fan, 2021-04-09:
OK，I'm  really grateful and appreciate for your advice!