IntelLabs/kAFL #36: Can't start fuzz except "Importing payload from path/to/imports/seed_00000" when starting fuzz

Junjie-Fan, 2021-04-14:
Dear Developer：when I comes to the final step,I create the seed file and make the seed "0123456789abcdef".Next I used the kafl_fuzz.py to start fuzz.But it only shows that Importing payload from path/to/imports/seed_00000.I wait a long time but there was no other happening next.and I check the import file.it's blank.Is there some steps I took wrongly?Or miss some important steps.

il-steffen, 2021-04-14:
Should be an FAQ. The seeds are copied to import/ folder upon startup and the fuzzer picks them up from there. The file will disappear after import and, if recognized as interesting input, a corresponding payload file will be created in corpus/.
There is no further output from the main program except for errors. See the main Readme on tools to watch status and inspect the files in workdir.

Junjie-Fan, 2021-04-14:
But,I also run the kafl_gui.py ,it show no any more useful information except "importing...".and i check the stasts.csv file and all the data is 0.Is this normal?I still doubt that computer did't start fuzz.Or something was wrong.

il-steffen, 2021-04-14:
Ok, that means it is indeed stuck somewhere. Try adding `-funky` and/or adding more seeds. Also make sure you are using the 'testing' branch as it fixes some bugs. You will have to build qemu again after checkout (rm -rf qemu*; ./install.sh qemu).

If none of this helps, make sure you are executing something useful. Injecting a crashing input as seed (or writing it to imports/) should result in a crash event. hprintf() and kafl_debug.py -action gdb are good friends.

Junjie-Fan, 2021-04-14:
I'm really grateful thank you for your patient and useful advice I will try the solutions!

vient, 2021-04-14:
You definitely need to try first to copy seeds to imports folder. I've observed this bug again yesterday when kAFL prints that it imported a seed but it actually did not, copying it again to imports folder helped.

il-steffen, 2021-04-14:
The message is printed from handle_import() so its pretty close to actual execution. There should be another message if the payload was deemed not interesting (no coverage or already known, or the coverage was not stable and failed to validate). Also if all seeds are rejected and nothing is in the queue, there is another error printed after a few secs.

Junjie-Fan, 2021-04-14:
I had tried this method to add the seed to imports folder before i sent this issue.But noting changed.

vient, 2021-04-14:
@il-steffen 
> If none of this happens and we are still stuck on import, that would be new. 

I did not see any additional messages appearing after "importing seed_00000" in around a minute. All slaves were stuck in waiting state except one which was stuck in input. I should note two things: this bug is not 100% reproducible and it occurred on my slightly modified version. I'll investigate it a bit later.

fr0zenrain, 2021-09-24:
i have met the same issue.
remove the qemu start argument "nographic", i can see windows guest loader.exe is hangs, but i'm not found the hangs reason.