IntelLabs/kAFL #46: Imported payload produced no new coverage, skipping

fr0zenrain, 2021-09-27:
i met this problem as above ,may be my seed wasn't add new coverage ,but i think no new coverage shouldn't reject fuzz.
some  communicator log message indicate fuzzer is ready, i was test the selffuzz_test.exe, i want start a normal fuzz test
what can i do?thanks!

il-steffen, 2021-09-28:
You need to supply seeds that trigger _some_ first coverage. Otherwise the queue is empty and no new work is scheduled.

Check if the -ip0 filter range is set properly and if you get any feedback in the returned bitmap.

You can also try to set kickstart = True in [slave.py#L90](https://github.com/IntelLabs/kAFL/blob/b5ed055ca60517b4692762b7bda9ebd6fbbcde99/kAFL-Fuzzer/fuzzer/process/slave.py#L90). This way the fuzzer will continue to try random payloads even when the queue is empty.

fr0zenrain, 2021-09-30:
thanks you reply.but i have cross new problem
the logs as follows:
Imported payload produced no new coverage,skipping..
[FATAL] Slave has died - check logs!
[FATAL] Master exit: All slaves have died.
waiting for Slave instances to shutdown...
[WARNING] Detected potential qemu zombies, please kill -9: [21017]

and when i review work_dir/debug.log,logs seems normal

[0:00:03.240795] [SLAVE 0]      `Imported payload produced no new coverage, skipping..
[0:00:03.241386] [SLAVE 0]      Launching import stage on node 0
[0:00:03.241412] [QEMU 0]       Send payload..
[0:00:03.241421] [QEMU 0]       [SEND] RELEASE
[0:00:03.308270] [QEMU 0]       [RECV] ACQUIRE
[0:00:03.308378] [QEMU 0]       Send payload..
[0:00:03.308389] [QEMU 0]       [SEND] RELEASE
[0:00:03.373198] [QEMU 0]       [RECV] ACQUIRE
[0:00:03.373310] [QEMU 0]       Send payload..
[0:00:03.373321] [QEMU 0]       [SEND] RELEASE
[0:00:03.439564] [QEMU 0]       [RECV] ACQUIRE
[0:00:03.439676] [QEMU 0]       Send payload..
[0:00:03.439687] [QEMU 0]       [SEND] RELEASE
[0:00:03.503953] [QEMU 0]       [RECV] ACQUIRE
[0:00:03.504054] [SLAVE 0]      `Imported payload produced no new coverage, skipping..
[0:00:03.504847] [SLAVE 0]      Launching import stage on node 0

when i debug the python code, sever.wait exception have nothing error information.i can't get the failed reason.
thanks!

fr0zenrain, 2021-10-14:
With in-depth analysis，i found loader.exe CreateProcess failed.because program buffer was initial data 0xff...
it seems write_virtual_memory failed, because cpu_get_phys_page_attrs_debug return -1.

some logs

[QEMU-PT] Iface: setup filter file: /dev/shm/kafl_tfilter
[QEMU-PT] Iface: filter file size: 1000000 (addr: 0x7fc205200000)
[QEMU-PT] Iface: new shm file/data/kAFL/work/program: (max size: 8000000) 8000000
[QEMU-PT] Iface: ptr=0x7fc1b0000000, mem dword=300905a4d
[QEMU-PT] Iface: new shm file/dev/shm/kafl_work_qemu_payload_0: (max size: 20000) 20000
[QEMU-PT] Iface: ptr=0x7fc21a665000, mem dword=0
[QEMU-PT] Core:  Enable hprintf support
[QEMU-PT] Core:  Patching PANIC address:        fffff8006020a1a0, longmode=1
[QEMU-PT] Mem:   phys_addr == -1:       fffff8006020a1a0
[QEMU-PT] Core:  Patching PANIC address:        fffff8006020a1c0, longmode=1
[QEMU-PT] Mem:   phys_addr == -1:       fffff8006020a1c0
[QEMU-PT] Mem:   phys_addr == -1:       ac0000
[QEMU-PT] Core:  Got program address:   ac0000 ret=0