IntelLabs/kAFL #47: macOS Virtual Machine creation

r-ld, 2021-10-14:
Hello,

I'm looking for information about the creation of macOS Virtual Machine.

With the help of the kholia/OSX-KVM project I've manage to create VM on macOS Big Sur or Monterey beta.

In order to create a snapshot I've removed the several options (CPU: invtsc ...)that were making the VM non-migratable.

I'm 'able' to create the snapshot but there are several issues with this approach:

    I cant use ram and overlay files, only overlay file is used for snapshot storing.

    On qemu-2.9.0 the savevm command is not stopping, I need to use the qemu-img info overlay.qcow2 command and kill qemu when the snapshot is shown with that command. (the problem is not present with qemu-4.2.0 btw)

    Before using the savevm command I need to remove the OpenCoreBoot, drive used by the kholia/OSX-KVM project, to prevent qemu for writing the snapshot in it.

My snapshot seems to be valid, as I'm able to load it, but I guess there is one problem:

    when I'm trying to send the vuln_test agent to crash the vulnerable_driver I'm always getting the phys_addr == -1 errors in read_virtual_memory and write_virtual_memory in memory_access.c.

    when I'm sending the info agent the error is not reproduced as I'm able to get the several informations written to the info_buffer.

    I have the same issue between the several tested macOS version (HighSierra and BigSur)

So it's possible to read/write at virtual address in userland but it's not possible when the address is in kernelland ?

I bet the way I'm creating the VM, and the snapshot, is not good (or incomplete):

    Does anyone have any idea why I'm encountering this problem ?
    Does anyone could share a better way to create the macOS VM, and snapshot ?

Thank you

fr0zenrain, 2021-10-20:
your issue looks like mine #46. it's fail to convert virtual address to physical, my system is windows 10.

r-ld, 2021-11-16:
Does anyone know how to disable KPTI/KAISER/doublemap() protection on macOS ?

fr0zenrain, 2021-11-24:
in loadr.c i find "ensure the virtual memory is really present in physical memory".this is the address transfer fail reason.
so how can i ensure that?