IntelLabs/kAFL #50: steps to make kAFL work

irenge, 2021-10-31:
Are there any steps to follow to make kAFL works  on Ubuntu?
I follow the below steps and it s not working

il-steffen, 2021-10-31:
Can you provide the next level of detail beyond "not working". Recent Ubuntu 20.4/20.10 should be no problem.

irenge, 2021-11-01:
When i launch this command 
 python3 ./kAFL-Fuzzer/kafl_info.py  --purge -work_dir $WORK -vm_dir snapshot/ -mem 512 -agent targets/linux_x86_64/bin/info/info
It complains
[FATAL] Failed to launch Qemu, please see logs. Error: [Errno 104] Connection reset by peer

il-steffen, 2021-11-01:
Fatal / connection reset is the result of Qemu exit. Qemu complains about your snapshot and likely exits immediately after launch. 

I have not seen the snapshot error. Can you check that the steps for creating the snapshot worked and you supply the correct image? Check that there is a snapshot kalled 'kafl' listed in the image, you can query the file properties with qemu-img.

closed 2021-11-02 by irenge