IntelLabs/kAFL #51: Windows guest VM with NIC attached doesn't complete qemu_handshake.

guardnox, 2021-12-13:
I created Windows 11 VM with "-net nic" option and I edited "kAFL-Fuzzer/common/qemu.py" by adding "-net nic" instead of "-net none". When I execute kafl_info.py the VM just got stuck on the qemu handshake.

il-steffen, 2021-12-13:
Stuck on qemu handshake usually means it is waiting for the guest to send the proper handshake. Guest may not be booting properly or the loader does not start.

guardnox, 2021-12-13:
I tried to remove **self.__debug_send(qemu_protocol.RELEASE)** from qemu.py and it's seems to work.
I'm not fully understand what going on there but I think qemu listening for RELEASE code but the guest doesn't send the hypercall or the guest sends the hypercall but the qemu not catch it somehow.

il-steffen, 2021-12-13:
Probably fine as long as its running. This protocol stuff has been replaced in the new Nyx backend and is much more reliable. I plan to push an update soon, latest January.

guardnox, 2021-12-13:
I didn't run kafl_fuzz yet but kafl_info is working now. 
I'm trying to install Npcap driver in the guest vm in order to inject packets to the NIC, which the harness will get from the fuzzer. 
Is it sounds suitable for kAFL to you ?

il-steffen, 2021-12-13:
Injecting "to the NIC" sounds like fuzzing "downwards" towards NIC emulation and external interface. Not sure if that makes sense from threat model and all. Injecting traffic upwards, from the network interface to protocol drivers or network services would make more sense. It can be hard for a fuzzer to generate a non-trivial message exchange but perhaps you capture some samples for use as seeds.

guardnox, 2021-12-13:
I'm actually inject traffic into the network interface and the driver catch the packet like it would came from external network. as you said it's more difficult to create non-trivial messages, thus I'm wondering whether kAFL is the proper way to do that.

il-steffen, 2021-12-13:
I would expect that the general network interface has been tested extensively, and I don't know how much driver-specific functions you can trigger from general pcap injection.

guardnox, 2021-12-13:
You probably right as far as Windows OS is concerned, but if I'll do that for embedded tcpip stacks I guess it will be more potentially exploited.

closed 2021-12-13 by il-steffen