IntelLabs/kAFL #53: Update Windows target and others

fuzzah, 2022-03-10:
Hi. I wish to contribute.
What needs to be done in order to migrate [targets](https://github.com/IntelLabs/kafl.targets) from kafl_user.h to nyx_api.h? I'm most interested in the Windows target.
I guess I would also need to read some additional info on the topic (kAFL, QEMU, snapshot fuzzing etc.). Any recommendations for that?

il-steffen, 2022-03-10:
Hi, welcome!

For an overview on the tool, I think the original kAFL paper and more recent Nyx paper are probably the best intro.
https://nyx-fuzz.com/papers/ => kAFL Feedback Fuzzer and Nyx Greybox Hypervisor Fuzzer.

If you ignore the Hypervisor and structure-aware fuzzer part in the Nyx paper, it basically shows an extension of kAFL with fast snapshots (and other capabilities merged in the code base as well).

Migrating to the new interface is not difficult. Here is my change for the Zephyr sample: https://github.com/IntelLabs/kafl.targets/commit/6cece87ae876e8dbd48c24edd85837e521d8953d
- include new nyx_api.h
- query host_config and set agent_config as part of fuzzer handshake.
- start fuzzing loop

The config query/set hypercalls are now required. The various new options are not tested at all, I basically just set it up for PT tracing + validate the payload buffer size expected by fuzzer frontend.

il-steffen, 2022-05-24:
Someone did the hard work, would be great to cleanup and integrate: https://github.com/Kharos102/kafl.targets

taehyun9203, 2023-02-05:
Hi, I do not know it is good timing to ask but I want to need help.

I tried to windows example in https://github.com/Kharos102/kafl.targets but failed.

I did following process to execute fuzzing.

I made one windows10 that installed vuln driver and compiled the vuln_test.c.

I enrolled vuln_test.exe (compiled by vuln_test.c) service when O/S was boot.

then converted the vmdk (I made the windows10 O/S in vmware) into qcow2.img

Then boot ubuntu again with 5.10.73-kafl+ start fufzzing.

I gave below command line.
`/home/test/kAFL# kafl fuzz --image windows10.qcow2 -m 16802 --redqueen --grimoire -D --radamsa --purge --log-crashes -p 2 --seed-dir=input/inputs --append ""`

The result was shown below.

```
No PT trace region defined.
00:00:00:     0 exec/s,    0 edges,  0% favs pending, findings: <0, 0, 0>
Worker-00 Launching virtual machine...
/home/test/kAFL/kafl/qemu/x86_64-softmmu/qemu-system-x86_64
	-enable-kvm
	-machine kAFL64-v1
	-cpu kAFL64-Hypervisor-v1,+vmx
	-no-reboot
	-net none
	-display none
	-chardev socket,server,id=nyx_socket,path=/dev/shm/kafl_test/interface_0
	-device nyx,chardev=nyx_socket,workdir=/dev/shm/kafl_test,worker_id=0,bitmap_size=65536,input_buffer_size=131072
	-device isa-serial,chardev=kafl_serial
	-chardev file,id=kafl_serial,mux=on,path=/dev/shm/kafl_test/serial_00.log
	-m 16802
	-drive file=windows10.qcow2
	-fast_vm_reload path=/dev/shm/kafl_test/snapshot/,load=off
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.07H:EBX.hle [bit 4]
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.07H:EBX.rtm [bit 11]
[QEMU-NYX] Warning: Invalid sharedir...
[QEMU-NYX] Booting VM to start fuzzing...
Worker-01 Launching virtual machine...
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.07H:EBX.hle [bit 4]
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.07H:EBX.rtm [bit 11]
[QEMU-NYX] Warning: Invalid sharedir...
[QEMU-NYX] Waiting for snapshot to start fuzzing...
```

Would you give me advice how to solve this problem?

Wenzel, 2023-06-12:
Hello @Kharos102, i'm trying to integrate the work you did on fixing the Windows targets for kAFL.

I have a WIP branch here:
https://github.com/Wenzel/kafl.targets/tree/fix_windows

I'm stuck at trying to find the right environment and `cl.exe` command line to compile the [`Driver.c`](https://github.com/Wenzel/kafl.targets/blob/fix_windows/windows_x86_64/src/vuln_driver/Driver.c)

I'm working with a Windows 10, Visual studio 2017 SDK and MSVC compiler is installed + WDK (10.0.17763.0)
Basically what i'm doing right now:

```python
C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\BuildTools\\VC\\Auxiliary\\Build\\vcvars64.bat
# append env vars with WDK Include and Lib dirs
# include: C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.17763.0\\km
# lib: C:\\Program Files (x86)\\Windows Kits\\10\\Lib\\10.0.17763.0\\km\\x64
cl Driver.c <cl_options>
```

I know it's been a long time, but if you recall how you approached it, it would help.
Thanks !

Kharos102, 2023-06-12:
Hey!

Pretty sure I just create a new project in Visual Studio based on the empty WDM project template, then replace the main .c code with the Driver.c code, don't usually compile it via cl directly.

Wenzel, 2023-06-14:
Hey @Kharos102

I managed to compile it with `cl.exe`, but I will update that later on with a real VS project and use `msbuild`, that's more convenient

Thanks !

Wenzel, 2023-06-28:
A PR has been that should fix this issue:
https://github.com/IntelLabs/kafl.targets/pull/21

Wenzel, 2023-07-19:
Closed with https://github.com/IntelLabs/kAFL/releases/tag/v0.7

closed 2023-07-19 by Wenzel