IntelLabs/kAFL #54: Malformed mainifest or wrong version of west?

docfate111, 2022-03-18:
Following the setup guide I did
```
git clone $this_repo ~/kafl; cd ~/kafl
make env       # create and activate environment
west update -k # download required sub-components
```
then I get the error
```
FATAL ERROR: Malformed manifest file: /home/t/fuzzing/kafl/manifest/west.yml 
  Schema file: /home/t/.local/lib/python3.8/site-packages/west/manifest-schema.yml
  Hint: /home/t/fuzzing/kafl/kafl: "self: import: .submanifests": file /home/t/fuzzing/kafl/kafl/.submanifests not found
```
Do I need to change the version of west or pip or python or something?
I have not made any edits to the west.yml. It is the same one as in the repo.

docfate111, 2022-03-18:
How do I change  /home/t/fuzzing/kafl/kafl/.submanifests to  /home/t/fuzzing/kafl/.submanifests?

docfate111, 2022-03-18:
Hi. its ok.
Is there a line missing to create scripts/create_env.sh in the Makefile?
```
$ git checkout -b b7ddae2
$ make env
make: *** No rule to make target 'scripts/create_env.sh', needed by '.env'.  Stop.
```

il-steffen, 2022-03-18:
`git checkout -b stable b7ddae2`; then follow old readme.

docfate111, 2022-03-18:
sorry to keep asking but is this not right?
```
 west init --mr stable  -m https://github.com/IntelLabs/kAFL.git
```
If I do 
```
west update -k
```
I get 
```
File "/home/t/.local/share/virtualenvs/kafl-UUNwUV7H/lib/python3.8/site-packages/west/manifest.py", line 182, in _mpath
    raise MalformedConfig('no "manifest.path" config option is set') from e
west.manifest.MalformedConfig: no "manifest.path" config option is set
```

il-steffen, 2022-03-18:
The issue should be fixed as of 433fda9. Please try `pipenv --rm`;  delete the workdir and try again.

docfate111, 2022-03-18:
Ok I tried 
```
git checkout -b stable 433fda9
```
the new README.md mentions make env as the first step but there is no target env in the Makfile. Also 
```
west update host_kernel
```
does not work after 
```
west init
```

il-steffen, 2022-03-18:
no, the readme says to checkout workdir branch as first step.

docfate111, 2022-03-19:
never mind it works

closed 2022-03-19 by docfate111