IntelLabs/kAFL #59: intelpt trace feature is not working properly.

insu123, 2022-04-11:
Thank you for awesome project.

I tested kafl, version kafl_0.2, on Ubuntu 20.04 x64 with fuzzing Windows 10 x64 kernel.
I want to fuzzing windows default kernel driver.
But target driver is not traced properly.
So I debugged windows qemu with gdb and target driver breakpoint hit properly.
But ntoskrnl.exe is traced well.
I think this problem occurs because of intelpt.
Can you give me an advice for me which code process this problem?

Thank you.

il-steffen, 2022-04-11:
Can you try the master branch? There have been many fixes since 0.2 and it should not be difficult to get the Windows example working again.

"Target not traced properly" is a bit too vague. If you have an example case working, the difference to tracing other targets is usually just to set the proper code range and CR3 filter. If you get decoding errors, try to snapshot/restore closer to target code and narrow the trace region to just the specific code sections you're interested in.

insu123, 2022-04-12:
Thank you for your reply.

1. I also tried the master branch. But starting windows in qemu with flags, -machine kAFL64-1, result int infinite boot. So I changed machine flag to q35 and booted well. But after starting fuzzing, when HYPERCALL_KAFL_ACQUIRE hypercall invoked, the windows occurs BSOD. so I tested kafl_0.2. 
1. I want to fuzzing windows network driver which is parsing packet. But this driver is not parsing packet using DeviceIoControl. So I made harness using raw socket and sending packet to localhost and driver parsed this packet well. I enabled SAMPLE_DECODED_DETAILED flag in logger.h. In /tmp/traces/sample_detailed_X files, There is no TNT packet at all.

il-steffen, 2022-04-12:
1) Can you try to play around with the devices emulated by qemu? It may be worth trying to switch the drive emulation between scsi and virtio, deactivate network, disable any fancy paravirtualization etc. There may also be a bug in virtio-blk interface that causes an assert on first qemu snapshot restore. But not a BSOD.

2) No trace likely means the filter is configured wrong. Try some simple example to ensure code range is correct, and disable CR3 filter until you have found the right setting.

insu123, 2022-04-12:
Oh I set CR3 filter wrong.
Thank you.

insu123, 2022-04-13:
I found the cr3 value which trace well.
But `kAFL_hypercall(HYPERCALL_KAFL_SUBMIT_CR3, 0);` submits wrong cr3 value.
How can I set cr3 value which i want to filter?

il-steffen, 2022-04-13:
Yes, I was also thinking of that.

Is rcx=0 the right magic value? I guess we don't need a disable signal, but some funny OS could actually set CR3 to zero? The only case I've seen is when building Zephyr with MMU=disable, in which case the CR3 filter does not matter.

insu123, 2022-04-14:
Thank you.
But if I set code range and not set CR3 filter, some code is traced well but windows go hang and trace go infinite loop.
Do you know why windows go hang?

schumilo, 2022-04-14:
Trace decoding without enabled CR3 as well as IP filters is not supported or even tested. I guess that this is also the reason why the decoding results in an infinite loop.

closed 2022-04-19 by insu123