IntelLabs/kAFL #61: Root Snapshot seems not work with Pre-Snapshot

xiolee4433, 2022-05-18:
Thank you for sharing great project!

I tried to test kafl sample driver fuzzing with pre-snapshot on Ubuntu 20.04 x64 host, Windows 11 x64 guest. 
Only thing I changed was inserting hypercall `kAFL_hypercall(HYPERCALL_KAFL_LOCK, 0);` as first instruction in the `vuln_test.c `. 
It did work well. Pre-image was dumped correctly, CR3 value was correct, handshake result was fine, but couldn't get coverage information. I thought this was a pre-image bug, so I tried the save/load snapshot function of qemu, but the result was same. 
Maybe Root Image does not store device state correctly?

xiolee4433, 2022-05-18:
Oh, Mutated input data was also fine.

il-steffen, 2022-05-18:
I have not worked with the pre-snapshot for some time. The KAFL_LOCK hypercall is deprecated since Nyx release, though not fully removed. The INFO hypercall used in the example is also deprecated and broken.

If you can confirm that the guest is resuming and executing correctly, the issue may be just to configure PT feedback correctly. Please double check the IP filter ranges. You can set them directly from the guest using RANGE_SUBMIT hypercall:
https://github.com/IntelLabs/kafl.linux/commit/13bbdb48084af2930185c6f86d298a62fb41cb39

Also ensure that you have disabled things like ALSR + KPTI in the guest and that the target actually returns (UI is reporting >1 exec/s).

I recently added a new hello world: https://github.com/IntelLabs/kafl.targets/tree/master/linux-kernel

schumilo, 2022-05-18:
For the record: KAFL_LOCK hypercall ***is not deprecated*** and creates a pre-snapshot if required (otherwise it does nothing, so no need to worry). But KAFL_INFO is indeed deprecated and shouldn't be used anymore.

il-steffen, 2022-05-18:
Ah, good to know the LOCK hypercall is still with us..

@xiolee4433: I just realized you want to target Windows 11, not Ubuntu. Have you checked this guide+patches: https://github.com/Kharos102/kafl.targets

The basic setup is the same - make env and make install should work fine. But you need some changes in the target itself to make it work with Nyx. See the changes in above repo.

xiolee4433, 2022-05-19:
Yes, I checked KAFL_INFO hypercall is deprecated. @il-steffen And Yes, I cheked it too, so it worked without pre-image. 
The only case I struggle now is with pre-image :(

xiolee4433, 2022-05-22:
I was wrong. After calling KAFL_LOCK, cpu state changed. cr3 value is not the same after loading pre-image snapshot

xiolee4433, 2022-05-22:
I don't know which process KVM judged as the calling guest process.

xiolee4433, 2022-05-23:
I solved this by changing cr3 value which is taken by my kernel driver.
Pre-image, Root-image was not a problem. 
Seems KVM's sending CR3 value was a problem.

il-steffen, 2022-05-24:
Sounds like you found the issue? The CR3 should be set exactly based on whatever the guest value is at time of the hypercall. There is a patch to make this more flexible, feel free to try and give feedback: https://github.com/nyx-fuzz/QEMU-Nyx/commit/88f2165541cb70bf432bc1a8580983502c21d4a6

closed 2022-06-29 by il-steffen