IntelLabs/kAFL #62: Breaking change on master branch

il-steffen, 2022-05-24:
### Heads up on planned re-organization of the installation flow + git repo structure

The current "workspace" installation approach where we separate the overall setup from python fuzzer and other components seems to work well. Folks can fork the "workspace" and pin some particular versions or add more repos for their desired fuzzing pipeline.
Problems are
- workspace as a branch is confusing, people tend to just click "Clone" on the main kAFL page
- setup with custom Makefile/install.sh is fragile and does not scale well for automated deployment 
- the kAFL-Fuzzer is now just one of the components to deploy, and will be better to maintain as such
- Docker [kind of works for automating kAFL deployment](https://github.com/Wenzel/kAFL/tree/docker_workspace) but we also tend to require a special kernel or other system-level properties

So going forward (next few days) **we plan to:**
- move kAFL-Fuzzer in the current master branch to a separate repo (kafl.fuzzer, to keep with the fantastic naming scheme)
- turn the 'kAFL' project into the top-level repo with scripts + docs to deploy and use kAFL/Nyx stack
- replace install steps + scripts with Ansible playbooks for local and remote installation ([feature branch](https://github.com/Wenzel/kAFL/tree/workspace_deployment))

Obviously this will **break history for any ongoing checkouts or forks**. The result of Ansible setup will be quite similar to current `make install` steps, but it may have some gaps initially. We can keep a branch "v0.4" of the current kAFL master for anyone too busy to switch their stuff to the new install.

I don't expect a lot of users being impacted here, but wanted to give a heads-up + reasoning anyway.

Let me know if you have any input. Ansible certainly feels like pulling in yet another big piece of infrastructure, but in my experience so far it has been very lightweight, basically just requiring sudo + ssh. It comes with a rich ecosystem to automate the system-level deployment tasks we need and it can also replace West, which has not been as unproblematic as I'd hoped.

vient, 2022-05-25:
Offtop: wonder if github gives any way to know stats of repo pulls (branches)?

il-steffen, 2022-05-25:
There is an "Insights" tab that claims some 115 "unique" clone ops over last 2 weeks. But I'm guessing 90% of that is just us testing Qemu versions and Ansible.

il-steffen, 2022-06-13:
Migration complete, old version is tagged as v0.4.

closed 2022-06-13 by il-steffen