IntelLabs/kAFL #82: Document system-wide changes made by Ansible

Wenzel, 2022-06-30:
As raised in https://github.com/IntelLabs/kAFL/pull/80#discussion_r910831103, we need to document the system-wide changes made by Ansible, so users can revert their system back into their original state if desired.

Since Ansible is a configuration management system, maybe it can output a summary of the system modifications at the end of the run ?

il-steffen, 2022-06-30:
We could use the dry-run/check mode to generate a useful overview up front. It currently fails as we use variables/outputs from prior jobs that aren't executed properly this way.

But the normal logging is probably just the standard output with different verbosity options. Either too high-level or too ansible-specific. There is a --diff option but I expected to be similar cryptic.

This looks more accessible but might again just give the titles of jobs. 
https://github.com/haidaraM/ansible-playbook-grapher

Maybe good enough to give an idea of whats gonna happen, plus logs.

Wenzel, 2022-09-19:
Partially solved by #103 
Future Sphinx docs PR will solve this

Wenzel, 2022-11-17:
closed by #122

closed 2022-11-17 by Wenzel