IntelLabs/kAFL #92: Read timeout error with make deploy

KainaatSingh, 2022-08-10:
The following error is observed when installing requirements.txt during deploy. Pip has a default timeout of 15 seconds that leads to this error when the internet connection is slow. I can recommend increasing the default timeout that helps overcome this issue.

[image]

Wenzel, 2022-08-10:
Thanks for reporting the issue.

I would avoid tweaking pip's behavior unless it's absolutely necessary.
kAFL's installer can't take responsability for the end-user connection hazards, and modifying pip's default values as a consequence.

If you are using a VPN, maybe you should double check the connection speed and reliability.
You are likely to experiment connection hiccups with other package managers, not just pip.

@il-steffen any thoughts ?

il-steffen, 2022-08-10:
Yes, seems like someone else's problem. The error handling is terrible but apparently also not our fault.

closed 2022-09-02 by il-steffen