IntelLabs/kAFL #116: ansible: run apt update first..

il-steffen, 2022-10-20:
When not updated for some time, apt tasks will try and fail to download outdated packets:

    The following additional packages will be installed:
      libasm1 libgit2-1.1 libhttp-parser2.9 libmbedcrypto3 libmbedtls12
      libmbedx509-0 libstd-rust-1.48 libstd-rust-dev rustc
    Suggested packages:
      cargo-doc rust-doc rust-src
    Recommended packages:
      cargo rust-gdb | rust-lldb
    The following NEW packages will be installed:
      cargo elfutils libasm1 libgit2-1.1 libhttp-parser2.9 libmbedcrypto3
      libmbedtls12 libmbedx509-0 libstd-rust-1.48 libstd-rust-dev rustc
    0 upgraded, 11 newly installed, 0 to remove and 0 not upgraded.
    Need to get 56.5 MB of archives.
    After this operation, 272 MB of additional disk space will be used.
    Err:1 http://ftp.de.debian.org/debian bullseye/main amd64 libhttp-parser2.9 amd64 2.9.4-4
      404  Not Found [IP: 141.76.2.4 80]

Wenzel, 2022-10-20:
Updating the APT cache is already a `pre_tasks` in [`site.yml`](https://github.com/IntelLabs/kAFL/blob/master/deploy/site.yml#L6).

So i'm confused as to what happened here ?
Did you run make deploy with a particular tag ?

il-steffen, 2022-10-20:
The error happened on the CCC repo, so I guess we don't have this pre-task there?

Wenzel, 2022-10-20:
No we don't, indeed.
I can make a PR.
https://github.com/intel/ccc-linux-guest-hardening/blob/master/deploy/site.yml#L6

Wenzel, 2022-10-20:
@il-steffen I don't have enough rights on ccc repo to transfer the issue there.
Can you try to do it ?
it's just on the right sidebar:
[image]

il-steffen, 2022-10-20:
"You can only transfer issues between repositories owned by the same user or organization account."

Wenzel, 2022-10-20:
ok, so let's close this.
See https://github.com/intel/ccc-linux-guest-hardening/pull/40