IntelLabs/kAFL #119: kafl_0.2 document error

5angjun, 2022-11-07:
Hi, I really interested in this kAFL fuzzing project.
so i trying to run kAFL until *[*kafl_0.2**  branch](https://github.com/IntelLabs/kAFL/tree/kafl_0.2)

i think there is some problem in kAFL/doc/TUTORIAL.Linux_fuzzing.md
--> in Option #2: Build from Filesystem (Automated) 
--> at the booting kernel /etc/rc.local script cannot command `modprobe` and `insmod` because they are not installed 
--> therefore, the kafl_vuln_test driver cannot be loaded

this is the result
```
[  OK  ] Started Daily Cleanup of Temporary Directories.
[  OK  ] Reached target Timers.
         Starting OpenBSD Secure Shell server...
         Starting LSB: Set the CPU Frequency Scaling governor to "ondemand"...
[  OK  ] Started Permit User Sessions.
[    1.017201] rc.local[238]: kernel.randomize_va_space = 0
[    1.017636] rc.local[238]: /etc/rc.local: 3: /etc/rc.local: modprobe: not found
[    1.018034] rc.local[238]: /etc/rc.local: 4: /etc/rc.local: modprobe: not found
[    1.018388] rc.local[238]: /etc/rc.local: 5: /etc/rc.local: insmod: not found
[  OK  ] Started LSB: Set the CPU Frequency Scaling governor to "ondemand".
[  OK  ] Started getty on tty2-tty6 if dbus and logind are not available.
[  OK  ] Started OpenBSD Secure Shell server.
[QEMU-PT] Core:  Creating snapshot <kafl> ...
[QEMU-PT] Core:  Done. Shutting down..
root@sangjunLab:~/kafl# WORK=~/work; mkdir -p $WORK
root@sangjunLab:~/kafl# python3 ./kAFL-Fuzzer/kafl_info.py --purge -vm_dir $TARGET -mem 512 -work_di 
    __                        __   ___    ________
   / /_____  _________  ___  / /  /   |  / ____/ /
  / //_/ _ \/ ___/ __ \/ _ \/ /  / /| | / /_  / /
 / ,< /  __/ /  / / / /  __/ /  / ___ |/ __/ / /___
/_/|_|\___/_/  /_/ /_/\___/_/  /_/  |_/_/   /_____/
=================================================== (C) 2020

<< ./kAFL-Fuzzer/kafl_info.py: Agent Info Dumper >>

kAFL Linux x86-64 Kernel Addresses (0 Modules)

START-ADDRESS      END-ADDRESS		DRIVER
0xffffffffc0060000-0xffffffffc006b000	autofs4
0xffffffffc003a000-0xffffffffc005f000	psmouse
0xffffffffc0027000-0xffffffffc0031000	ahci
0xffffffffc001d000-0xffffffffc0026000	libahci
0xffffffffc000f000-0xffffffffc0017000	i2c_i801
0xffffffffc0009000-0xffffffffc000e000	i2c_smbus
0xffffffffc0002000-0xffffffffc0008000	lpc_ich
0xffffffff81000000-0xffffffff81e50d80	Kernel Core

root@sangjunLab:~/kafl# 

```

I'm audit this project line by line so really want to run this tool :point_up_2: 
i really appreciate it if you could fix this problem.

my detailed environment setting is
```
root@sangjunLab:~/kafl# uname -a
Linux sangjunLab 5.11.4-kAFL+ #1 SMP Mon Nov 7 21:58:09 KST 2022 x86_64 x86_64 x86_64 GNU/Linux
root@sangjunLab:~/kafl# cat /etc/issue
Ubuntu 20.04.4 LTS \n \l
```
and i command for setting this.
https://www.notion.so/realhacker/TEST-kafl_0-2-07234f84ab164821afce42ce71149ffe

il-steffen, 2022-11-07:
Hi there,

The v0.2 branch was made for archive purpose before rebasing to the Qemu-Nyx release. Many more things have changed since then, and we cannot support it.

I recommend to take the latest master branch and work based on that. The userspace example is unfortunately also not updated, but you can try the linux-kernel or Zephyer examples to get started.

5angjun, 2022-11-07:
hmm. i want to use kAFL legacy or kafl_0.2.

there is any way to run original kAFL??

i planned to study qemu_nyx after studying original kAFL.

if there is any way to run that?

thanks

il-steffen, 2022-11-07:
You can try the old branch, the linux userspace examples should work better there than in the latest version (since we did not update them yet).

The example you show above shows that the basic tools are working. It appears that modprobe is just not installed. Maybe you have to add it explicitly in the debootstrap image creation? Something like this?
```
sudo debootstrap --arch=amd64 --variant=minbase --include vim,ssh,kmod [...]
```

5angjun, 2022-11-07:
ohhh i works. very interesting!!

Thank you for help me. 

Now i'm going to audit kAFL original to master branch :+1: 

Thank you so much for your help

closed 2022-11-07 by 5angjun