IntelLabs/kAFL #121: [Question] Can latest kAFL works on target with no source code?

jingfelix, 2022-11-13:
Hi! 

I am working on fuzzing Linux kernel using kAFL. Some third-party distributions use modified kernel and may not publish their source code, so I am wondering if the latest version of kAFL can fuzz targets without source code. 

Also I notice that a new example named "linux-user" is working in progress. Hope to see tutorial about it!

Thanks again for creating and maintaining such a great tool.

il-steffen, 2022-11-13:
Hi,

I cannot tell you what you can fuzz or not, but kAFL has been used for Windows and OSX fuzzing so in general this should be possible.

Updated linux userspace tutorial is planned, yes.

jingfelix, 2022-11-15:
Thanks! I will consider proceeding my work after the userspace tutorial is ready.

closed 2022-11-15 by jingfelix

Wenzel, 2022-11-15:
Thanks @il-steffen for answering this question !
@jingfelix feedback from the community is extremely valuable to us, so thank you for sharing your problem and questions with us.

I just opened the [Github Discussions tab](https://github.com/IntelLabs/kAFL/discussions) to better engage and connect with our community, get their questions regarding the project or some of its components, their specific use cases, etc ..

Feel free to use it !