IntelLabs/kAFL #141: broken memory lookup for Q35 and >3.2GB RAM

il-steffen, 2022-12-13:
When running Q35 setup with >3.2GB ram, the hypercall GET_PAYLOAD() will error out with
```
qemu-system-x86_64: /home/user/kAFL/kafl/qemu/nyx/snapshot/memory/block_list.c:52: snapshot_page_blocklist_add: Assertion `phys_addr < self->phys_area_size' failed.
```

The assert is here: https://github.com/nyx-fuzz/QEMU-Nyx/blob/8a88edc2a1b06cd7b8b0d7954061cbebe9e01f65/nyx/memory_access.h#L33

Reason is probably lack of support for larger VM size for Q35 platform. Sergej hinted that we need to fix the memory access functions based on this here: https://github.com/nyx-fuzz/QEMU-Nyx/blob/1acaa75a8b093e3d1c23667782783bea36ec16e3/hw/i386/pc_q35.c#L139

il-steffen, 2022-12-13:
Cc @x86-sec

schumilo, 2023-01-02:
WIP patch: https://github.com/nyx-fuzz/QEMU-Nyx/pull/47