IntelLabs/kAFL #145: Any documents to show how to use "kafl cov"?

LeoneChen, 2022-12-23:


il-steffen, 2022-12-30:
Our linux kernel example covers it. I recently updated the README.md for the new commandline and added some more usage notes around it:
https://github.com/il-steffen/kafl.targets/tree/master/linux-kernel#4-next-steps

TODO: Should merge this updated text to our docs/tutorial. The README can just link to the tutorial.

LeoneChen, 2022-12-30:
Thanks~

And I found a ghidra script that can show coverage result, which may be helpful for others. (Descibed in [here](https://github.com/IntelLabs/kAFL.fuzzer#coverage-and-debug))

```
make env
./kafl/fuzzer/scripts/ghidra_run.sh $WORKDIR $TARGET_BINARY kafl/fuzzer/scripts/ghidra_cov_analysis.py
```

il-steffen, 2022-12-30:
TODO: merge updated text in [kafl.targets/linux-kernel/README.md](https://github.com/IntelLabs/kafl.targets/tree/master/linux-kernel#4-next-steps)  to our docs/tutorial. The README can just link to the tutorial.

See also: #126