IntelLabs/kAFL #150: kafl_fuzz.py path

5angjun, 2023-01-13:
Hello.

I stuck this problem.

There is no kafl_fuzz.py
[image]

`
cd ~
git clone https://github.com/IntelLabs/kAFL.git
cd kAFL

make deploy
# update-grub and change kernel

make env

cd ~/kAFL/kafl/fuzzer/

there is no kafl_fuzz.py
`

Wenzel, 2023-01-13:
Hi @5angjun !

Thanks for reporting this issue.
Indeed `kafl_fuzz.py` is missing since the kafl command line has been refactored to act as a single `kafl` entrypoint with subcommands, like `git` for example.

You can find this in the next kafl.fuzzer changelog, unpublished yet:
https://github.com/IntelLabs/kafl.fuzzer/blob/master/.github/RELEASE.md

 At the same time were are updating the tutorial with this new command line, the PR hasn't been merged yet:
https://github.com/IntelLabs/kAFL/pull/135

So in short, you should now switch from `kafl_fuzz.py` to `kafl fuzz`.

Furthermore, a Docker image is coming soon for kafl, if you prefer that option in the future:
https://github.com/IntelLabs/kAFL/pull/136

If you have more topics or concerns you would like to start engaging with, you can start a discussion here !
https://github.com/IntelLabs/kAFL/discussions

Thanks

5angjun, 2023-01-13:
oh thanks

but is there any seed file to run fuzz followed tutorial??

i command using this commands.

i need some advice

```
sudo su
## Dependency ##
apt-get update -y
apt install curl git wget file zsh sudo vim libssl-dev libffi-dev build-essential libssl-dev libc6-i386 libc6-dbg gcc-multilib make gcc gdb -y
apt install python git curl wget vim zsh gdb python3 python3-pip  -y
apt install gawk bison flex openssl libssl-dev libelf-dev lz4 dwarves -y
apt install qemu-utils qemu-system-x86 python3 python3-venv g++-mingw-w64-x86-64 zstd -y
apt install python3 python3-venv -y
sudo apt install gawk bison flex openssl libssl-dev libelf-dev lz4 dwarves zstd -y

## git clone ##
cd ~
git clone https://github.com/IntelLabs/kAFL.git
cd kAFL

## Install kAFL ##
make deploy
make env

## Download Target kernel ##
mkdir ~/kAFL/linux-kernel && cd ~/kAFL/linux-kernel
git clone -b kafl/fuzz-5.15-4 https://github.com/IntelLabs/kafl.linux.git linux-guest
cd linux-guest

## change .config to custom config  And Build##
wget https://raw.githubusercontent.com/IntelLabs/kafl.targets/master/linux-kernel/config.vanilla.virtio
mv ./config.vanilla.virtio .config
make -j$(nproc)

## set KAFL CONFIG ##
cd ~/kAFL/kafl/fuzzer/
wget https://raw.githubusercontent.com/IntelLabs/kafl.targets/master/linux-kernel/kafl_config.yaml
export KAFL_CONFIG_FILE=kafl_config.yaml

## output Dir ##
mkdir output

## FUZZ ##
kafl fuzz --purge \
-w /dev/shm/kafl \
--redqueen \
--grimoire \
-D --radamsa \
--kernel ~/kAFL/linux-kernel/linux-guest/arch/x86/boot/bzImage \
-t 0.1 \
-ts 0.01 \
-m 512 \
--log-crashes \
-p 12 \
--work-dir ./output

```
[image]

il-steffen, 2023-01-16:
The setup script above looks good, screenshot also looks like kAFL is installed correctly (it launches and manages to boot the kernel, you can verify by looking at Qemu logs in $KAFL_WORKDIR/{qemu,serial,hprintf}*log).

The relevant error message is "ABORT: Attempt to finish kAFL run but never initialized". This is an 'abort' hypercall issued by the harness inside the kernel source. The error is triggered when we reach the end of a fuzz execution loop without consuming any fuzzing input (meaning we can stop fuzzing because the configuration is broken).

It looks like you supplied the correct kernel config and kafl config, but something is wrong. The kernel .config will activate the virtio injection, but the qemu commandline shown in your screenshot does not have any virtio devices enabled. The kafl_config.yaml you downloaded enables a `virtio-net` device and also sets `hprintf` parameter in the `qemu --append` option. Both are missing in your screenshot, so something is wrong how you supplied this config.

Wenzel, 2023-01-16:
Thanks @il-steffen for your inputs.

To better understand what went wrong while loading the config, you can use the `--verbose` kafl switch to turn on verbose logging.

This will dump important information regarding which configuration files were loaded, and what the whole config looks like when the fuzzer starts.

closed 2023-01-24 by 5angjun