IntelLabs/kAFL #172: pipe target output to hcat show nothing

LeoneChen, 2023-02-21:
Hello! In order to show some information in work_dir, `hprintf` in source code can always output string, however, if I use `./target 2>&1 | vmcall hcat` in shell, usually I can't get anything, but sometimes it show someting

il-steffen, 2023-02-27:
Did you find any specific bugs? 

This is usually an issue with buffering in the pipe or ./target writing to other destinations like kernel log and not directly to console.

The most reliable option is to use hprintf() directly in your agent.

LeoneChen, 2023-02-27:
Sorry, I can't figure out it, it may be my own problem in using it, I will use hprintf() directly. If I figure out it someday, I will note it here. Thanks for your help!