IntelLabs/kAFL #173: deploy check for pmu_name fails for alderlake cpus

fuzzah, 2023-03-02:
Here's my result of running `make deploy`:
```
TASK [intellabs.kafl.kernel : Checking that pmu_name is not broadwell] *********
fatal: [localhost]: FAILED! => changed=false 
  cmd:
  - grep
  - --invert-match
  - broadwell
  - /sys/devices/cpu/caps/pmu_name
  delta: '0:00:00.001088'
  end: '2023-03-02 11:22:22.906029'
  failed_when_result: true
  msg: non-zero return code
  rc: 2
  start: '2023-03-02 11:22:22.904941'
  stderr: 'grep: /sys/devices/cpu/caps/pmu_name: No such file or directory'
  stderr_lines: <omitted>
  stdout: ''
  stdout_lines: <omitted>

PLAY RECAP *********************************************************************
localhost                  : ok=3    changed=0    unreachable=0    failed=1    skipped=0    rescued=0    ignored=0   

make[1]: *** [Makefile:36: deploy] Error 2
make[1]: Leaving directory '/home/user/git/kAFL/deploy'
make: *** [Makefile:27: deploy] Error 2
```

In fact there are two paths:
/sys/devices/cpu_atom/caps/pmu_name
/sys/devices/cpu_core/caps/pmu_name
Both contain "alderlake_hybrid". The CPU is Core i9-13900K

Temporary workaround: delete respective check in the file `deploy/intellabs/kafl/roles/kernel/tasks/pre_tasks.yml`.

Wenzel, 2023-03-02:
hi @fuzzah 
thanks for reporting this issue, it's much appreciated.
According to LKML:
https://lore.kernel.org/lkml/YE+avkajxbGJMOoM@krava/T/

> AlderLake uses a hybrid architecture utilizing Golden Cove cores
> (core cpu) and Gracemont cores (atom cpu). Each cpu has dedicated
> event list. Some events are available on core cpu, some events
> are available on atom cpu and some events can be available on both.
> 
> Kernel exports new pmus "cpu_core" and "cpu_atom" through sysfs:
> /sys/devices/cpu_core
> /sys/devices/cpu_atom

If both exposes the `pmu_name`, i will simply add one of those paths as an additional check, if the file exists.

fuzzah, 2023-03-02:
@Wenzel , your fix works.

And here are additional issues which took me so long to deal with and respond back to you.

## Downgrading kernel may break drivers
After running `make deploy` on Ubuntu **22.04** the 5.19 kernel was replaced by the 5.10-kafl kernel, which permanently disabled all networking on that machine after reboot :(

I quickly gave up on finding solutions to that, because no similar version of `linux-modules-extra` for the 5.10 kernel is provided in 22.04 repos. Decided to install Ubuntu **20.04**.

## Problems with ansible installation on Ubuntu 20.04
On updated Ubuntu **20.04** `make deploy` failed, because pip couldn't install ansible==7.1.0. Turns out the default python on 20.04 is 3.8, but ansible requires 3.9. Installing python 3.9 alone didn't help, had to resort to this solution of symlinking apt libs in python's dist-packages:
```bash
cd /usr/lib/python3/dist-packages
sudo ln -s apt_inst.cpython-38-x86_64-linux-gnu.so apt_inst.so
sudo ln -s apt_pkg.cpython-38-x86_64-linux-gnu.so apt_pkg.so
```
source: https://stackoverflow.com/a/69107017

fuzzah, 2023-03-02:
Upd: on Ubuntu 20.04 the kernel was downgraded too, which also disabled all networking :(

Also no linux-modules-extra of version 5.10 available in apt.
Any remedy to that apart from compiling drivers?

closed 2023-03-03 by Wenzel

fuzzah, 2023-03-03:
Did some homework.
The network card is Intel i225-v. In order for Linux kernel to support it, the kernel should be configured with the `CONFIG_IGC=M` option as described here: https://www.kernelconfig.io/config_igc?q=&kernelversion=5.10.73&arch=x86
Seems like newer kernels enable this by default.

Going to compile 5.10.73-kafl on my own.
@Wenzel, could you please advise in what file should I put the option?

Wenzel, 2023-03-03:
Hi @fuzzah 

> And here are additional issues which took me so long to deal with and respond back to you.

First, thank you so much for the energy you put in to figure out how to install kAFL and sending that feedback to us.
One of my goals is to tear down the entry barrier, and reduce the friction as much as possible to increase kAFL adoption by its users and the community in general.

Those issues were not in my line of sight, but thanks to you, we can have a deeper look into this.

> After running make deploy on Ubuntu 22.04 the 5.19 kernel was replaced by the 5.10-kafl kernel, which permanently disabled all networking on that machine after reboot :(

I have to review the way we build and release our kernel debian package, and will try to use the Ubuntu 22.04 generic configuration as a default from now on.

> For some reason make still asks configuration questions in terminal. Any way to make it non-interactive?

did you run the following commands ?
~~~
make olddefconfig
./scripts/config --enable XXX
~~~

`make olddefconfig` should sets every option to their default value, if any.
that should give you a non interactive build.

> On updated Ubuntu 20.04 make deploy failed, because pip couldn't install ansible==7.1.0. Turns out the default python on 20.04 is 3.8, but ansible requires 3.9. Installing python 3.9 alone didn't help, had to resort to this solution of symlinking apt libs in python's dist-packages:

I will create a separate issue to track this down. 👍

PR #174 (2023-03-03): kernel: fix checking pmu_name for AlderLake
fix #173