IntelLabs/kAFL #175: Problems with make deploy and Python 3.9 requirement on ubuntu 20.04

Wenzel, 2023-03-03:
From https://github.com/IntelLabs/kAFL/issues/173#issuecomment-1452063651

# Problems with ansible installation on Ubuntu 20.04
On updated Ubuntu **20.04** `make deploy` failed, because pip couldn't install ansible==7.1.0. Turns out the default python on 20.04 is 3.8, but ansible requires 3.9. Installing python 3.9 alone didn't help, had to resort to this solution of symlinking apt libs in python's dist-packages:
```bash
cd /usr/lib/python3/dist-packages
sudo ln -s apt_inst.cpython-38-x86_64-linux-gnu.so apt_inst.so
sudo ln -s apt_pkg.cpython-38-x86_64-linux-gnu.so apt_pkg.so
```
source: https://stackoverflow.com/a/69107017

➡️ Next step: try this in a docker container

Wenzel, 2023-03-03:
With this Dockerfile, I can `make deploy` kAFL userland components in a Docker container based on Ubuntu 20.04.

The only thing I really had to do was to create an alternative for `python3`, and it wasn't created by Ubuntu after installing `python3.9` as I expected.
~~~Dockerfile
FROM ubuntu:20.04

ARG DEBIAN_FRONTEND=noninteractive
# we need python3-venv otherwise when creating the venv, python will screams that "ensurepip" is not available.
RUN apt-get update && apt-get install -y python3.9 python3.9-venv python3-venv git build-essential
# no python3 alternative has been created, we need to build one ourselves and link it to python3.9 (!)
RUN update-alternatives --install /usr/local/bin/python3 python $(which python3.9) 50
RUN git clone https://github.com/IntelLabs/kAFL
WORKDIR kAFL
RUN make deploy -- --skip-tags kernel
~~~

fuzzah, 2023-03-06:
@Wenzel your instructions work!
Seems like it was my fault using update-alternatives in a different way and also installing `python-is-python3`.

Wenzel, 2023-03-06:
Thanks for the confirmation !
I will think again about rolling back to Python3.8 and Ansible 6.7.0, as installing Python3.9 doesn't create or update the alternative and isn't transparent for the end-user on Ubuntu 20.04.

closed 2023-03-06 by Wenzel