IntelLabs/kAFL #183: kafl.qemu Crashes When Installing Windows

Anhang512, 2023-04-14:
Hello,
Qemu got stuck while choosing where to install the windows system. At the same time, the host got stuck. I had to reboot with the hardware power button.
The branch of kafl.qemu I use is `kafl_stable`. The command for creating a VM is as follows:
```
qemu-img create -f qcow2 windows.qcow2 50G
./kAFL/kafl/qemu/x86_64-softmmu/qemu-system-x86_64 -machine q35 -enable-kvm -m 1024 -hda ./windows.qcow2 -cdrom  ./windows.iso
```

Wenzel, 2023-04-14:
Hi @Anhang512 and thanks for reporting this issue.

Can you give us more context here ?
Are you following the Windows example from [kafl.targets](https://github.com/IntelLabs/kafl.targets/tree/master/windows_x86_64) ? This example code is outdated and needs to be updated.
A ticket is already tracking this issue here: https://github.com/IntelLabs/kAFL/issues/53

Can you give us the `dmesg` log of the last boot to see if there was any stacktrace originating in KVM ?

This command should give you the previous boot's log
```shell
journalctl -b -1 -e
```

Anhang512, 2023-04-14:
Hi @Wenzel , Thank you for your help.

I noticed the problem with [kafl.targets](https://github.com/IntelLabs/kafl.targets/tree/master/windows_x86_64) and also looked at https://github.com/IntelLabs/kAFL/issues/53. These are for setting the target environment.
But before setting target virtual machine, I first need to create a target virtual machine.

Following the tutorial [kAFL stable-0.5](https://github.com/IntelLabs/kAFL/blob/stable-v0.5/docs/source/how_to/windows_tutorial.md), I used kafl.qemu to create a windows virtual machine and crashed in the process.

The result of command  `journalctl -b -1 -e` is:
4月 14 17:29:52 tianyi-pc kernel: nouveau 0000:01:00.0: gr: GPC0/TPC0/MP trap: global 00000004 [MULTIPLE_WARP_ERRORS] warp 40009 [ILLEGAL_INSTR_ENCODING]
4月 14 17:29:52 tianyi-pc kernel: nouveau 0000:01:00.0: gr: TRAP ch 2 [007faf1000 systemd-logind[803]]
4月 14 17:29:52 tianyi-pc kernel: nouveau 0000:01:00.0: gr: GPC0/TPC0/MP trap: global 00000004 [MULTIPLE_WARP_ERRORS] warp 50009 [ILLEGAL_INSTR_ENCODING]
4月 14 17:29:52 tianyi-pc kernel: nouveau 0000:01:00.0: gr: TRAP ch 2 [007faf1000 systemd-logind[803]]
4月 14 17:29:52 tianyi-pc kernel: nouveau 0000:01:00.0: gr: GPC0/TPC0/MP trap: global 00000000 [] warp 3e0009 [ILLEGAL_INSTR_ENCODING]
4月 14 17:29:52 tianyi-pc kernel: nouveau 0000:01:00.0: fifo: fault 00 [READ] at 000000fcd233c000 engine 00 [GR] client 0d [GPC0/GCC] reason 00 [PDE] on channel 6 [007f79d000 Xorg[1050]]
4月 14 17:29:52 tianyi-pc kernel: nouveau 0000:01:00.0: fifo: channel 6: killed
4月 14 17:29:52 tianyi-pc kernel: nouveau 0000:01:00.0: fifo: runlist 0: scheduled for recovery
4月 14 17:29:52 tianyi-pc kernel: nouveau 0000:01:00.0: fifo: engine 0: scheduled for recovery
4月 14 17:29:52 tianyi-pc kernel: nouveau 0000:01:00.0: Xorg[1050]: channel 6 killed!
4月 14 17:30:01 tianyi-pc CRON[10418]: pam_unix(cron:session): session opened for user root by (uid=0)
4月 14 17:30:01 tianyi-pc CRON[10419]: (root) CMD ([ -x /etc/init.d/anacron ] && if [ ! -d /run/systemd/system ]; then /usr/sbin/invoke-rc.d anacron start >/dev/null; fi)
4月 14 17:30:01 tianyi-pc CRON[10418]: pam_unix(cron:session): session closed for user root
4月 14 17:33:14 tianyi-pc kernel: INFO: task kworker/u16:3:9211 blocked for more than 120 seconds.
4月 14 17:33:14 tianyi-pc kernel:       Not tainted 5.10.73-kafl+ #1
4月 14 17:33:14 tianyi-pc kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
4月 14 17:33:14 tianyi-pc kernel: task:kworker/u16:3   state:D stack:    0 pid: 9211 ppid:     2 flags:0x00004000
4月 14 17:33:14 tianyi-pc kernel: Workqueue: events_unbound nv50_disp_atomic_commit_work [nouveau]
4月 14 17:33:14 tianyi-pc kernel: Call Trace:
4月 14 17:33:14 tianyi-pc kernel:  __schedule+0x1d4/0x630
4月 14 17:33:14 tianyi-pc kernel:  schedule+0x4a/0xb0
4月 14 17:33:14 tianyi-pc kernel:  schedule_timeout+0xb8/0xf0
4月 14 17:33:14 tianyi-pc kernel:  ? nouveau_fence_enable_signaling+0x2c/0x70 [nouveau]
4月 14 17:33:14 tianyi-pc kernel:  ? __dma_fence_enable_signaling+0x36/0x60
4月 14 17:33:14 tianyi-pc kernel:  dma_fence_default_wait+0x197/0x1e0
4月 14 17:33:14 tianyi-pc kernel:  ? dma_fence_release+0xf0/0xf0
4月 14 17:33:14 tianyi-pc kernel:  dma_fence_wait_timeout+0x42/0x60
4月 14 17:33:14 tianyi-pc kernel:  drm_atomic_helper_wait_for_fences+0x7f/0xf0 [drm_kms_helper]
4月 14 17:33:14 tianyi-pc kernel:  nv50_disp_atomic_commit_tail+0x85/0x840 [nouveau]
4月 14 17:33:14 tianyi-pc kernel:  ? __switch_to+0x131/0x3c0
4月 14 17:33:14 tianyi-pc kernel:  nv50_disp_atomic_commit_work+0xd/0x10 [nouveau]
4月 14 17:33:14 tianyi-pc kernel:  process_one_work+0x1c4/0x320
4月 14 17:33:14 tianyi-pc kernel:  worker_thread+0x4e/0x400
4月 14 17:33:14 tianyi-pc kernel:  kthread+0x125/0x140
4月 14 17:33:14 tianyi-pc kernel:  ? process_one_work+0x320/0x320
4月 14 17:33:14 tianyi-pc kernel:  ? __kthread_bind_mask+0x70/0x70
4月 14 17:33:14 tianyi-pc kernel:  ret_from_fork+0x22/0x30

Wenzel, 2023-04-14:
So, first good news: the kernel bug isn't located in kAFL code, nor in KVM.

That stacktrace originates from the [`nouveau`](https://nouveau.freedesktop.org/) open-source Nvidia graphic driver
`tianyi-pc kernel:  ? nouveau_fence_enable_signaling+0x2c/0x70 [nouveau]`

I suspect you have a very recent Nvidia card, with an edge case not supported by nouveau in our 5.10.73 kernel.

Fortunately, we already have a [kAFL enabled 6.0 kernel](https://github.com/IntelLabs/kAFL/pull/178) available.

Can you pick up this [`linux-image-6.0.0-nyx+_6.0.0-nyx+-1_amd64.deb`](https://github.com/IntelLabs/kafl.linux/releases/download/kvm-nyx-6.0-v1-alpha/linux-image-6.0.0-nyx+_6.0.0-nyx+-1_amd64.deb) kernel package, install it and try again ?

Anhang512, 2023-04-17:
Thank you for your help, the problem was solved by replacing the experimental computer with an AMD graphics card, but was not solved by the new kernel.

kafl.qemu will crash and the host user session will restart with the new kernel. Fortunately, the host won't crash. The message of `dmesg` after restart of session is:
...
```
[ 1342.389028] nouveau 0000:01:00.0: gr: TRAP ch 2 [007faf1000 systemd-logind[822]]
[ 1342.389035] nouveau 0000:01:00.0: gr: GPC0/TPC0/MP trap: global 00000004 [MULTIPLE_WARP_ERRORS] warp 10009 [ILLEGAL_INSTR_ENCODING]
[ 1342.389044] nouveau 0000:01:00.0: gr: TRAP ch 2 [007faf1000 systemd-logind[822]]
[ 1342.389050] nouveau 0000:01:00.0: gr: GPC0/TPC0/MP trap: global 00000004 [MULTIPLE_WARP_ERRORS] warp d0009 [ILLEGAL_INSTR_ENCODING]
[ 1342.391127] nouveau 0000:01:00.0: fifo: fault 00 [READ] at 0000004b776f1000 engine 00 [GR] client 0d [GPC0/GCC] reason 00 [PDE] on channel 2 [007faf1000 systemd-logind[822]]
[ 1342.391136] nouveau 0000:01:00.0: fifo: channel 2: killed
[ 1342.391138] nouveau 0000:01:00.0: fifo: runlist 0: scheduled for recovery
[ 1342.391142] nouveau 0000:01:00.0: fifo: engine 0: scheduled for recovery
[ 1342.391149] nouveau 0000:01:00.0: systemd-logind[822]: channel 2 killed!
```

closed 2023-04-17 by Anhang512