IntelLabs/kAFL #190: does "ERROR: KVM does not support NYX_PT (0)!" influence the kafl

gjm-anban, 2023-04-26:
I ran the command "make deploy", and an error occurred. However, it continued to run the next step. Does this error affect Kafl's work?
[image]

Wenzel, 2023-04-26:
Hi @gjm-anban and thanks for opening an issue.

This error results from testing whether the current kernel support the NYX API or not.
It's used as a conditional inside the playbook to determine whether a Nyx kernel should be downloaded and installed.

The only to build this conditional was to run the test, ignore the result, and catch the exit code in a variable to be tested afterwards.
https://github.com/IntelLabs/kAFL/blob/master/deploy/intellabs/kafl/roles/kernel/tasks/main.yml#L34

Everything is working as expected, but I understand it might seem confusing 🙂 

If you have ideas to improve this, please open an issue !

gjm-anban, 2023-04-26:
> Are you still running this on top of VMware? That will not work. Please give output of `lscpu` or `grep intel_pt /proc/cpuinfo`...

I try in another machine
[image]

il-steffen, 2023-04-26:
Looks like [IceLake](https://www.intel.com/content/www/us/en/products/sku/215277) and Linux is detecting the Intel PT feature. Should work fine.

closed 2023-05-04 by gjm-anban