IntelLabs/kAFL #191: how to use the payload

gjm-anban, 2023-05-04:
I fuzz the linux by https://intellabs.github.io/kAFL/tutorials/fuzzing_linux_kernel.html.
I found some payload here
[image]
Could anyone tell how to use the payload to reproduce the crash. I didn't find tutorials.

Wenzel, 2023-05-04:
Hi @gjm-anban ,

You can have a look at the `kafl debug` subcommand to replay a single payload.

The command line is similar to `kafl fuzz`, except that it takes an `input` file parameter, which is the payload.

I agree that documentation is missing on how to proceed with a fuzzing campaign, i'll add a note to do it this month !

Thanks for the feedback !

gjm-anban, 2023-05-05:
thanks, I can run the payload now
`KAFL_CONFIG_FILE=./kafl_config.yaml kafl debug --kernel linux-guest/arch/x86/boot/bzImage -m 512 --input /dev/shm/kafl_root/corpus/crash/payload_00125 --action gdb --resume`

liujf628995, 2023-09-08:
hello，@Wenzel.
I tried using kafl debug to test Payload, but the program would always stop waiting for the snapshot to start.I tried to copy the qemu command and execute it in bash,it also had not any output.Here is the kafl debug log,I don't know how to run it correctly.

kafl debug --resume   --kernel /home/liu/linux-guest/arch/x86/boot/bzImage --memory 2048  --input /dev/shm/kafl_root/corpus/kasan/payload_00001 --action gdb

No trace region configured! Intel PT disabled!
Starting Qemu + GDB with payload /dev/shm/kafl_root/corpus/kasan/payload_00001
Connect with gdb to release guest from reset (localhost:1234)
Worker-1337 Launching virtual machine...
/home/liu/kAFL/kafl/qemu/x86_64-softmmu/qemu-system-x86_64
	-enable-kvm
	-machine kAFL64-v1
	-cpu kAFL64-Hypervisor-v1,+vmx
	-no-reboot
	-net none
	-display none
	-chardev socket,server,id=nyx_socket,path=/dev/shm/kafl_root/interface_1337
	-device nyx,chardev=nyx_socket,workdir=/dev/shm/kafl_root,worker_id=1337,bitmap_size=65536,input_buffer_size=131072
	-device isa-serial,chardev=kafl_serial
	-chardev file,id=kafl_serial,mux=on,path=/dev/shm/kafl_root/serial_1337.log
	-m 2048
	-s
	-S
	-kernel /home/liu/linux-guest/arch/x86/boot/bzImage
	-append root=/dev/vda1 rw hprintf=4 nokaslr oops=panic nopti mitigations=off
	-netdev user,id=mynet0
	-device virtio-net,netdev=mynet0
	-fast_vm_reload path=/dev/shm/kafl_root/snapshot/,load=on
[QEMU-NYX] Max Dirty Ring Size -> 1048576 (Entries: 65536)
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.07H:EBX.hle [bit 4]
qemu-system-x86_64: warning: host doesn't support requested feature: CPUID.07H:EBX.rtm [bit 11]
[QEMU-NYX] Dirty ring mmap region located at 0x7fbdedd9c000
[QEMU-NYX] Warning: Invalid sharedir...
[QEMU-NYX] Waiting for snapshot to start fuzzing...

> Hi @gjm-anban ,
>
> You can have a look at the `kafl debug` subcommand to replay a single payload.
>
> The command line is similar to `kafl fuzz`, except that it takes an `input` file parameter, which is [...]

Wenzel, 2023-09-08:
Hi @liujf628995,

When using the GDB kafl debug action, QEMU starts with `-S`:
~~~
-S              freeze CPU at startup (use 'c' to start execution)
~~~

so you need to connect to the QEMU instance with gdb and release the execution

~~~
gdb
(gdb) target remote :1234
(gdb) continue
~~~

liujf628995, 2023-09-08:
> Hi @liujf628995,
> 
> When using the GDB kafl debug action, QEMU starts with `-S`:
> 
> ```
> -S              freeze CPU at startup (use 'c' to start execution)
> ```
> 
> so you need to connect to the QEMU instance with gdb and release the execution
> 
> ```
> gdb
> (gdb) target remote :1234
> (gdb) continue
> ```

@Wenzel Thanks for your help!