IntelLabs/kAFL #199: does it only run on physical host machine and cannot start in virtual machines like on vmware？

liujf628995, 2023-05-16:
My computer is a Win10 machine,I tried to make kafl on linux by vmwarebut could not find intel_pt in /proc/cpuinfo.Then I found the intel_pt info on WSL in the same host machine,I thought that means my cpu is support for intel_pt.
Can this project only run on physical Linux machines?

Wenzel, 2023-05-17:
Hi @liujf628995 

kAFL is based on a modified KVM, so it requires a Linux host machine, bare-metal.
Nesting kAFL inside another VM is an open ticket at this point: https://github.com/IntelLabs/kAFL/issues/142

liujf628995, 2023-05-18:
> Hi @liujf628995
> 
> kAFL is based on a modified KVM, so it requires a Linux host machine, bare-metal. Nesting kAFL inside another VM is an open ticket at this point: #142

Thanks for your anwser! I think it will be helpful that writing the operational requirement into README.I have tried many times to locate this problem.

Wenzel, 2023-06-20:
> I think it will be helpful that writing the operational requirement into README.I have tried many times to locate this problem.

The hardware and software requirements are specified in the documentation:
I think it will be helpful that writing the operational requirement into README.I have tried many times to locate this problem.

And there is a note specifying that VM-based setups are not supported:
[image]

closed 2023-06-23 by Wenzel