IntelLabs/kAFL #200: Can kAFL run in host mode without qemu?

LeoneChen, 2023-05-30:
Hello~ Can kAFL run in host mode without qemu?

Wenzel, 2023-06-04:
Hi @LeoneChen 

I'm not exactly sure what you mean by "host mode", but kAFL is tracing the execution of a QEMU/KVM VM, and not the host machine.

LeoneChen, 2023-06-04:
Hi!

In fuzzing user space target.

Since Qemu-Nyx dosen't support some special hardware feature (e.g. Intel SGX, and integrated Qemu-SGX and Qemu-Nyx seems very troublesome), I think a fuzz engine that can run in host (non VM) and utilize intel PT to collect coverage is helpful, much better then instrumented coverage feedback, (snapshot of VM is not necessary to me)

But maybe seems not possible in current software stack of kAFL. And seems no any other fuzz engines can utilize Intel PT to collect coverage.

Wenzel, 2023-06-23:
Tracing the host is not what kAFL intends to do, only tracing VMs running on KVM.
You can ask the fuzzing community if such a tool exist ?

Closing this issue as it's unrelated to kAFL.

closed 2023-06-23 by Wenzel