IntelLabs/kAFL #204: [UEFI] ERR: TNT 483 at position <0x0fabcf17,0x0fabcf17>

francesco-ev, 2023-06-19:
Hello, I was trying to fuzz UEFI using [this](https://github.com/IntelLabs/kafl.targets/tree/master/uefi_ovmf_64) by running the provided run.sh script (`./run.sh dxe_null` and then `./run.sh fuzz`) and initially I was getting this error:
```bash
Worker-00 Failed to connect to Qemu: [Errno 2] No such file or directory: '/dev/shm/kafl_uefi/aux_buffer_0'
```

<details>
  <summary>Full output</summary>

```bash
(.venv) francesco@xps:~/kAFL/kafl/examples/uefi_ovmf_64$ ./run.sh fuzz

    __                        __  ___    ________
   / /_____  _________  ___  / / /   |  / ____/ /
  / //_/ _ \/ ___/ __ \/ _ \/ / / /| | / /_  / /
 / ,< /  __/ /  / / / /  __/ / / ___ |/ __/ / /___
/_/|_|\___/_/  /_/ /_/\___/_/ /_/  |_/_/   /_____/
===================================================

<< kAFL Fuzzer >>

Warning: Launching without --seed-dir?
Warning: Requested 8 workers but 0 out of 8 vCPUs seem busy?
00:00:00:     0 exec/s,    0 edges,  0% favs pending, findings: <0, 0, 0>
Worker-00 Launching virtual machine...
/home/francesco/kAFL/kafl/qemu/x86_64-softmmu/qemu-system-x86_64
	-enable-kvm
	-machine kAFL64-v1
	-cpu kAFL64-Hypervisor-v1,+vmx
	-no-reboot
	-net none
	-display none
	-chardev socket,server,id=nyx_socket,path=/dev/shm/kafl_uefi/interface_0
	-device nyx,chardev=nyx_socket,workdir=/dev/shm/kafl_uefi,worker_id=0,bitmap_size=65536,input_buffer_size=131072,ip0_a=0x2000000,ip0_b=0x2f00000,ip1_a=0xf000000,ip1_b=0xff00000
	-device isa-serial,chardev=kafl_serial
	-chardev file,id=kafl_serial,mux=on,path=/dev/shm/kafl_uefi/serial_00.log
	-m 256
	-bios /home/francesco/kAFL/kafl/examples/uefi_ovmf_64/bios.bin
	-append nokaslr oops=panic nopti mitigations=off console=ttyS0
	-hda fat:rw:/home/francesco/kAFL/kafl/examples/uefi_ovmf_64/fake_hda
	-fast_vm_reload path=/dev/shm/kafl_uefi/snapshot/,load=off
WARNING: Image format was not specified for 'json:{"fat-type": 0, "backing": {"driver": "vvfat_write_target"}, "dir": "/home/francesco/kAFL/kafl/examples/uefi_ovmf_64/fake_hda", "driver": "vvfat", "floppy": false, "rw": true, "write-target": {"driver": "qcow", "file": {"driver": "file", "filename": "/var/tmp/vl.EZtM3z"}}}' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
[QEMU-NYX] Max Dirty Ring Size -> 1048576 (Entries: 65536)
qemu-system-x86_64: -append only allowed with -kernel option
Worker-00 Failed to connect to Qemu: [Errno 2] No such file or directory: '/dev/shm/kafl_uefi/aux_buffer_0'
Worker-00 Shutting down Qemu after 0 execs..
Worker-00 Failed to launch Qemu.
Worker 0 sent ABORT..
Manager exit: Workers aborted before becoming ready. Likely broken VM or agent setup.
Waiting for Workers to shutdown...
Worker-05 Shutting down Qemu after 0 execs..
Worker-04 Shutting down Qemu after 0 execs..
Worker-01 Shutting down Qemu after 0 execs..
Worker-03 Shutting down Qemu after 0 execs..
Worker-06 Shutting down Qemu after 0 execs..
Worker-02 Shutting down Qemu after 0 execs..
Worker-07 Shutting down Qemu after 0 execs..
```
</details>

I solved the problem by adding `qemu_append:` to the end of [kafl.yaml](https://github.com/IntelLabs/kafl.targets/blob/master/uefi_ovmf_64/kafl.yaml), but now if I try to run the fuzzer again I get this result:
```bash
ERR: 	TNT 483 at position <0x0fabcf17,0x0fabcf17>
[QEMU-NYX] Warning: libxdc_decode returned decoder_error
```

<details>
  <summary>Full output</summary>

[... 213 lines cut ...]
FSOpen: Open 'harness.efi' Success
[Security] 3rd party image[0] can be loaded after EndOfDxe: PciRoot(0x0)/Pci(0x1,0x1)/Ata(Primary,Master,0x0)/HD(1,MBR,0xBE1AFDFA,0x3F,0xFBFC1)/harness.efi.
InstallProtocolInterface: 5B1B31A1-9562-11D2-8E3F-00A0C969723B EB65040
Loading driver at 0x0000E4E4000 EntryPoint=0x0000E4E534F kAFLApp.efi
InstallProtocolInterface: BC62157E-3E33-4FEC-9920-2D3B36D750DF EB9A798
ProtectUefiImageCommon - 0xEB65040
  - 0x000000000E4E4000 - 0x0000000000002440
InstallProtocolInterface: 752F3136-4E16-4FDC-A22A-E5F46812F4CA FE87578
System Table address: 0x0F9EC018
kAFLDxe: NOOP!
kAFLDxe: FUZZ!
SmmDxeFuzz: Calling HarnessRun...
Mapping info: kAFL buffer in heap 0x000000000F087000
Payload size as pages: 0x20
HYPERCALL_KAFL_GET_PAYLOAD
Payload [AB, AB, AB, AB]
No CR3 filtering, crossing SMM boudaries
Main loop go !
@HarnessRun(0x000000000FABB00A)
```
</details>

Also, by modifyind the function [RunkAFLTarget](https://github.com/IntelLabs/kafl.targets/blob/master/uefi_ovmf_64/kAFLDxeTargetNullPkg/Library/kAFLDxeTargetLib/kAFLDxeTargetLib.c#L36) by adding random crashes (either using `kAFL_hypercall(HYPERCALL_KAFL_PANIC, 0);` or something else like `*((unsigned int*)0) = 0xDEAD;`) kAFL reports no crashes.

Am I doing something wrong?

Wenzel, 2023-06-23:
Hi  @francesco-ev !

> I solved the problem by adding qemu_append

Good call.
This is an open issue we have to refactor the default settings, especially `qemu_append`:
https://github.com/IntelLabs/kafl.fuzzer/issues/64

> ERR: 	TNT 483 at position <0x0fabcf17,0x0fabcf17>
> [QEMU-NYX] Warning: libxdc_decode returned decoder_error

This issue means that libxdc couldn't decode the trace provided by Intel PT.
cc @il-steffen if you have some insights with libxdc

> Also, by modifyind the function [RunkAFLTarget](https://github.com/IntelLabs/kafl.targets/blob/master/uefi_ovmf_64/kAFLDxeTargetNullPkg/Library/kAFLDxeTargetLib/kAFLDxeTargetLib.c#L36) by adding [...]

Is your code public somewhere i could try this on my end ?

il-steffen, 2023-06-23:
> ERR: TNT 483 at position <0x0fabcf17,0x0fabcf17>

Have to check with @schumilo for decode problems. Note it printed the code pointer above - maybe that gives you a hint why it wasn't able to decode.

il-steffen, 2023-06-23:
Just seeing this is the UEFI sample, then it is likely a setup/config issue.

@Wenzel @x86-sec  I know we had it working but not sure how streamlined / out-of-the-box this example is?

francesco-ev, 2023-06-25:
> Is your code public somewhere i could try this on my end ?

Sure, I uploaded the code here: [kAFLDxeTargetLib](https://github.com/francesco-ev/kafl.targets/tree/master/uefi_ovmf_64/kAFLDxeTargetNullPkg/Library/kAFLDxeTargetLib)