IntelLabs/kAFL #211: vagrant failed to compile/load kafl_windows box

kotee4ko, 2023-07-23:
All steps performed as in https://intellabs.github.io/kAFL/tutorials/windows/windows_template.html
NYX-KVM is compiled with custom kernel and loaded.
Fuzzing of linux_kernel can be performed via default steps.

```
|root@w00t|:{/usr/src} #_ uname -a
Linux w00t 5.10.73-custom #1 SMP Fri Jul 21 18:47:25 EDT 2023 x86_64 GNU/Linux
|root@w00t|:{/usr/src} #_ packer -v
1.9.2
|root@w00t|:{/usr/src} #_ vagrant -v
Vagrant 2.3.6
```

```
|root@w00t|:{/usr/src/kAFL/kafl/examples/templates/windows} #_ make build ; make import;
source /usr/src/kAFL/kafl/examples/templates/windows/../../venv/bin/activate && packer build -var-file win10.pkrvars.hcl -var cpus=16 -var memory=8192 windows.pkr.hcl
Warning: Bundled plugins used

This template relies on the use of plugins bundled into the Packer binary.
The practice of bundling external plugins into Packer will be removed in an
upcoming version.

To remove this warning, add the following section to your template:

packer {
  required_plugins {
    qemu = {
      source  = "github.com/hashicorp/qemu"
      version = "~> 1"
    }
    ansible = {
      source  = "github.com/hashicorp/ansible"
      version = "~> 1"
    }
    vagrant = {
      source  = "github.com/hashicorp/vagrant"
      version = "~> 1"
    }
  }
}

Then run 'packer init' to manage installation of the plugins

qemu.windows: output will be in this color.

==> qemu.windows: Retrieving ISO
==> qemu.windows: Trying https://go.microsoft.com/fwlink/p/?LinkID=2208844&clcid=0x409&culture=en-us&country=US
==> qemu.windows: Trying https://go.microsoft.com/fwlink/p/?LinkID=2208844&checksum=sha256%3Aef7312733a9f5d7d51cfa04ac497671995674ca5e1058d5164d6028f0938d668&clcid=0x409&country=US&culture=en-us
==> qemu.windows: https://go.microsoft.com/fwlink/p/?LinkID=2208844&checksum=sha256%3Aef7312733a9f5d7d51cfa04ac497671995674ca5e1058d5164d6028f0938d668&clcid=0x409&country=US&culture=en-us => /root/.cache/packer/d731b3f758e61d53033aa8a67d3d8a3050aa1122.iso
==> qemu.windows: Creating floppy disk...
    qemu.windows: Copying files flatly from floppy_files
    qemu.windows: Copying file: answer_files/10/Autounattend.xml
    qemu.windows: Copying file: scripts/fixnetwork.ps1
    qemu.windows: Copying file: scripts/setup_winrm_public.bat
    qemu.windows: Done copying files from floppy_files
    qemu.windows: Collecting paths from floppy_dirs
    qemu.windows: Resulting paths from floppy_dirs : []
    qemu.windows: Done copying paths from floppy_dirs
    qemu.windows: Copying files from floppy_content
    qemu.windows: Done copying files from floppy_content
==> qemu.windows: Found port for communicator (SSH, WinRM, etc): 4209.
==> qemu.windows: Looking for available port between 5900 and 6000 on 0.0.0.0
==> qemu.windows: Starting VM, booting from CD-ROM
    qemu.windows: The VM will be run headless, without a GUI. If you want to
    qemu.windows: view the screen of the VM, connect via VNC without a password to
    qemu.windows: vnc://0.0.0.0:5963
==> qemu.windows: Overriding default Qemu arguments with qemuargs template option...
==> qemu.windows: Waiting 5s for boot...
==> qemu.windows: Connecting to VM via VNC (0.0.0.0:5963)
==> qemu.windows: Typing the boot commands over VNC...
[... 297 lines cut ...]
    qemu.windows (vagrant): Compressing: metadata.json
Build 'qemu.windows' finished after 21 minutes 33 seconds.

==> Wait completed after 21 minutes 33 seconds

==> Builds finished. The artifacts of successful builds are:
--> qemu.windows: 'libvirt' provider box: packer_windows_libvirt.box
vagrant box remove kafl_windows || true
The box you requested to be removed could not be found. No
boxes named 'kafl_windows' could be found.
vagrant box add packer_windows_libvirt.box --name kafl_windows
==> box: Box file was not detected as metadata. Adding it directly...
==> box: Adding box 'kafl_windows' (v0) for provider: 
    box: Unpacking necessary files from: file:///usr/src/kAFL/kafl/examples/templates/windows/packer_windows_libvirt.box
The "metadata.json" file for the box 'kafl_windows' was not found.
Boxes require this file in order for Vagrant to determine the
provider it was made for. If you made the box, please add a
"metadata.json" file to it. If someone else made the box, please
notify the box creator that the box is corrupt. Documentation for
box file format can be found at the URL below:

https://www.vagrantup.com/docs/boxes/format.html
make: *** [Makefile:17: import] Error 1

```

Wenzel, 2023-07-23:
hi @kotee4ko
thanks for reporting the issue.

i see you are using packer 1.9.2, which is very recebt release (https://github.com/hashicorp/packer/blob/v1.9.2/CHANGELOG.md#192-july-19-2023) and i didn't had the chance to test my templates against it.

i would recommend to retry with packer v1.9.1 because the missing metadata json file is of packer's vagrant plugin's responsability.

tell me if that works !

kotee4ko, 2023-07-23:
Thanks for fast reply!

```
(.venv) |root@w00t|:{/usr/src/kAFL/kafl/examples/templates/windows} #_ packer -v
1.9.1

...snipped...

==> qemu.windows (vagrant): Creating a dummy Vagrant box to ensure the host system can create one correctly
==> qemu.windows (vagrant): Creating Vagrant box for 'libvirt' provider
    qemu.windows (vagrant): Copying from artifact: output-windows/win10.qcow2
    qemu.windows (vagrant): Using custom Vagrantfile: Vagrantfile_template
    qemu.windows (vagrant): Compressing: Vagrantfile
    qemu.windows (vagrant): Compressing: box.img
    qemu.windows (vagrant): Compressing: metadata.json
Build 'qemu.windows' finished after 20 minutes 21 seconds.

==> Wait completed after 20 minutes 21 seconds

==> Builds finished. The artifacts of successful builds are:
--> qemu.windows: 'libvirt' provider box: packer_windows_libvirt.box
make[2]: Leaving directory '/usr/src/kAFL/kafl/examples/templates/windows'
make[2]: Entering directory '/usr/src/kAFL/kafl/examples/templates/windows'
vagrant box remove kafl_windows || true
The box you requested to be removed could not be found. No
boxes named 'kafl_windows' could be found.
vagrant box add packer_windows_libvirt.box --name kafl_windows
==> box: Box file was not detected as metadata. Adding it directly...
==> box: Adding box 'kafl_windows' (v0) for provider: 
    box: Unpacking necessary files from: file:///usr/src/kAFL/kafl/examples/templates/windows/packer_windows_libvirt.box
The "metadata.json" file for the box 'kafl_windows' was not found.
Boxes require this file in order for Vagrant to determine the
provider it was made for. If you made the box, please add a
"metadata.json" file to it. If someone else made the box, please
notify the box creator that the box is corrupt. Documentation for
box file format can be found at the URL below:

https://www.vagrantup.com/docs/boxes/format.html
make[2]: *** [Makefile:17: import] Error 1
make[2]: Leaving directory '/usr/src/kAFL/kafl/examples/templates/windows'
```

Wenzel, 2023-07-24:
One thing you could try is to run vagrant in debug mode with

```shell
VAGRANT_LOG=debug vagrant box add packer_windows_libvirt.box --name kafl_windows |& tee vagrant.log
```

See if anything weird comes out regarding metadata.json ?

kotee4ko, 2023-07-24:
Strange things happens... 
`apt purge vagrant* && rm -rf /usr/lib/.vagrant /root/.vagrand.d/ && apt install vagrant` solve the problem.

Hope this help somebody.

@Wenzel  Thanks you for you're time!

closed 2023-07-24 by kotee4ko