IntelLabs/kAFL #212: Redqueen doesn't handle strcmp() calls on linux userspace target

Wenzel, 2023-07-23:
Greetings!

I reach some troubles with redqueen feature: it doesn't handle `strcmp()` calls on linux userspace target.

Launch command:
```
kafl fuzz \
        --kernel /boot/vmlinuz-$(uname -r) \
        --initrd forkserver.cpio.gz \
        --memory 512 \
        --sharedir $PWD \
        --purge -t 2 -ts 1 --log-hprintf --log-crashes \
        --append "nokaslr oops=panic nopti mitigations=off console=ttyS0 hprintf=7" \
        -ip0 0x401000-0x4af000 \
        --redqueen --redqueen-hammer --redqueen-simple --redqueen-hashes --grimoire --radamsa \
        -p 1 --qemu-path /usr/src/QEMU-Nyx/x86_64-softmmu/qemu-system-x86_64

```

Some debugging of qemu-nyx approve, that redqueen strcmp() hooks are reached and works as expected, or, at least, see strcmp() calls is target and could access guest args.

The target, as always, is simple, statically built, small harness with several branches to test integers and strings extractions.

```
In file: /usr/src/QEMU-Nyx/nyx/redqueen.c
   948 static void format_strcmp(uint8_t *buf1, uint8_t *buf2)
   949 {
   950     char  out_buf[REDQUEEN_MAX_STRCMP_LEN * 4 + 2];
   951     char *tmp_hex_buf = &out_buf[0];
   952     for (int i = 0; i < REDQUEEN_MAX_STRCMP_LEN; i++) {
 ► 953         tmp_hex_buf += sprintf(tmp_hex_buf, "%02X", (uint8_t)buf1[i]);
   954     }
   955     *tmp_hex_buf++ = '-';
   956     for (int i = 0; i < REDQUEEN_MAX_STRCMP_LEN; i++) {
   957         tmp_hex_buf += sprintf(tmp_hex_buf, "%02X", (uint8_t)buf2[i]);
   958     }

pwndbg> p tmp_hex_buf 
$5 = 0x7fffeb000310 "C9"
pwndbg> p buf2
$6 = (uint8_t *) 0x7fffeb0004a0 "PWNED!!!"
pwndbg> p buf1
$7 = (uint8_t *) 0x7fffeb000460 "ɵ\032<\237b22\032\062\062\016\020g\022UUU"
```

full backtrace:

```
pwndbg> bt full
#0  0x0000555555c0f3a4 in format_strcmp (buf1=buf1@entry=0x7fffeb000460 "ɵ\032<\237b22\032\062\062\016\020g\022UUU", buf2=buf2@entry=0x7fffeb0004a0 "PWNED!!!") at /usr/src/QEMU-Nyx/nyx/redqueen.c:953
        i = <optimized out>
        out_buf = "C9\000\000\000\000\000\000 p\034\334\377\177\000\000\000\000\000\000\000\000\000\000@\244\210\004\000\000\000\000@s\311VUU\000\000\000\003홨\"\261\065`\003\000\353\377\177\000\000\b\000\000\000\000\000\000\000@\244\210\004\000\000\000\000\360\003\000\353\377\177\000\000h\376\377\377\377\377\377\377 p\034\334\377\177\000\000\016p\375\367\377\177\000\000S+\254UUU\000\000\b\000\000\000\000\000\000\000\260\"\265VUU\000\000\340\353\377\377\377\177\000\000\b\000\000\000\000\000\000\000@\244\210\004\000\000\000\000\000\003홨\"\261\065\357\200H", '\000' <repeats 13 times>, "\360\003\000\353\377\177\000\000\000\360\377\377\377\377\017\000\357\200H\000\000\000\000\000\210K\300UUU"...
        tmp_hex_buf = 0x7fffeb000310 "C9"
        res = 0x555555ac21c7 <flatview_read_continue+119> "L\001\365M\001\364L)\363\017\204Q\001"
        env = <optimized out>
        __func__ = "format_strcmp"
        rip = <optimized out>
        __PRETTY_FUNCTION__ = "format_strcmp"
#1  0x0000555555c0f948 in test_strcmp (arg1=140737136034976, arg1@entry=140737353969678, arg2=arg2@entry=4751599) at /usr/src/QEMU-Nyx/nyx/redqueen.c:1002
        cpu = 0x555556c97340
        buf1 = "ɵ\032<\237b22\032\062\062\016\020g\022UUU\000YXﾭ\300\000\000\000\357\276ޭ\276", '\363' <repeats 23 times>, "\255\336쭾\357\256", <incomplete sequence \374>
        buf2 = "PWNED!!!\000Something went wrong\n\000This should never happen :)\n\000xeon"
        __PRETTY_FUNCTION__ = "test_strcmp"
#2  0x0000555555c10024 in test_strcmp_sys_v () at /usr/src/QEMU-Nyx/nyx/redqueen.c:1035
        env = <optimized out>
        arg1 = 140737353969678
        arg2 = 4751599
        __func__ = "test_strcmp_sys_v"
        mode = 1
        cpu = <optimized out>
[... 115 lines cut ...]
Worker-00 RedQ-Dict: Have performed 0 iters
Received new input (exit=crash): b'!NYX\xef\xbe\xad\xde'ndings: <0, 0, 0>
00:02:58: Got    8 from    6: exit=C,  9/ 0 bits,  0 favs, 0.13msec, 0.0KB (afl_splice)
Worker-00 HAVOC times: afl: 6.0, splice: 14.7, grim: 0.0, rdmsa: 0.00, 0>
```
So, the magic string was found and successfully injected to input, the harness react by crashing, as expected, and kAFL log this. But not new path in corpus/regular neither correct (crashing) payloads in corpus/crashes are present.
```
(.venv) |root@w00t|:{/usr/src/kAFL/kafl/examples/linux-user/forkserver} #_ hexyl /dev/shm/kafl_root/corpus/crash/payload_00008 
┌────────┬─────────────────────────┬─────────────────────────┬────────┬────────┐
│00000000│ 21 4e 59 58 ef be ad de ┊                         │!NYX××××┊        │
└────────┴─────────────────────────┴─────────────────────────┴────────┴────────┘
(.venv) |root@w00t|:{/usr/src/kAFL/kafl/examples/linux-user/forkserver} #_ hexyl /dev/shm/kafl_root/corpus/
crash/   kasan/   regular/ timeout/ 
(.venv) |root@w00t|:{/usr/src/kAFL/kafl/examples/linux-user/forkserver} #_ hexyl /dev/shm/kafl_root/corpus/regular/payload_0000
payload_00001  payload_00002  payload_00003  payload_00004  payload_00005  payload_00006  payload_00007  
(.venv) |root@w00t|:{/usr/src/kAFL/kafl/examples/linux-user/forkserver} #_ hexyl /dev/shm/kafl_root/corpus/regular/payload_00007 
┌────────┬─────────────────────────┬─────────────────────────┬────────┬────────┐
│00000000│ 21 4e 59 58 ef be ad de ┊                         │!NYX××××┊        │
└────────┴─────────────────────────┴─────────────────────────┴────────┴────────┘
```  
------------
Any advices?
Thanks.

_Originally posted by @kotee4ko in https://github.com/IntelLabs/kAFL/issues/12#issuecomment-1646964469_

kotee4ko, 2023-07-23:
---upd---

It seems, that if there are unused-bytes in payload data before place WHERE redq-dict C-string must be put -- kAFL can't save such testcase to corpus correct.
In next harness between `0xdeadbeef` constant and C-String `"PWNED!!!"` two unused bytes present.
```
  94   │     if (len >= 4) {
  95   │ 
  96   │       /* set a byte in the bitmap to guide your fuzzer */
  97   │       //((uint8_t *)trace_buffer)[0] = 0x1;
  98   │       if (payload_buffer->data[0] == '!') {
  99   │ 
 100   │         //((uint8_t *)trace_buffer)[1] = 0x1;
 101   │         if (payload_buffer->data[1] == 'N') {
 102   │ 
 103   │           //((uint8_t *)trace_buffer)[2] = 0x1;
 104   │           if (payload_buffer->data[2] == 'Y') {
 105   │ 
 106   │             //((uint8_t *)trace_buffer)[3] = 0x1;
 107   │             if (payload_buffer->data[3] == 'X') {
 108   │ 
 109   │               //((uint8_t *)trace_buffer)[4] = 0x1;
 110   │               /* Notifiy the hypervisor and the fuzzer that a "crash" has
 111   │                * occured. Also a string is passed by this hypercall (this is
 112   │                * currently not supported by AFL++-Nyx) */
 113   │               if (*(int*)&(payload_buffer->data[4]) == 0xdeadbeef) {
 114   │             if (!strcmp(&(payload_buffer->data[10]), "PWNED!!!")) {
 115   │             hprintf("PWNED\n");
 116   │             char *magic;
 117   │             magic = &(payload_buffer->data[20]);
 118   │             if (!memcmp(magic, "RedQueen", sizeof("RedQueen"))) {
 119   │                 void **crash = 0xdeadbeef;
 120   │                 *crash = 0xcafefeed;
 121   │                           kAFL_hypercall(HYPERCALL_KAFL_PANIC_EXTENDED,
 122   │                                  (uintptr_t) "Something went wrong\n");
 123   │             }
 124   │             }
 125   │               }
 126   │             }
 127   │ 
 128   │           }
 129   │ 
 130   │         }
 131   │ 
 132   │       }
 133   │ 
 134   │     }
```

Backend pass that code and increase edges count up to 59, but:

```
(.venv) |root@w00t|:{/usr/src/kAFL/kafl/examples/linux-user/forkserver} #_ hexyl /dev/shm/kafl_root/corpus/regular/payload_00007
┌────────┬─────────────────────────┬─────────────────────────┬────────┬────────┐
│00000000│ 21 4e 59 58 ef be ad de ┊                         │!NYX××××┊        │
└────────┴─────────────────────────┴─────────────────────────┴────────┴────────┘
(.venv) |root@w00t|:{/usr/src/kAFL/kafl/examples/linux-user/forkserver} #_ hexyl /dev/shm/kafl_root/corpus/regular/payload_00008
┌────────┬─────────────────────────┬─────────────────────────┬────────┬────────┐
│00000000│ 21 4e 59 58 ef be ad de ┊                         │!NYX××××┊        │
└────────┴─────────────────────────┴─────────────────────────┴────────┴────────┘
```

In case of next branches:

```
 113   │               if (*(int*)&(payload_buffer->data[4]) == 0xdeadbeef) {
 114   │             if (!strcmp(&(payload_buffer->data[8]), "PWNED!!!")) {

```

all works as expected

```
(.venv) |root@w00t|:{/usr/src/kAFL/kafl/examples/linux-user/forkserver} #_ hexyl /dev/shm/kafl_root/corpus/regular/payload_00008
┌────────┬─────────────────────────┬─────────────────────────┬────────┬────────┐
│00000000│ 21 4e 59 58 ef be ad de ┊ 50 57 4e 45 44 21 21 21 │!NYX××××┊PWNED!!!│
└────────┴─────────────────────────┴─────────────────────────┴────────┴────────┘
```

I have some experience with AFL's cmplog backend stuff, in qemu userspace.
But such sort of behaviour is highly discouraged, so i think i  need some help.
Thanks.

kotee4ko, 2023-07-24:
Please, can anybody test this behaviour and discard or approve it?
In my setup it seems that when any space between parts of payload present -- it became unpassable constrain for kAFL.

Wenzel, 2023-07-24:
poke @il-steffen
any chance you can have a quick at this redqueen issue ?
thanks !

il-steffen, 2023-07-25:
> So, the magic string was found and successfully injected to input, the harness react by crashing, as expected, and kAFL log this. But not new path in corpus/regular neither correct (crashing) payloads in corpus/crashes are present.

Thanks for debugging this. Can you check the logic on frontend side as well? :-)

The Redqueen analysis + first mutation stage is here:
https://github.com/IntelLabs/kafl.fuzzer/blob/master/kafl_fuzzer/worker/state_logic.py#L346

The "state logic" implements the mutation pipeline of the workers. For the Redqueen stage, it first performs coloring + tracing using `execute_redqueen()`, which skips the crash + feedback analysis disabled (using execute_naked() function). Qemu writes the relevant traces as files in the workdir/redqueeen-something/ folder where they are picked up by the frontend (Redqueen "InfoGatherer" stuff). It then runs the proposals using the normal "execute()" call where crashes + new coverage is analyzed normally. There is also a "havoc" variant of this, called as part of the havoc stage, where it simply uses tokens found by Redqueen as a dictionary: https://github.com/IntelLabs/kafl.fuzzer/blob/master/kafl_fuzzer/worker/state_logic.py#L473

I expect your issue is related to how the redqueen mutator handles the input strings. I am not sure if the initial stage should have found the input already, but in principle the colorizer will first do some input expansion so the unused bytes should have been passed. If the trace logic found the compare, you should be able to see it written to the traces, where it is picked up by the InfoGatherer stuff. 

Your log indicates that the input was eventually found in the later havoc stage using the RQ-dictionary:
> Worker-00 RQ-Dict: attempting b'PWNED!!!\x00Somethi'
> Worker-00 RedQ-Dict: Have performed 0 iters   (???????)
> Received new input (exit=crash): b'!NYX\xef\xbe\xad\xde'ndings: <0, 0, 0>

However, the supposedly crashing payload also seems wrong here. This "received new input" is what the worker eventually hands off to the manager process as the crashing input, and the manager will store it after checking that the bitmap is unique. But it seems that even the worker/self.execute() got the wrong input here, so maybe the execution is out of sync somehow?

kotee4ko, 2023-07-26:
Kindly appreciated for reply, guys.

I was able to move on a bit with strings.
```
|root@w00t|:{/usr/src/kAFL} #_ find /dev/shm/kafl_root/corpus/regular/
/dev/shm/kafl_root/corpus/regular/
/dev/shm/kafl_root/corpus/regular/payload_00008
/dev/shm/kafl_root/corpus/regular/payload_00007
/dev/shm/kafl_root/corpus/regular/payload_00006
/dev/shm/kafl_root/corpus/regular/payload_00005
/dev/shm/kafl_root/corpus/regular/payload_00004
/dev/shm/kafl_root/corpus/regular/payload_00001
/dev/shm/kafl_root/corpus/regular/payload_00002
/dev/shm/kafl_root/corpus/regular/payload_00003
|root@w00t|:{/usr/src/kAFL} #_ hexyl /dev/shm/kafl_root/corpus/regular/payload_00008
┌────────┬─────────────────────────┬─────────────────────────┬────────┬────────┐
│00000000│ 21 4e 59 58 ef be ad de ┊ 00 00 50 57 4e 45 44 21 │!NYX××××┊00PWNED!│
│00000010│ 21 21 00 50             ┊                         │!!0P    ┊        │
└────────┴─────────────────────────┴─────────────────────────┴────────┴────────┘
```
There were several issues, at least.
0) https://github.com/IntelLabs/kafl.fuzzer/blob/82c87450bc6812773be8b051b465ec010aaf8710/kafl_fuzzer/worker/state_logic.py#L477C13-L477C13
why limitation of 256?
1) Redqueen's data was append  to the dict with junky null bytes in front and in tail.
https://github.com/IntelLabs/kafl.fuzzer/blob/82c87450bc6812773be8b051b465ec010aaf8710/kafl_fuzzer/technique/redqueen/cmp.py#L193
Should we `.strip(b'\x00')` before processing?

2) Sometimes even just zero-bytes were append to the dict.
https://github.com/IntelLabs/kafl.fuzzer/blob/82c87450bc6812773be8b051b465ec010aaf8710/kafl_fuzzer/technique/havoc_handler.py#L284
should we check `len(.strip(b'\x00')) > 0` here?

3) As was told here https://github.com/IntelLabs/kafl.fuzzer/blob/82c87450bc6812773be8b051b465ec010aaf8710/kafl_fuzzer/technique/havoc_handler.py#L297C1-L297C83
variable and out-of-bound offsets -- very good idea, because without them we couldn't discover that the target has/need more large payload, etc.

Btw, in CMPLOG impl of qemuafl there was very cool stuff, which split ints to bytes. I don't sure if it possible to do that trick with intel-pt (but think it should), if it isn't done yet?  

4) https://github.com/IntelLabs/kafl.fuzzer/blob/82c87450bc6812773be8b051b465ec010aaf8710/kafl_fuzzer/technique/havoc_handler.py#L318
if we got static dict we would never use rq?

5) https://github.com/IntelLabs/kafl.fuzzer/blob/82c87450bc6812773be8b051b465ec010aaf8710/kafl_fuzzer/technique/havoc_handler.py#L284C2-L284C2
What about uint16_t ? two bytes?
And what about integers (with nullbytes) at all? 
If i understand right the dict is one for all types of rq findings?

Atm it pass first strcmp() and totally stacked on second memcmp().
```
 124   │         if (*(int*)&(payload_buffer->data[4]) == 0xdeadbeef) {
 125   │           if (!strcmp(&(payload_buffer->data[10]), "PWNED!!!")) {
 126   │             hprintf("PWNED\n");
 127   │             char *magic;
 128   │             magic = &(payload_buffer->data[20]);
 129   │             if (!memcmp(magic, "RedQueen", sizeof("RedQueen"))) {
 130   │                 void **crash = 0xdeadbeef;
 131   │                 *crash = 0xcafefeed;
 132   │                           kAFL_hypercall(HYPERCALL_KAFL_PANIC_EXTENDED,
 133   │                                  (uintptr_t) "Something went wrong\n");
 134   │             }
 135   │           }
 136   │         }
 137   │       }
 138   │ 

```
```
|root@w00t|:{/usr/src/kAFL} #_ grep -rn PWNED /dev/shm/kafl_root/
/dev/shm/kafl_root/hprintf_03.log:1:PWNED
grep: /dev/shm/kafl_root/aux_buffer_9: binary file matches
grep: /dev/shm/kafl_root/aux_buffer_8: binary file matches
grep: /dev/shm/kafl_root/aux_buffer_7: binary file matches
grep: /dev/shm/kafl_root/aux_buffer_6: binary file matches
[... 94 lines cut ...]
Worker-04 RedQ-Dict: Have performed 0 iters
Worker-06 HAVOC times: afl: 4.8, splice: 11.9, grim: 0.0, rdmsa: 0.00, 0>
perf updated for node 4: 0.13 => 0.34
Worker-06 __perform_rq_dictt: rq_dict len 2, [{4203020: {b'\xde\xad\xbe\xef'}, 4203045: {b'PWNED!!!\x00PWNED\n\x00'}}];
payload_array len 4, [b'!N4\xd4']

Worker-06 RedQ-Dict: Have performed 0 iters
Worker-09 HAVOC times: afl: 9.6, splice: 23.7, grim: 0.0, rdmsa: 0.00, 0>
perf updated for node 6: 0.13 => 0.26
Worker-09 __perform_rq_dictt: rq_dict len 0, [{}];
payload_array len 4, [b'!NYX']

Worker-09 RedQ-Dict: Have performed 0 iters
Worker-08 HAVOC times: afl: 4.2, splice: 10.3, grim: 0.0, rdmsa: 0.00, 0>
perf updated for node 7: 0.14 => 0.60
Worker-08 __perform_rq_dictt: rq_dict len 0, [{}];
payload_array len 20, [b'!NYX\xef\xbe\xad\xde\x00\x00PWNED!!!\x00P']

Worker-08 RedQ-Dict: Have performed 0 iters
Worker-02 HAVOC times: afl: 3.0, splice: 7.4, grim: 0.0, rdmsa: 0.0 0, 0>
perf updated for node 1: 0.13 => 0.54
Worker-02 __perform_rq_dictt: rq_dict len 0, [{}];
payload_array len 8, [b'!NYX\xef\xbe\xad\xde']

```

il-steffen, 2023-07-26:
Whew, thanks!

Suggestion @Wenzel: Lets collect/create a couple redqueen trace files that we load into RedqueenInfoGatherer() and call get_proposals() on. Then fix up some of this code to generate good inputs.

The kafl debug tool has some very old, very unlikely to work, code for this: https://github.com/IntelLabs/kafl.fuzzer/blob/master/kafl_fuzzer/debug/core.py#L321

kotee4ko, 2023-10-16:
Hi guys, any upd on this?

Wenzel, 2023-10-17:
Hi @kotee4ko,

this issue has been in my TODO list for a while, unfortunately it's not trivial to investigate and solve.
On top of that i'm not familiar with RedQueen's internals and implementation details on kAFL.
I can try to allocate some time on Friday, and estimate how much work is needed.

I'd like to point out that I really appreciate the quality of your bug report though, thank you for this.

m1lua, 2024-12-11:
hi. any news on this?