IntelLabs/kAFL #220: There is an error in windows Tutorial

5angjun, 2023-09-02:
I recommanded you to test your `windows tutorial` in Cleaned ubuntu 22.04

I use intel i-7 12700K CPU that supported intel PT.
`
bob@bob:~/Desktop$ uname -a
Linux bob 6.0.0-nyx+ #1 SMP PREEMPT_DYNAMIC Wed Mar 8 22:55:35 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
bob@bob:~/Desktop$ 

`
Can i have a feedback from this issue?
`
git clone https://github.com/IntelLabs/kAFL.git
cd kAFL

git checkout release_v0.7
make deploy
make env
kafl fuzz

### make windows vm ###
make deploy -- --tags examples,examples-template-windows
cd ~/kAFL/kafl/examples/templates/windows
make build  
make import

sudo apt install qemu qemu-kvm libvirt-clients libvirt-daemon-system virtinst bridge-utils -y
sudo systemctl enable libvirtd -y
sudo systemctl start libvirtd -y
sudo apt install -y ebtables -y
sudo setfacl -m user:$USER:rw /var/run/libvirt/libvirt-sock

sudo systemctl enable libvirtd
sudo systemctl start libvirtd

### Error occured in this Stage ##
cd ../../windows_x86_64
make init

`
[image]

5angjun, 2023-09-02:
I found the solution of this problem

Need to Run All command & script in User mode not Super root mode.

I think you add the warning message that alert to user to run Ansible or make command  in user mode.

closed 2023-09-02 by 5angjun

Wenzel, 2023-09-02:
hi @5angjun 

Thank you for taking the time to share your experience and the solution you found.
We appreciate your contribution to the project.

Indeed, we assume that all the commands will be run as non-root user, since root is only required for administrative tasks.
As this is common practise among Linux users, I don't think adding a note is necessary in the documentation as this point.

If this type of issue is raising again, I will consider it however.

Again, thank you for your inputs and feel free to open more issue, should you encounter any problem with kAFL.