IntelLabs/kAFL #222: GET_HOST_CNOFIG magic/version mismatch! error

kbh1860, 2023-09-03:
Hello! I try to run kAFL targeted windows -> referenced https://github.com/Kharos102/kafl.targets

When I input this command to fuzzing

`kafl fuzz -w ./outTest --image /home/rolling/windows/overlay_0.qcow2 -m 3000 --radamsa-path=/home/rolling/kAFL/kafl/radamsa/bin/radamsa --qemu-path=/home/rolling/kafl.qemu/x86_64-softmmu/qemu-system-x86_64 --purge --seed-dir=/home/rolling/kAFL/seedTest --redqueen --append ""
`
but i got error.. like this
[image]

can I get hint for this error..??

**my environment**
`
(.venv) rolling@rolling-System-Product-Name:~/kAFL$ uname -a
Linux rolling-System-Product-Name 6.0.0-nyx+ #1 SMP PREEMPT_DYNAMIC Wed Mar 8 22:55:35 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

(.venv) rolling@rolling-System-Product-Name:~/kAFL$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal
`

Wenzel, 2023-09-03:
Hi,

that error message is referenced here in Kharos's fork:
https://github.com/Kharos102/kafl.targets/blob/master/windows_x86_64/src/fuzzer/vuln_test.c#L99

What it means is that the [`NYX_HOST_MAGIC`](https://github.com/Kharos102/kafl.targets/blob/master/nyx_api.h#L139) used by the kafl agent in the guest doesn't match the Nyx API version used by QEMU on the host.

I don't know about Kharos's fork at this point, but I've merged most of his changed into the official kafl.targets, and built a Windows tutorial on top of it.
https://intellabs.github.io/kAFL/

I advise you to give it a try, as I'm maintaining and providing support for these.

kbh1860, 2023-09-04:
Thank you for your advice!

I read the docs that you provided, But I don't know how to insert target driver.. :cry: 

I can found compile the harness routine in **/home/rolling/kAFL/kafl/examples/windows_x86_64/Makefile**
[image]
like this

But How can i found the targeted device driver file(.sys)..?? We can only fuzzing example driver..??

If you know the related issue, can you advice to me.. :pray:

5angjun, 2023-09-04:
i stucked with this same problem that fuzz target driver not for example driver.

We some specific docs how to loading pre compiled driver / Program that use driver in qemu image.

It is having difficulties because it operates in a different way from kAFL0.2.

We need some kindful docs to set our target drivers in fuzzing qemu image

Wenzel, 2023-09-04:
> I read the docs that you provided, But I don't know how to insert target driver.. 😢

Thanks for this feedback, i'm committed to fix the gaps and improve the documentation, especially the tutorial.

> But How can i inserted the targeted device driver file(already builded .sys file)..? We can only fuzzing example driver..??

The way I structured the Windows driver example is that one userspace component (`vuln_test.exe`) will be the harness and send a buffer to the vulnerable driver (`kAFLvulnerabledriver.sys`).

When you run `make provision_driver` on the host, this will
- compile the userspace `vuln_driver.c` code with `x86_64-w64-mingw32-gcc`, like you found in the Makefile: x86_64-w64-mingw32-gcc
- restore the guest snapshot
- provision it with [`setup_target.yml`](https://github.com/IntelLabs/kafl.targets/blob/master/windows_x86_64/setup_target.yml) playbook
- the playbook will upload both the `vuln_test.exe` final binary, as well as the driver **sources** to [compile it](https://github.com/IntelLabs/kafl.targets/blob/master/windows_x86_64/setup_target.yml#L35) within the **guest**. In fact, you need Windows SDK and WDK to compile a driver, so this wouldn't be possible on the host with `x86_64-w64-mingw32-gcc`.
- and finally install it as a driver

Now your usecase is different, since you already have a compiled `.sys` driver.

You could put your .sys driver file in the `bin` directory, so it would be automatically uploaded in the guest with:
https://github.com/IntelLabs/kafl.targets/blob/master/windows_x86_64/setup_target.yml#L15

At this stage, the questions are:
- how to you plant to install your driver ?
- does your driver contains the kAFL agent harness, or will it be triggered by userspace like in the example ?

Your issue and use case is very valuable to me to better understand how you are using kAFL as an end-user, and how I can remove blockers and improve the docs.

Thanks !

kbh1860, 2023-09-04:
First, Thank you for your reply :D

And, Currently I think the problem that I asked you.

My environment is **my driver doesn't contain the kAFL agent harness**, so i must build harness in userspace (previous fuzz step)

So, I Think edit the `setup_target.yml` to build a qemu's image 

[image]
like this. I edit the win_copy's `src` to my device driver file path.

and I edit the win_command that register the driver.

Lastly, I edit the harness(vuln_test.c ) that comfortably to my device driver

Finally, I entered the `make init` and Can get the effective result!

Thank you for your advice!

closed 2023-09-04 by kbh1860