IntelLabs/kAFL #224: fuzzing for drivers that are loaded only when a particular process is running

hyjun0407, 2023-09-08:
How do I fuzz for drivers that have dependencies between drivers (IofCallDriver) and work normally only when certain processes are running? In Kafl 0.2, I could run a specific process and execute Loader.exe and then take a snapshot, but in 0.7, I don't know if there's a similar way.

Wenzel, 2023-09-08:
Hi @hyjun0407 

Thanks for your question and interest in kAFL.
The [`Loader.exe`](https://github.com/IntelLabs/kafl.targets/blob/kafl-v0.5/windows_x86_64/src/loader/loader.c) you are mentionning has been deprecated and removed from the kafl.targets repo.

In fact, It was relying on [`HYPERFCALL_KAFL_GET_PROGRAM`](https://github.com/IntelLabs/kafl.targets/blob/kafl-v0.5/windows_x86_64/src/loader/loader.c#L181), which is now [deprecated](https://intellabs.github.io/kAFL/reference/hypercall_api.html#deprecated).

However, the feature that [you are referring to](https://github.com/IntelLabs/kafl.targets/blob/kafl-v0.5/windows_x86_64/src/loader/loader.c#L168) (the ability to take a pre-snapshot in a specific context) still exists thanks to [`HYPERCALL_KAFL_LOCK`](https://intellabs.github.io/kAFL/reference/hypercall_api.html#lock).

The examples haven't been updated yet to highlight that feature.

In the meantime, you can take the code from Loader.c, tweak it to your needs, and capture that pre-snapshot.

If this fits your use case, please give my your feedback, i would be happy to integrate it into the examples and update the documentation accordingly.

hyjun0407, 2023-09-08:
My use case is what I want to take snapshots after running a particular process that load driver. from what I understand, Lock Hypercall can do that (because the document says I can snapshot it when you want.. Please tell me if you misunderstood) If added to the project, it would be good.

Wenzel, 2023-09-08:
> My use case is what I want to take snapshots after running a particular process that load driver

Yes the **LOCK** hypercall is the right feature here.
However, if you can easily automate how you bring the guest into this context, assuming it doesn't involve user interaction) (via powershell script for example), you can try to update the [`setup_target.yml`](https://github.com/IntelLabs/kafl.targets/blob/master/windows_x86_64/setup_target.yml#L48) playbook provided here.

It configures `vuln_test.exe` to be started during the Windows boot sequence, but you can plug anything there 🙂 

Otherwise, yes the **LOCK** hypecall is your friend !

cc @il-steffen, if you'd like to add anything

hyjun0407, 2023-09-08:
Unfortunately, I've already tried everything using ansible playbook, but it's been excluded because it includes installation through gui. (I've already tried all automation through playbook) However, there is some confusion. After creating a Windows base image through 'make build', copy the base image and run qemu, call the LOCK hypercall after doing the desired action, and then position the snapshot with the --snapshot argument when start fuzzing? I didnt read some api and hypercall fully, so I could misunderstood :/ Thanks for your assistance

hyjun0407, 2023-09-08:
And additionally, it would be very convenient if you could add a script that would automatically run qemu like the loader in kafl0.2, and allow users to save snapshots when they want (for those who are not familiar with ansible, or have elements to interact with users eg,.gui)

Wenzel, 2023-09-08:
> However, there is some confusion. After creating a Windows base image through 'make build', copy the base image and run qemu, call the LOCK hypercall after doing the desired action, and then position the snapshot with the --snapshot argument when start fuzzing?

I haven't thought through the combination of the LOCK hypercall with Vagrant's workflow and interface yet.

but removing Vagrant, yes, you would first run your image with kafl fuzz until that LOCK hypercall, specifying the pre-snapshot [`-snapshot`](https://intellabs.github.io/kAFL/reference/fuzzer_configuration.html#qemu-snapshot) parameter.

The run it a second time to resume the execution this snapshot and fuzz your target from there.

Again, I need to build a concrete example on top of this.

Wenzel, 2023-09-08:
> And additionally, it would be very convenient if you could add a script that would automatically run qemu like the loader in kafl0.2, and allow users to save snapshots when they want (for those who are not familiar with ansible, or have elements to interact with users eg,.gui)

I'm not familiar with kAFL 0.2, nor the loader.
If I understand correctly, you would like a quick script to only execute the Loader in the guest, and especially the LOCK hypercall to generate pre-snapshots ?

hyjun0407, 2023-09-09:
For example, if I want to fuzz  a.sys of a Chrome, use the GUI installer to install Chrome(by launching qemu), and I would like to have a 1-click-program that saves the status as a snapshot after checking that sys is loaded.