IntelLabs/kAFL #228: Vagrant needs qemu-img

M3noetius, 2023-09-11:
Good afternoon,

In order to compile QEMU-NYX the --disable--tools is disabled [[here](https://github.com/nyx-fuzz/QEMU-Nyx/blob/qemu-nyx-4.2.0/compile_qemu_nyx.sh#L56)]. 

Am I suppose to go to QEMU-nyx master [[here](https://github.com/nyx-fuzz/QEMU-Nyx/tree/master)] and compile the tool ? 
I see that the changes are at the "block-backend,c" and I guess this will have an affect on the image used. [[here](https://github.com/nyx-fuzz/QEMU-Nyx/blob/qemu-nyx-4.2.0/block/block-backend.c#L50)] [[git_blame](https://github.com/nyx-fuzz/QEMU-Nyx/commit/cd702b528c25655632fe3d3c720cdf55a6b9ca9d#diff-a75c6eb35db61021f6306b7c74a9ce914875909178f9cb244ceccf221de3de32R344)]

Thank you

Wenzel, 2023-09-11:
Hi @M3noetius and thank you for opening an issue.

I wasn't aware of the problem you are reporting, since I'm simply relying on the system package to get `qemu-img` tool, from `/usr/bin/qemu-img` (`qemu-system-x86` package).

I'm not fully understanding all the QEMU-Nyx modifications at this stage.
However, the `BlockBackend` struct seems to be defined when QEMU_NYX is **undefined**.
``` c
#ifndef QEMU_NYX
struct BlockBackend {
    char *name;
    int refcnt;
    BdrvChild *root;
```

and judging from the `blk_new` functions, the only field added by Nyx is the `cow_cache`:
https://github.com/nyx-fuzz/QEMU-Nyx/blob/qemu-nyx-4.2.0/block/block-backend.c#L345

In any case, Packer and the Windows template here isn't configured to use the QEMU-Nyx binary that you built previously.
In fact, the `PATH` hasn't been modified to include `kafl/kafl/qemu/x86_64-softmmu/qemu-system-x86_64`.

So in my case, I also the system packaged QEMU to build the Windows template, and then fuzz it with QEMU-Nyx, doesn't seem to be a problem.

@il-steffen can you shed some lights regarding the `block-backend.c` modifications, and whether they are relevant for QEMU images being built ?

Wenzel, 2023-10-02:
Closed by https://github.com/IntelLabs/kAFL/pull/234

closed 2023-10-02 by Wenzel